espressif/arduino-esp32 #1971: ESP32 LoRa Heltec Serial

Maddesh1337, 2018-10-16:
### Hardware:
Board:							ESP32 LoRa Heltec
IDE name:							Arduino IDE
Flash Frequency:					80Mhz
Computer OS:                                            Windows 10

### Description:
I am new to Arduino so please excuse me, if I may ask a "stupid" question here. But I am stucked in my project and dont know how to solve it.
I try to build a sender-unit for the LoRaWAN. Therefore I tried to start small and start with reading out the data of my digital electricity meter via D0-interface. I was able to build a ESP8266 Wemos D1 mini working (based on online projects) but on the ESP32 LoRa Heltec board the Serial2.availble is not working. I am using GND and RX pin (PIN 17). 
If you wonder about the multiple serial prints: I just wanted to check which part is working and which is not.
Do you got a hint for me to solve the problem?

At the moment the serial print out is

`Serial available
Configuring access point...HTTP server started
    
0...
7...
0...
7...
0...
7...
0...
7...
0...
7...
0...
7...`

Here is the code I am using.
`
// Buffer for serial reading
int  serIn;             // var that will hold the bytes-in read from the serialBuffer
byte datagram[1000];    // array that will hold the different bytes
int  serindex = 0;    // index of serInString[] in which to insert the next incoming byte

float global[15];
int a = 0;
int b = 0;
int c = 0;
int d = 0;
float kwh = 0;
float gmin = 0;
float ghour = 0;
float gday = 0;
float dmin = 0;
float dhour = 0;
float dday = 0;

int  serialbyte;        // Byte to store serial input data

void setup() {

  Serial2.begin(9600,SERIAL_8N1, 16, 17); //Baud rate, parity mode, RX, TX
  Serial.begin(9600);
  
  while (!Serial2) {
    delay(5);
    Serial.println("Kein Signal Serial2");
    }
    
  Serial.println("Serial available");

  Serial.println();
  Serial.print("Configuring access point...");

  // ----------------------------------------------------if DNSServer is started with "*" for domain name, it will reply with
  //----------------------------------------------------- provided IP to all DNS request

  // --------------------------------------------------- replay to all requests with same HTML

  Serial.println("HTTP server started");
  Serial.println("    ");

}

void loop() {
    Serial.println("0...");
          //
          // Clear Serial Data
          //
  while (Serial2.available()) {
    Serial.println("1...");
    while (Serial2.available()) {
      
        Serial.println("2...");
        serialbyte = Serial2.read(); //Read Buffer to clear
        Serial.println("3...");
    }
    //Serial.print("F");
    delay(10);  // wait approx 10 bytes at 9600 Baud to clear bytes in transmission
}

  //
  // assume, that there is now a pause. Wait for start of transmission
  //
  int flashcount = 0;
  while (Serial2.available()) {    //captive portal hängt sich hier auf
        Serial.println("4...");
    flashcount++;
    // Serial.println(flashcount);
    if (flashcount == 400) {
          Serial.println("5...");
    }
    else if (flashcount > 500) {
          Serial.println("5...");
      flashcount = 0;
    }
    else {
          Serial.println("6...");
      delay(5);  // wait 5 ms for new packets
    }
  }

  // We got some bytes. read until next pause
  // Serial.println("Reading serial data");
  Serial.println("7...");
  Serial2.setTimeout(500);   // Set Timeout to 500ms.
  serindex = Serial2.readBytes(datagram, 1000);

  if (serindex > 0) {
    Serial.println("8...");
    Serial.print("Datagram received. Total Bytes: "); Serial.println(serindex);

    //------------------------------------------------------------- Um dargestellte Werte zu vergleichen
    
    kwh =   (0.0001*(datagram[153] * 16777216 + datagram[154] * 65536 + datagram[155] * 256 + datagram[156] * 1 ));

    
    for (b = 0; b < 15; b++) {
      if (b < 14) {
        global[b] = global[b + 1];
      }
    }
    global[14] = (0.1 * (datagram[215] * 16777216 + datagram[216] * 65536 + datagram[217] * 256 + datagram[218] * 1 ));
    Serial.print("14"); Serial.print(" : "); Serial.print(global[14], DEC); Serial.println("W");

} 
}`

Jeroen88, 2018-10-16:
> I am using GND and RX pin (PIN 17).

I do not understand what your issue is, however the default Serial2 Rx pin is 16. Did you connect the meter serial port to pin 16?
Futher while(!Serial2) ... is not necessary, this test will always return true.

Maddesh1337, 2018-10-16:
Thank you for your fast reply!
 Thats a mistake here. I am using pin 16. I use a phototransistor on the D0-interface with is connected to GND and (RX) pin 16 at the ESP32.

My problem is, that I not "reading" the incoming data from the meter with the ESP32, but the code is working at the EPS8266 Wemos D1 mini.

Jeroen88, 2018-10-16:
Could be anything, Is the phototransistor connected to the ESP32 in a similar way as to the Wemos (You need +5v or +3v3 too I assume). Are the port parameters exactly the same (baudrate, configuration (SERIAL_8N1)). Is the signal inverted?
I use Serial2 on GPIO35 and that works like a charm (pass 35 in stead of 16 to Serial2.begin)
Your test program could be far more easy (not tested):

```
void setup() {
  Serial.begin(9600); // Or Serial.begin(115200) and use your serial monitor at this baudrate
  Serial.begin(9600, SERIAL_8N1, 16, 17);
}

void loop() {
  while(true) {
    int c = Serial2.read();
    if(c >= 0) Serial.write(c);
    yield();
  }
}
```
Serial2.read() will return -1 if nothing is available, so this basically writes every incoming data on Serial2 to Serial.
Good luck!

lbernstone, 2018-10-16:
Maddesh, a phototransistor is not typically using a [serial connection](https://en.wikipedia.org/wiki/Universal_asynchronous_receiver-transmitter). It is very unclear what device you are trying to connect to. You may also want to check the schematic of you dev board.  Those LoRa boards often use pin 16 for the reset pin on the OLED, which may interfere with serial comms.

Maddesh1337, 2018-10-18:
Thank you guys for your tips. I did some more tests to try to get to the point why is it not working.

To cut out the possible mistake in my code i tried 100% the same code at the ESP8266 and EPS32. I expected that this will work at both ... but it only worked at the esp8266. I tried all different HardwareSerials at the ESP32, no effect.
So I looked at the signals the ESPs are working with coming from the phototransistor. With the ESP8266 the signals are really nice at the oscilloscope, clear lows and highs. On the other hand, the ESP32 connected, the signals are really bad! There are no clear highs and lows. The peaks were on different levels and I think that would be the problem. Internal and/or external pull ups to RX made a improvement but I could not manage to get a signal as good as with the ESP8266. 
Is it possible, that the ESP32 has some higher pin-capacity or something like that, so he isnt able to supply the current/voltage? It looks like he is to slow for the incoming signal, so the voltage is not rising to clear highs... 
I hope I could describe my thoughts and findings good enough that you can understand the problem.

Jeroen88, 2018-10-18:
I do not know to much about hardware, but I am using both ESP8266 and ESP32 HardwareSerial's successfully to read out the same serial device with more or less the same sketch. Could you explain what you are trying to connect? Is the port on your electricity meter a digital port or a pulse output? In case of the latter you should not use HardwareSerial but pulse detection. If it is a digital port, the use of a phototransistor is a bit odd, unless you are trying to read the Infrared port. But then you need more than just a phototransistor: https://wiki.volkszaehler.org/hardware/controllers/ir-schreib-lesekopf-ttl-ausgang
One more thing: Gnd / 5v and 3v3 are at different positions on the ESP32 compared to ESP8266

lbernstone, 2018-10-18:
Maddesh, the ESP32 is not a power supply.  The regulator on board most development kits should be adequate to provide peak consumption requirements of the chip itself, which is about 500mA @ 3.3V.  If you are consuming more than that, particularly if you are running off a laptop USB port, you may not get enough power to your peripherals.

Maddesh1337, 2018-10-22:
Good morning!
thank you again for your suggestions. I just solved the problem.
@lbernstone : I am using the transistor to read the infrared-signal coming from the digital meter. So it just "opens/closes". So I looked again at the signals (oscilloscope) and tried again to get a clear high and low status. After some testing with pull up resistances I was able to get the signal the ESP32 need to get the high and low ... ~2.8k Oh, 3.3V. 
So at the end this is what I am doing: Reading the infrared-signal from the digital current meter via phototransistor. The 2 Pins of the PT are connected to GND and RX of the ESP32 (I chose GPIO35 as input). In addition I need to pullup the RX pin with the pullup-resistance 2.8k Ohms at 3.3V to get the signal strong enough for the ESP32 - I'm still not sure why :)

Now I just need to calculate the data to [kWh] and transmit it via LoRa. [Not done yet, but step by step!]

Thank you guys again for your suggestions!!

Thank you guys fr

closed 2019-08-15 by stale[bot]