espressif/arduino-esp32 #1972: ESP 32 is not connecting to wifi after using external power.

anandpanchal, 2018-10-16:
Hello Everyone,

One of my project I am sending post request using esp32. While checking my project powered by my laptop it is working perfectly. But my issue is that, esp32 is not connecting after using external power. 

I meant, I am giving power using vin pin and esp32 is not connecting to wifi. But it works fine while it is powered by laptop.

stickbreaker, 2018-10-16:
Simple answer:  Your external power supply is not working.

Chuck.

martinius96, 2018-10-18:
@anandpanchal I agree with @stickbreaker . Your power supply is "soft".
Type here it's specifications, and verify it with tester.

fpovoski, 2019-01-05:
Its not a "soft" PS, but a reference board design issue. The cause of this problem appears that some ESP32 boards the USB chip is not powered from the Vin (VDD5V) pin (19) but from the VDDUSB with a diode between.  This is unlike the ESP8266 boards which are the opposite and is correct to enable the USB chip to be powered at 5v whether plugged into the USB or powered solely by the 5v input pin.  It would then cause DTR and RTS to be in an indeterminate voltage affecting EN and IO0 throwing the ESP into a reboot when the higher current surge happens during WIFI communication pulls the rail down. IT happens on an older Makerfocus ESP32 but I will try another ESP32 board to see if occurs there.  Maybe its been fixed in newer boards, nevertheless, I use 8266s powered from the Vin pin all the time with no problems.

hervema, 2019-04-01:
Hello, not sure to get if there is a workaround to able to use an ESP32 board, WIFI and external power ? :-( I have the same problem and can't resolve it ! Thanks.

closed 2019-08-15 by stale[bot]

Bhaudipmehta, 2020-10-28:
Hi , I am facing the same issue and found nothing as a solution but you can workaround like connect you external supply to USB output and it works well for me.

fpovoski, 2020-10-28:
The designer obviously added the diode to prevent damage to the board if the Vin chip is powered at a higher voltage.  They forget about the voltage drop inherent in the diode. Been a while since I looked at this but find a board like the D1 mini ESP32 that doesn't have the diode in place. I've been using the ESP32-CAM boards for projects without the cam (although most are the one processor version of ESP32), since it is nearly the same small form factor as the 8266 D1 mini and still has alot of pins exposed.

lotrasoft, 2020-11-06:
I have the same issue with all the ESP32 boards that I have bought recently from Amazon and AliExpress.

DamyHao, 2021-01-19:
Same problem here, ESP32 from AZDelivery. It's very frustating and it seems there isn't a solution yet.

chegewara, 2021-01-19:
> Same problem here, ESP32 from AZDelivery. It's very frustating and it seems there isn't a solution yet.

What solution do you expect? Do you want someone from espressif support came to you and check voltage with V-meter or scope?
Sorry, it is a bit sarcastic, but it has nothing to do with software. It is related to your hardware and power source.
It may be voltage drop on long wires, or thin wires or anything else. Unless you or someone else can point it is software problem there is nothing support can do with it.
I am having dozens boards, some are powered with native USB (esp32S2), some with https://pl.aliexpress.com/item/4001205248833.html (M5 stack with 96 WS2812) and ive got issues only with very bad quality usb cable. In other hand one of my clients had issues with long wires connected between battery and custom PCB.

lotrasoft, 2021-01-19:
I found the issue with my board. I was wiring my circuit based on wrong pinout. I learned that there are lots of ESP32 DEV boards around, and each have different pinouts. My board's pinout was different from official DEV board; and as a result, I was supplying the voltage to the wrong pin, and that's why it was only working when it was connected to USB.

roblatour, 2021-02-28:
I am having this issue too esp devkit 1 c.  I am connecting the power to 5.5vin pin and a ground pin.  works when powered thru USB, but not externally.  Have connected it to a reliable power supply when connecting externally - 5v 2amps.  On board led lights up, but will not connect to wifi.

martinius96, 2021-02-28:
@roblatour I think problem is in your board. Devkit is defective board that have same capacitors at EN and BOOT signal. So startup characteristics for both signals are same and it will start ESP32 in download mode and must be restarted via EN to boot correctly to running mode.

roblatour, 2021-02-28:
thanks - is there any way around that?

martinius96, 2021-02-28:
yes, for instance pullup GPIO 0 (that is BOOT button signal) to 3V3 via 10K resistor.
Or change capacitor onboard :)

roblatour, 2021-02-28:
ok - thanks - I'll have to read up on that - I'm really a bit of a beginner - so I appreciate the feedback

fpovoski, 2021-02-28:
Rob, This was a common problem with the Dev board.  A while back I posted analysis of the circuit they use and have to check back to it to remember the work around.  Of the top of my head it had to do with the reverse current diode causing a voltage drop and not suppling enough current during that high draw wifi connect.  Let me see if i can find what I posted and get back to you.

roblatour, 2021-03-01:
fpovoski, thank you I would appreciate that.

roblatour, 2021-03-20:
@fpovoski just wondering if you were able to find anything, unfortunately I have a bunch of these models and they are of diminished use to me without a way to connect to wifi while on battery

roblatour, 2021-03-20:
@fpovoski I found this post:
https://esp32.com/viewtopic.php?t=7886

which says: 
Simply add a 470uF16V or larger rated electrolitic capacitor between the VCC and Ground

Does that make sense to you?

fpovoski, 2021-03-21:
Any bulk cap will keep the rail from dipping too low.  If it eliminates the problem for you great but I think if you can and would ever have usb hooked up the same time as the external 5v, jumper out the zener diode.  Effectively having the 5v pin without a diode powering the voltage regulator.

https://esp32.com/viewtopic.php?t=7886

fpovoski, 2021-03-21:
Sorry i haven’t tried jumpering out the zener, but if I get a chance today I will

raclobao, 2022-06-03:
While working on a College project, we stepped on the problem that many reported: the wifi connection isn't established if we supply the ESP32 pico kit with the 5V pin. The solution for us was using a smartphone USB adapter (Output 5V, 1.67A). So basically bypassing the 5V pin...

catchme1412, 2022-09-28:
When I changed the baud rate in the code from 115200 to 9600, it started working for me.

Anonymate054, 2023-09-02:
> When I changed the baud rate in the code from 115200 to 9600, it started working for me.

It worked for me too, but only for a few minutes. :c

arunsaispk12, 2024-01-01:
> Simple answer:  Your external power supply is not working.
> 
> Chuck.

I think you got that wrong

What mentioned over there is not connecting it means that the board is working but not the wifi

fpovoski, 2024-01-24:
> Following what is written here (cause i have the same issue), i made a simple test: remove the diode to make a bypass
> IMG_20240124_134558.jpg (view on web)<https://github.com/espressif/arduino-esp32/assets/27675549/6739512d-4942-4d7b-a52f-7bd4c78f96bd>
> IMG_20240124_135036.jpg (view on web)<https://github.com/espressif/arduino-esp32/assets/27675549/cd293c03-a78c-458b-8a7e-ca0711551373>
>
> It's simple fixing the issue
> Simple ans efficient (but yes less secure i know)

The exact fix I would recommend.
 Why is that diode there?  Simply to prevent current flow back into the USB power if you have the USB connected and +5V  applied to the EXT_5V pin.
Why is the ESP32 browning out?  The Schottky Barrier Diode (D1) when it is forward biased (power supplied from USB)  will drop the USB voltage by 0.3 - 0.5 volts. The designer  of  D1_Mini32 connected the LDO to the forward side of the DIODE which is subject to the 0.3 - 0.5v drop.  So what is feed to the Vin of the LDO is worse case 4.5v if +5V is actually being supplied to the USB_5V pin.   But he forgot to consider any voltage drop the USB_5V input might have due to cable losses, etc.  This coupled with the fact the ME6211 LDO on the board requires a Vin=Vout+1v to maintain regulation leaves a 0.2v margin. It also has an additional Dropout dependent on output current.  So if the USB_5V drops by just 0.2V or anything near 500ma of current is being drawn, this LDO will go out of regulation.
Is this the proper circuit to achieve this? No, its a cheap and dirty way to do it.  A MOSFET switch circuit would have been proper, or use of an LDO with less of a dropout voltage.  This is why the  cheap D1-Mini32 clones that used the "initial schematic"  cause the ESP32 to brownout when powered from USB.  Jumper out D1 or use the EXT_5v pin for power and all is well.

arunsaispk12, 2024-01-24:
Thanks for the explanation.
But I'll be more happy if the image is more clear even though I could
understand it

fpovoski, 2024-01-24:
Sorry,
Let me give you a link so you can download the schematic and datasheets.

[https://res-h3.public.cdn.office.net/assets/mail/file-icon/png/zip_16x16.png]D1-Mini32.zip<https://1drv.ms/u/s!AhTxS9MzL6TohuNwZMzqaThHGLMaQQ?e=r5C5Nj>

arunsaispk12, 2024-01-25:
Thanks, I'll check it out.

thermalogic, 2024-04-28:
Use the adjustable DC step-down module Mini360  to adjust the voltage to slightly lower than 5v to solve the problem that ESP32 is not connecting after using external power.

wintal, 2024-08-16:
I know this isn't the best place for this, but it might help someone.
For the NodeMCU-32 boards I have, the ground directly next to the 5v pin is... weird. When powered from USB it reads 2v to ground, so possibly something to do with the 3.3v pin, and also reads as +2v from the other ground. Powering using the 5v pin and this ground will not work. Powering using the 5v pin and the other ground works fine.

TL;DR - turns out the quality of dirt cheap boards is... questionable. But it worked eventuially.

arunsaispk12, 2024-08-16:
Thanks for the tip!