espressif/arduino-esp32 #1974: BLE scanner is an unexpected stop during scanning

jensen0915, 2018-10-17:
### Hardware:
Board:							ESP32 Dev Module (HiLetgo ESP-WROOM-32, 211-161007, FCC ID: 2AC7Z-ESPWROOM32)
IDE name:					    Arduino IDE
Flash Frequency:		        80Mhz
PSRAM enabled:                  no
Upload Speed:			        921600
Computer OS:                    Windows 10

### Description:
The BLE scanner is an unexpected stop during scanning (around few hours).
The loop() is freezing.

I dont' know how to slove this problem. I'm trying to do a BLE scanner to scan ibeacon data.

The debug message is stucked at "dumpGapEvent(): [status: 1]"

### Sketch:  

#if CONFIG_FREERTOS_UNICORE
#define ARDUINO_RUNNING_CORE 0
#else
#define ARDUINO_RUNNING_CORE 1
#endif

#include <BLEDevice.h>
#include <BLEUtils.h>
#include <BLEScan.h>
#include <BLEAdvertisedDevice.h>

#include "HardwareSerial.h"

#define SERIAL1_RXPIN 17
#define SERIAL1_TXPIN 18

#include <sstream>
#include <Wire.h>

const int SLAVE_ADDRESS = 1;
char incomingByte = 0;

#define MAX_SCAN_NUM 500
String scan_uuid[MAX_SCAN_NUM];
int scan_lifetime[MAX_SCAN_NUM];
unsigned long previousMillis = 0;
#define SCAN_MAX_LIFETIME 40
int scan_counter = 0;

std::stringstream ss;

#define SERIAL_PRINT

int scanTime = 30; //In seconds

boolean foundQRcode = false;
boolean serial1Enable = false;
String qrcode_uuid = "";
char qrcodeData[200];

class MyAdvertisedDeviceCallbacks: public BLEAdvertisedDeviceCallbacks {
    void onResult(BLEAdvertisedDevice advertisedDevice) {
      Serial.printf("%s\n", advertisedDevice.toString().c_str());
    }
};

void setup() {
  Serial.begin(115200);
}

void loop() {
  // put your main code here, to run repeatedly:

  BLEDevice::init("");
  BLEScan* pBLEScan = BLEDevice::getScan(); //create new scan
  pBLEScan->setAdvertisedDeviceCallbacks(new MyAdvertisedDeviceCallbacks());
  pBLEScan->setActiveScan(true); //active scan uses more power, but get results faster

  pBLEScan->setInterval(0x50);
  pBLEScan->setWindow(0x30);

  BLEScanResults foundDevices = pBLEScan->start(scanTime);

  delay(500);
}
```

### Debug Messages:
```
[D][BLEScan.cpp:196] start(): >> start(duration=30)
06:04:40.815 -> [D][BLEScan.cpp:221] start(): << start()
06:04:40.815 -> [D][BLEUtils.cpp:1091] dumpGapEvent(): Received a GAP event: ESP_GAP_BLE_SCAN_PARAM_SET_COMPLETE_EVT
06:04:40.815 -> [D][BLEUtils.cpp:1229] dumpGapEvent(): [status: 0]
06:04:40.862 -> [D][BLEUtils.cpp:1091] dumpGapEvent(): Received a GAP event: ESP_GAP_BLE_SCAN_START_COMPLETE_EVT
06:04:40.862 -> [D][BLEUtils.cpp:1305] dumpGapEvent(): [status: 1]
```

chegewara, 2018-10-17:
Its hard to say what is the reason, but if i would have guess then i would say it can be memory leak.
Maybe this will help you, at the bottom you can read that normal using pScan->start from loop still leads to memory leak, i advise to create new task for every start scan:
https://github.com/nkolban/esp32-snippets/issues/651

But you will have to replace BLEScan.cpp and BLEScan.h with those from gist:
https://gist.github.com/chegewara/6fc38e4127a7e18bb5b0bb4367aa33d1

Also there is still few more bugs in library i am trying to fix, some of them has been fixed now and are in repository of mine because needs to be tested.

PS this part of code should be in setup(), there is no need to call it in loop:
```
BLEDevice::init("");
BLEScan* pBLEScan = BLEDevice::getScan(); //create new scan
pBLEScan->setAdvertisedDeviceCallbacks(new MyAdvertisedDeviceCallbacks());
pBLEScan->setActiveScan(true); //active scan uses more power, but get results faster

pBLEScan->setInterval(0x50);
pBLEScan->setWindow(0x30);
```

jensen0915, 2018-10-17:
> Its hard to say what is the reason, but if i would have guess then i would say it can be memory leak.
> Maybe this will help you, at the bottom you can read that normal using pScan->start from loop still leads to memory leak, i advise to create new task for every start scan:
> [nkolban/esp32-snippets#651](https://github.com/nkolban/esp32-snippets/issues/651)
> 
> But you will have to replace BLEScan.cpp and BLEScan.h with those from gist:
> https://gist.github.com/chegewara/6fc38e4127a7e18bb5b0bb4367aa33d1
> 
> Also there is still few more bugs in library i am trying to fix, some of them has been fixed now and are in repository of mine because needs to be tested.
> 
> PS this part of code should be in setup(), there is no need to call it in loop:
> 
> ```
> BLEDevice::init("");
> BLEScan* pBLEScan = BLEDevice::getScan(); //create new scan
> pBLEScan->setAdvertisedDeviceCallbacks(new MyAdvertisedDeviceCallbacks());
> pBLEScan->setActiveScan(true); //active scan uses more power, but get results faster
> 
> pBLEScan->setInterval(0x50);
> pBLEScan->setWindow(0x30);
> ```

Thanks for your quick reply. 
I'm trying to replace new files (BLEScan.cpp and BLEScan.h) from the git.
But, I have faced a problem as shown in the following:

-----------------------------------------------------
C:\Users\jensen\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.0\libraries\BLE\src\BLEScan.cpp: In member function 'void BLEScan::handleGAPEvent(esp_gap_ble_cb_event_t, esp_ble_gap_cb_param_t*)':

C:\Users\jensen\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.0\libraries\BLE\src\BLEScan.cpp:118:24: error: 'class BLEAdvertisedDevice' has no member named 'setAddressType'

      advertisedDevice->setAddressType(param->scan_rst.ble_addr_type);
-----------------------------------------------------

Could you provide the URL so that I can download new files to solve the problem? 
(I have tried to find new BLEAdvertisedDevice.cpp and BLEAdvertisedDevice.h from git (espressif/arduino-esp32), but the files didn't contain setAddressType).

Many thanks!

chegewara, 2018-10-17:
You can delete line with setAddressType, it is only if you wish to connect with peer device and peer device got not public address. But if you still have this error then you can get new files from here:
https://github.com/chegewara/esp32-snippets-enchancements-test/tree/multi_connect_test

jensen0915, 2018-10-18:
Hi Chegeware, 

I have followed your instruction and tested the stability of the system.
So far, the system has been running at 24 hours and no problem so far. 

Thanks for your help.

closed 2018-10-18 by jensen0915