espressif/arduino-esp32 #1976: Propose change to setHostname

cv007, 2018-10-18:
### Hardware:
Board: ESP-WROOM-32D module

### Description:
A proposal to the way the WiFiSTAClass::setHostname() works so wifi sta reconnects automatically also sets the hostname instead of reverting to the built-in 'espressif' hostname.

I use WiFiMulti [ addAP() and run() ], and the only opportunity to set the hostname is to do a WiFi.enableSTA(true) before run()- which works only when the connection happens on the first call to run(). If it takes more than one attempt, the hostname is 'lost' and reverts to the built in name.

I'm unsure if this is the best/correct way to go about this, but what I have done-
this function-
https://github.com/espressif/arduino-esp32/blob/85032b226c7775bae510e954112823a2ae32181a/libraries/WiFi/src/WiFiSTA.cpp#L583

I have changed to-

        * Set the station interface Host name.
        * @param  hostname  pointer to const string or NULL to use cached hostname
        * @return true on   success
        */
        bool WiFiSTAClass::setHostname(const char * hostname)
        {
            static String _hostname;
            if(hostname) _hostname = hostname;
            if(WiFiGenericClass::getMode() == WIFI_MODE_NULL || _hostname.length() == 0){
                return false;
            }
            return tcpip_adapter_set_hostname(TCPIP_ADAPTER_IF_STA, _hostname.c_str()) == 0;
        }

which will locally cache the hostname when set. If a call is made with a NULL hostname the cached copy is used to set the hostname (if there is a cached copy). This gives an opportunity for some code in the connection process to set the hostname at the proper time.

I'm not sure where that proper time is, but this is what I did-
this function-
https://github.com/espressif/arduino-esp32/blob/85032b226c7775bae510e954112823a2ae32181a/libraries/WiFi/src/WiFiGeneric.cpp#L340

            } else if(event->event_id == SYSTEM_EVENT_STA_START) {
                WiFiSTAClass::_setStatus(WL_DISCONNECTED);
                WiFi.setHostname(NULL); //<<<< this line added <<<<
                setStatusBits(STA_STARTED_BIT);

which I think will get called whenever wifi sta 'starts'. I am assuming this would take care of the several esp_wifi_connects() in WiFiSTA.cpp or whatever else is used.

The result will be that you can set the hostname once (early) and it should persist through any disconnect/reconnect/scan or any other events that can cause the hostname to be 'lost'.

I'm not sure if there is a better way, or if this would be useful to anyone else. I have used this code only for a little while, but I am always getting the hostname set with these changes. I get my nvs stored hostname then make a call to setHostname() early in my code (my router is an older wrt54g with tomato firmware, and uses this hostname from the dchp request, where I can access my esp32 by hostname from a Linux pc without any other services like mdns).

I'm probably not capable of doing any pull requests, so if anyone thinks this is a good idea and wants to do that, go right ahead please.

cv007, 2018-10-28:
The setHostname function I previously wrote could also be written without bringing in a String, which is probably a better idea. 
```
bool WiFiSTAClass::setHostname(const char * hostname)
{
    static char _hostname[TCPIP_HOSTNAME_MAX_SIZE+1];
    if(hostname) strlcpy(_hostname, hostname, sizeof(_hostname));
    if(WiFiGenericClass::getMode() == WIFI_MODE_NULL) return false;
    return tcpip_adapter_set_hostname(TCPIP_ADAPTER_IF_STA, _hostname) == 0;
}
```
The better solution would probably be a re-write of the esp api function 'tcpcpip_adapter_set_hostname' as it already has a static buffer set aside for each interface which it fills in when a hostname is provided. 
https://github.com/espressif/esp-idf/blob/bd90f534fb1cdf7d4406767f9067398d51a12305/components/tcpip_adapter/tcpip_adapter_lwip.c#L1147
but since we are working on the other side of the fence, we deal with what they currently have.

MMartinezV, 2018-11-09:
I've resolved this in a different (but similar) way. Just creating a setHostname method in WiFiMulti.

1) In the headers declare a protected hostname variable and the setHostname method.
<pre><code>
// in WiFiMulti.h
class WiFiMulti
{
  public:
  ...
  void setHostname(const char* hostname);
  ...
  protected:
    static char * _hostname;
}
</code></pre>
2) In the cpp file:
<pre><code>
// Initialize _hostname just in case the setHostname() method is not called.
char * WiFiMulti::_hostname = "espressifWiFiMulti"; 

void WiFiMulti::setHostname(const char* hostname)
{
    _hostname=strdup(hostname);
}

// and then, just after the WiFi.begin(bestNetwork.ssid, ...) line:
WiFi.setHostname(_hostname);
</code></pre>

closed 2019-08-15 by stale[bot]