espressif/arduino-esp32 #1977: esp_log_level_set did not work as expected

moelski, 2018-10-18:
### Hardware:
Board:							ESP32 Dev Module
Core Installation/update date:			18.10.2018
IDE name:							Arduino IDE
Flash Frequency:					80Mhz
PSRAM enabled:                                          no
Upload Speed:						921600
Computer OS:                                            Windows 10

### Description:
I need some configurable Logging system and tried the espressif macros. Logging works great but setting the Log Level in Code with esp_log_level_set does not. 
The Loglevel in the Arduino IDE is set to Verbose. 

Why did I see 
[D][Debug_Test.ino:10] setup(): 1Debug
[V][Debug_Test.ino:11] setup(): 1Verbose
in the Log Output? 

Would be great to enable / disable logging parts during runtime. 

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp

#include "esp_log.h"

void setup() {
  // put your setup code here, to run once:
    Serial.begin(115200); 
    esp_log_level_set("*", ESP_LOG_INFO);  
    ESP_LOGE("TAG", "1Error");
    ESP_LOGW("TAG", "1Warning");
    ESP_LOGI("TAG", "1Info");
    ESP_LOGD("TAG", "1Debug");
    ESP_LOGV("TAG", "1Verbose");

    esp_log_level_set("*", ESP_LOG_VERBOSE);  
    ESP_LOGE("TAG", "Error");
    ESP_LOGW("TAG", "Warning");
    ESP_LOGI("TAG", "Info");
    ESP_LOGD("TAG", "Debug");
    ESP_LOGV("TAG", "Verbose");
}

void loop() {
  // put your main code here, to run repeatedly:

}
```

### Debug Messages:
```
[E][Debug_Test.ino:7] setup(): 1Error
[W][Debug_Test.ino:8] setup(): 1Warning
[I][Debug_Test.ino:9] setup(): 1Info
[D][Debug_Test.ino:10] setup(): 1Debug
[V][Debug_Test.ino:11] setup(): 1Verbose
[E][Debug_Test.ino:14] setup(): Error
[W][Debug_Test.ino:15] setup(): Warning
[I][Debug_Test.ino:16] setup(): Info
[D][Debug_Test.ino:17] setup(): Debug
[V][Debug_Test.ino:18] setup(): Verbose

```

voicevon, 2019-07-25:
to me,  I tried for  esp_log_level_set("*", ANY_POSSIBLE_LOG_LEVEL)
 It only output the ERROR_LEVEL information.  
I am using Vs_code + Platformio.  Esp32 on Arduino.   All is the latest version.

closed 2019-10-07 by stale[bot]

cakira, 2020-07-14:
To print the log, you need to define CORE_DEBUG_LEVEL as a build flag in the platformio.ini file.
For instance, mine is:

build_flags =
    -D CORE_DEBUG_LEVEL=4
    -D DEBUG

However, I'm here because searching for an answer regarding esp32-hal-log.h vs esp-log.h. Due to inclusion rules I can only use the former, which is simpler.  :-(

elotn, 2020-07-16:
I have the same problem with esp_log_level_set with arduino-esp32.  It doesn't work. With esp-idf it works fine.
I was trying PlatformIO and Visual micro with the same result. Board is OLIMEX esp32-evb but I think board is not the issue.
Arduino IDE 1.8.9
Arduino-esp32 - latest master
It's easy to replicate it in PlatformIO with framework arduino. Thanks for any suggestions.

CODE:
```
#include <Arduino.h>

void setup() {
  // put your setup code here, to run once:
    Serial.begin(115200); 
    esp_log_level_set("*", ESP_LOG_INFO);  
    ESP_LOGE("TAG", "1Error");
    ESP_LOGW("TAG", "1Warning");
    ESP_LOGI("TAG", "1Info");
    ESP_LOGD("TAG", "1Debug");
    ESP_LOGV("TAG", "1Verbose");

    esp_log_level_set("*", ESP_LOG_VERBOSE);  
    ESP_LOGE("TAG", "Error");
    ESP_LOGW("TAG", "Warning");
    ESP_LOGI("TAG", "Info");
    ESP_LOGD("TAG", "Debug");
    ESP_LOGV("TAG", "Verbose");
}

void loop() {
  // put your main code here, to run repeatedly:

}
```
OUTPUT:
```
[E][main.cpp:8] setup(): 1Error
[W][main.cpp:9] setup(): 1Warning
[I][main.cpp:10] setup(): 1Info
[D][main.cpp:11] setup(): 1Debug
[V][main.cpp:12] setup(): 1Verbose
[E][main.cpp:15] setup(): Error
[W][main.cpp:16] setup(): Warning
[I][main.cpp:17] setup(): Info
[D][main.cpp:18] setup(): Debug
[V][main.cpp:19] setup(): Verbose

platformio.ini
[env:esp32-evb]
platform = espressif32
board = esp32-evb
framework = arduino
build_flags = -DCORE_DEBUG_LEVEL=5
```

chegewara, 2020-07-16:
> To print the log, you need to define CORE_DEBUG_LEVEL as a build flag in the platformio.ini file.
> For instance, mine is:
> 
> build_flags =
> -D CORE_DEBUG_LEVEL=4
> -D DEBUG
> 
> However, I'm here because I'm searching for answers regarding esp32-hal-log.h vs esp-log.h. Due to inclusion rules, I can only use the former, which is simpler, and I would like to use esp-log.h :-(

there is no `esp-log.h` its `esp_log.h`

me-no-dev, 2021-01-13:
Arduino uses it's own log functions. ESP_LOG works for IDF only (and acts on already compiled libs). Use `log_x()` instead. And it is not runtime adjustable. ESP_LOGX are turned into log_x in esp32-hal-log.h

N0ury, 2023-04-01:
`And it is not runtime adjustable`

it's not understandable

jamesalawrence, 2025-01-30:
>  it is not runtime adjustable

I was able to make it runtime adjustable. I was able to make esp_log_level_set() work for me in Arduino code. As a bonus, I added some additional macros to support compiling conditionally on log level. The full solution is as follows:
I created my_log.h with the following content:
`#ifndef MYLOG_H
#define MYLOG_H

// Library/Arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-632e0c2a/esp32/include/log/include/
#include <esp_log.h>
// Library/Arduino15/packages/esp32/hardware/esp32/3.0.7/cores/esp32/esp32-hal-log.h
#include <esp32-hal-log.h>

#undef CONFIG_LOG_MAXIMUM_LEVEL
#define CONFIG_LOG_MAXIMUM_LEVEL CORE_DEBUG_LEVEL

#undef ESP_LOGE
#undef ESP_LOGW
#undef ESP_LOGI
#undef ESP_LOGD
#undef ESP_LOGV

#define ESP_LOGE( tag, format, ... ) ESP_LOG_LEVEL_LOCAL(ESP_LOG_ERROR,   tag, ARDUHAL_LOG_FORMAT(E, format) __VA_OPT__(,) __VA_ARGS__)
#define ESP_LOGW( tag, format, ... ) ESP_LOG_LEVEL_LOCAL(ESP_LOG_WARN,    tag, ARDUHAL_LOG_FORMAT(W, format) __VA_OPT__(,) __VA_ARGS__)
#define ESP_LOGI( tag, format, ... ) ESP_LOG_LEVEL_LOCAL(ESP_LOG_INFO,    tag, ARDUHAL_LOG_FORMAT(I, format) __VA_OPT__(,) __VA_ARGS__)
#define ESP_LOGD( tag, format, ... ) ESP_LOG_LEVEL_LOCAL(ESP_LOG_DEBUG,   tag, ARDUHAL_LOG_FORMAT(D, format) __VA_OPT__(,) __VA_ARGS__)
#define ESP_LOGV( tag, format, ... ) ESP_LOG_LEVEL_LOCAL(ESP_LOG_VERBOSE, tag, ARDUHAL_LOG_FORMAT(V, format) __VA_OPT__(,) __VA_ARGS__)

#undef ARDUHAL_LOG_FORMAT
#define ARDUHAL_LOG_FORMAT(letter, format)				\
  ARDUHAL_LOG_COLOR_##letter "[%s:%u] %s(): " format ARDUHAL_LOG_RESET_COLOR,  \
    pathToFileName(__FILE__), __LINE__, __FUNCTION__

#endif // MYLOG_H`
Then include the file in my code in place of including esp_log.h:
`#include <mylog.h>`
This replaces the macros for ESP_LOGx as defined in esp_log.h with macros that work with esp_log_level_set().