espressif/arduino-esp32 #1981: Connect with mDNS server

kamilbystryk, 2018-10-20:
I have Linux computer with Avahi, which works as MQTT broker and Web application. Mobile phone (iOS) and other computers (Windows, Linux) can connect with MQTT broker or web application using name.local address.
When I try to connect with MQTT broker or http using Arduino/ESP8266 I have no problem (both, using IP address or mDNS server name). When I try to use ESP32 I can connect via IP without any problem, but when I change address to DNS name (wg. name.local) I can't connect.
Also, demo resolving mDNS host can't find server:

`#include <ESPmDNS.h>               
#include <WiFi.h>               

void setup() {
  WiFi.begin("**", "**");
  Serial.begin(115200);
  while ( !WiFi.isConnected() )
  {
    delay(100);
  }
  Serial.println( "Wifi connected" );

  //initialize mDNS service
  esp_err_t err = mdns_init();
  if (err)
  {
    Serial.println( "MDNS Init failed: %d\n" );
    Serial.println(err);
  }
    
  //esolve_mdns_host( hostName );
}

void loop() {
resolve_mdns_host("name.local");
}

void resolve_mdns_host(const char * host_name)
{
    Serial.print(  "Query: ");
    Serial.println( host_name);

    struct ip4_addr addr;
    addr.addr = 0;

    esp_err_t err = mdns_query_a(host_name, 2000,  &addr);
    if(err)
    {
      if(err == ESP_ERR_NOT_FOUND)
      {
        Serial.println(  "Host was not found!");
        return;
      }
      Serial.println(  "Query Failed");
      return;
    }
}`

francescolavra, 2018-10-26:
The ".local" suffix should not be included in the host name passed to `mdns_query_a()`; try with `resolve_mdns_host("name")`.

g01d10x, 2019-05-09:
Just found something in the .cpp file which might help us all with these troubles .. could someone verify this change I'm proposing?

`
bool MDNSResponder::begin(const char* hostName){
  if(mdns_init()){
    log_e("Failed starting MDNS");
    return false;
  }
  WiFi.onEvent(_on_sys_event);
  _hostname = hostName;
  _hostname.toLowerCase();
  // Dave's input here, used to be:
  // if(mdns_hostname_set(hostName)) {
  // but that doesn't seem to make sense to me, so making it a 'not'
  if(!mdns_hostname_set(hostName)) {
    log_e("Failed setting MDNS hostname");
    return false;
  }
  return true;
}`

It seems to me that tif the if(mans_hostname_set(hoseName)) is successful, then the function should be allowed to return as a true, vs failing as a false as it was in the default.

lbernstone, 2019-05-09:
mdns_hostname_set returns ESP_OK (0) on success.  Otherwise it returns a specific (non-zero) error code which can be used for troubleshooting.  This is how almost all esp-idf functions are structured.  This will not change.

g01d10x, 2019-05-10:
OK thanks for that @lbernstone .. that clears up my puzzle.

closed 2019-08-14 by stale[bot]