espressif/arduino-esp32 #1983: Loop Through File Names?

daviddawe1982, 2018-10-21:
Hello,
After a lot of searching to no avail is there a way to list all file names?
Eg.
/dir1/file1.file
/file2.file
etc?
The file system is not big at all, I am trying to build a web interface and the ability to see file paths is the only thing missing.
Thanks

CelliesProjects, 2018-10-22:
I did it like this:
````
    else if ( request->hasArg( "files" ) )
    {
      File root = fs.open( "/" );
      if ( !root )
      {
        return request->send( 503, HTML_HEADER, "Storage not available." );
      }
      if ( !root.isDirectory() )
      {
        return request->send( 400, HTML_HEADER, "No root on Storage.");
      }
      File file = root.openNextFile();
      if ( !file )
      {
        return request->send( 404 );
      }
      response = request->beginResponseStream( HTML_HEADER );
      while ( file )
      {
        if ( !file.isDirectory() )
        {
          response->printf( "%s,%s\n", file.name(), humanReadableSize( file.size() ).c_str() );
        }
        file = root.openNextFile();
      }
      return request->send( response );
    }
````

arminsalcin, 2018-10-23:
Try this 

```C
void listDir(fs::FS &fs, const char * dirname, uint8_t levels){
    Serial.printf("Listing directory: %s\r\n", dirname);

    File root = fs.open(dirname);
    if(!root){
        Serial.println("- failed to open directory");
        return;
    }
    if(!root.isDirectory()){
        Serial.println(" - not a directory");
        return;
    }

    File file = root.openNextFile();
    while(file){
        if(file.isDirectory()){
            Serial.print("  DIR : ");
            Serial.println(file.name());
            if(levels){
                listDir(fs, file.name(), levels -1);
            }
        } else {
            Serial.print("  FILE: ");
            Serial.print(file.name());
            Serial.print("\tSIZE: ");
            Serial.println(file.size());
        }
        file = root.openNextFile();
    }
}

void setup(){
    Serial.begin(115200);
    if(!SPIFFS.begin(true)){
        Serial.println("SPIFFS Mount Failed");
        return;
    }
  listDir(SPIFFS, "/", 0);
}

void loop(){

}
```

closed 2019-08-15 by stale[bot]