espressif/arduino-esp32 #1985: Testing EEPROM Library  problem

gigirizzi, 2018-10-21:
hi, i try the Example Testing EEPROM Library but the code faul in 

if (!EEPROM.begin(EEPROM.length())) { 
  Serial.println("Failed to initialise EEPROM");
    Serial.println("Restarting...");
    delay(1000);
    ESP.restart();

arminsalcin, 2018-10-23:
@gigirizzi Try using preferences you have example , i found it better than eeprom .

gigirizzi, 2018-10-23:
Ok Tnx.

Inviato da iPhone

francescolavra, 2018-10-26:
If you check the return value of `EEPROM.length()`, you will see that it's 0, and `EEPROM.begin()` errors out if you call it with 0 as argument; instead, you should choose a non-zero EEPROM size value, and pass that value the `begin()` method. Note that if the size value is larger than the flash sector size (4 KB), it will be truncated to 4 KB.

closed 2019-08-15 by stale[bot]