espressif/arduino-esp32 #1987: esp32 loses serial connection when writing to servo via async webserver

sirluckyluke, 2018-10-22:
Hi folks,

i am working on a project where i am trying to control a servo via the async webserver and the esp32 servo-library. In one case the servo should be "speed controlled" which meas the servo position will be increased step by step with a variable waiting time in between. 

The problem occurs when the servo position is updated faster than 25 to 30 Hz, using the servo.write(position) function from the library. At this threshold the esp32 loses the serial connection to the pc and the serial link disappear in /dev/serial , but everything else on the esp32 is working properly as far as i could tell without the serial monitor. I have tried two ways of doing the position update, one with a while loop and a delay between every position increment and the other timer based (writing to the servo in the callback), but the result is still the same. 

Has someone experienced the same behavior and found a solution or a workaround? 

### Software:
ESP32 Servo-Library from John K. Bennett
AsyncWebserver from me-no-dev

### Hardware:
Board:							ESP32 Dev Module
Core Installation/update date:			latest
IDE name:					       Arduino IDE 1.8
Computer OS:                                        Ubuntu 16.04 LTS

Maximvdw, 2018-10-24:
Are you running the board from an external power source, or only the USB that you use for serial monitoring

sirluckyluke, 2018-10-25:
I used only the USB power, but switched to an external power source yesterday and it seems to be working now.

Thank you for pointing me in the right direction:)

closed 2018-10-25 by sirluckyluke