espressif/arduino-esp32 #1990: Help with OTA ESP32 - Urgent

lucasflbarbosa, 2018-10-22:
### Hardware:
Board: DOIT ESP32 DEVKIT V1
Core Installation/update date: 11/jul/2017
IDE name: Platform.io
Flash Frequency: 40Mhz
Upload Speed: 115200
Computer OS: Windows 10

### Description:
I need a lot of your help, I have an esp32 and I can do OTA with small files, but I have a program that uses 1.5MB and I can not do OTA for another program of the same size. I have already modified the partition table but even then at the time I am doing OTA it returns me an error: "Not enough space". I do not know what else to do!

Below are images in the order of:
My partition table
Build Status

Can you help me ?

### Debug Messages:
```
Not Enough Space

```

[image]
[image]
[image]

lucasflbarbosa, 2018-10-25:
Someone?

lucasflbarbosa, 2018-10-31:
Please, Someone???

lbernstone, 2018-10-31:
Try running espota.py manually with the debug option.

lucasflbarbosa, 2018-10-31:
I did not succeed! I have already changed esp32doit-devkit-v1.json and it continues the same thing. My json file is attached.

{
  "build": {
    "core": "esp32",
    "extra_flags": "-DARDUINO_ESP32_DEV",
    "f_cpu": "240000000L",
    "f_flash": "40000000L",
    "flash_mode": "dio",
    "ldscript": "esp32_out.ld",
    "mcu": "esp32",
    "variant": "doitESP32devkitV1",
	"partitions": "partitions"
  },
  "connectivity": [
    "wifi",
    "bluetooth",
    "ethernet",
    "can"
  ],
  "frameworks":  [
    "arduino",
    "espidf"
  ],
  "name": "DOIT ESP32 DEVKIT V1",
  "upload": {
    "flash_size": "4MB",
    "maximum_ram_size": 950000,
	"maximum_size": 4194303,
    "require_upload_port": true,
    "speed": 115200,
    "wait_for_upload_port": true
  },
  "url": "http://www.doit.am/",
  "vendor": "DOIT"
}

chegewara, 2018-10-31:
@lucasflbarbosa 
I would say that all DevKitC should have 4MB flash, but from your screenshot we can see that yours have 2MB or at least logs shows that board is recognized with 1.94MB.

Check `SYSTEM` line.

lucasflbarbosa, 2018-11-01:
Sorry, had modified the .json to 4MB, here below follows again the prints of the files. Still failing! But when I do OTA from a smaller file it works.

esp32doit-devkit-v1.json

{
  "build": {
    "core": "esp32",
    "extra_flags": "-DARDUINO_ESP32_DEV",
    "f_cpu": "240000000L",
    "f_flash": "40000000L",
    "flash_mode": "dio",
    "ldscript": "esp32_out.ld",
    "mcu": "esp32",
    "variant": "doitESP32devkitV1",
	"partitions": "partitions"
  },
  "connectivity": [
    "wifi",
    "bluetooth",
    "ethernet",
    "can"
  ],
  "frameworks":  [
    "arduino",
    "espidf"
  ],
  "name": "DOIT ESP32 DEVKIT V1",
  "upload": {
    "flash_size": "4MB",
    "maximum_ram_size": 520000,
	"maximum_size": 4194304,
    "require_upload_port": true,
    "speed": 921600,
    "wait_for_upload_port": true
  },
  "url": "http://www.doit.am/",
  "vendor": "DOIT"
}

Partition Table

# Name,   Type, SubType, Offset,  Size, Flags
nvs,      data, nvs,     0x9000,  0x5000,
otadata,  data, ota,     0xe000,  0x2000,
app0,     app,  ota_0,   0x10000, 0x1E0000,
app1,     app,  ota_1,   0x1F0000,0x1E0000,
eeprom,   data, 0x99,    0x3F0000,0x1000,
spiffs,   data, spiffs,  0x3F1000,0xF000,

chegewara, 2018-11-01:
Maybe connection timeout or something else with wifi connection.

PS does spiffs partition is not too small?

lucasflbarbosa, 2018-11-01:
I do not use SPIFFS, and while I'm doing OTA I check on the serial output that it spits the following message: "Not enough space" and gives it the fail.

chegewara, 2018-11-01:
Yes, but you dont have where you have "Not enough space", or you have it? Maybe some logs would help. 10-20 lines before crash.

lucasflbarbosa, 2018-11-01:
I have space, at least I believe I have! Since my first firmware upload is by serial cable, where I place the partition below! In this case I have almost 2MB for APP0 and 2MB for APP1! My firmware is 1.5MB, and the firmware I'm going to put through OTA is also 1.5MB, so according to the partition table, it should work! Or not?

lucasflbarbosa, 2018-11-01:
I created these files inside my project folder in ATOM + PLATFORMIO, so I believe this is partitioning my memory from ESP32. Or is something wrong?

[Files.zip](https://github.com/espressif/arduino-esp32/files/2539518/Files.zip)

chegewara, 2018-11-01:
I cant disagree with that, thats why it is so hard to resolve this issue and thats why i am asking for logs.

lucasflbarbosa, 2018-11-01:
what log files are these?

chegewara, 2018-11-01:
> I'm doing OTA I check on the serial output that it spits the following message: "Not enough space" and gives it the fail.

lucasflbarbosa, 2018-11-06:
Friends, thank you! I found out what was the error, I was passing in the Update.begin () call a partition value below the one I have, even though I having a large partition I passed the wrong value in the call of that Update.begin () function.

For example:

I had in my partition table the equivalent to app_1 the value of 0x1F0000, but at the time of the function call I was doing Update.begin (0x140000) instead of Update.begin (0x1F0000).

Thank you all for your help!

closed 2018-11-06 by lucasflbarbosa