espressif/arduino-esp32 #1995: Wifi interrupt function for receiving data

josefinHeilig, 2018-10-23:
Hi everyone,

I connected several ESP32s as clients each with a static IP with a local Wifi server (my computer) using UDP.

I am wondering if there is a way to use a kind of wifi on event function which only interrupts the main task of the ESP, when the needed information has filled up the read buffer.

The udp.read() function is slowing down the main loop in irregular intervals which is problematic.

How should I approach this? 

So far I only found examples for WifiEvent when the ESP is used as a server.

Should I maybe write my own function inside of the Wifi library so I can already be specific on the data that I am sending to the clients?
If yes how?

Thanks in advance and I hope I am clear enough about my problem. If not I am there for clarification.

Cheers, jh

bertmelis, 2018-10-23:
https://github.com/espressif/arduino-esp32/tree/master/libraries/AsyncUDP/src

Of is TCP needed?

josefinHeilig, 2018-10-23:
Hi, we are using UDP. At the moment we have 10 omniwheel robots steered over wifi. They were just moving a bit jerky because the motors were not moving while the udp buffer was filling.
So I will have a look at the AsyncUDP, it sounds promising.
Thanks :)