espressif/arduino-esp32 #1996: New freertos function

arminsalcin, 2018-10-23:
### Hardware:
Board:							node32
Core Installation/update date:			22/oct/2018
IDE name:							Arduino IDE
Flash Frequency:					..
PSRAM enabled:                                          ..
Upload Speed:						..
Computer OS:                                            Mac OSX

### Description:

Hello , i want to try using freertos function called  "xTaskAbortDelay " but as i see in freertos files in core this function doesn't even exist , how can i add function ? , i also doesn't see tasks.c file which i think there would be code execution for functions . Also how functions are added to core ?

lbernstone, 2018-10-23:
freertos is a part of [esp-idf](https://github.com/espressif/esp-idf). The functions in idf are available in arduino through compiled libraries. Many freertos functions are not SMP compatible, so are not included in the esp32.

arminsalcin, 2018-10-23:
Hmm so what is solution ?

lbernstone, 2018-10-23:
[To use as a component of ESP-IDF](https://github.com/espressif/arduino-esp32/blob/master/docs/esp-idf_component.md). The function is in tasks.c, just blocked by a define.  You can use it, but YMMV.

arminsalcin, 2018-10-23:
So i can't upload programs via arduino ide anymore ?

lbernstone, 2018-10-23:
I am saying it is not included Arduino because it will not function properly in all situations.  If you choose to use it, you will need to compile everything from scratch.

arminsalcin, 2018-10-24:
@chegewara  Thank you so much for explaining i will find another way to get task from blocked state , and wait for esp-idf to get updated

chegewara, 2018-10-24:
@arminsalcin There is few option, i would recommend event groups from freeRTOS.
`xEventGroupWaitBits` and `xEventGroupSetBits`

arminsalcin, 2018-10-24:
@chegewara I'm now using `xTaskNotifyWait`  to delay and `xTaskNotify` to get out of blocked state  , i writed code for someone needs .

```C

TaskHandle_t suspendTask , loopTask;

int loadDelay = 5000;

void setup() {
  Serial.begin(9600);

  xTaskCreatePinnedToCore(suspendTask1, "SuspendTask", 7000, NULL, 1, &suspendTask, 0);
  delay(500);
  xTaskCreatePinnedToCore(loopTask1, "Looptask1", 7000, NULL, 1, &loopTask, 0);
  delay(500);

}

void loop() {
  // put your main code here, to run repeatedly:

}

void suspendTask1(void * parameter) {
    TickType_t xLastWakeTime = xTaskGetTickCount();
  for (;;) {
   // this task will unblock looptask after 5 seconds 
    vTaskDelayUntil( &xLastWakeTime, 5000/ portTICK_PERIOD_MS );
     xTaskNotify(loopTask, 0, eNoAction);
  }
}

void loopTask1(void * parameter) {
uint32_t ulNotifiedValue;
  for (;;) {
//this task will wait for one minute if no notifications it will proceed to code you write after NotifyWait
  xTaskNotifyWait( 0x00, ULONG_MAX, &ulNotifiedValue, loadDelay / portTICK_PERIOD_MS );

//some code to do 
  }

}

```

chegewara, 2018-10-24:
I didnt work with TaskNotify yet, but yes, its good choice i think. If its solved you can close issue.

closed 2018-10-24 by arminsalcin

cyberman54, 2019-04-08:
I had the same problem (`xTaskAbortDelay` not in current ESP-IDF) and solved it with the suggested solution, using xTaskNotifyWait instead of VTaskDelay. Thanks for this clever post.