espressif/arduino-esp32 #1997: Bluetooth classic coexistance problem wifi

Maximvdw, 2018-10-23:
I am making a Bluetooth scanner (classic Bluetooth) that logs back to a server.

The program works fine using the modified IDF discovery for classic Bluetooth. But as soon as I set Wifi to Sta mode, the eir packet gets corrupted.

I still receive results from the scanning, but the raw EIR data no longer contains a logical format.

Has anyone ever had issues like this? The BLE API seems to do fine in getting the advertisement data

chegewara, 2018-10-24:
I would not recommend to use `WiFi.mode(WIFI_OFF);` with bluetooth, because it can cause issues with bt. It is possible, at least ive seen few some esp-idf issue post that that turning off wifi causing turning off radio. It should be fixed right now in esp-idf master but most likely still not fixed in arduino-esp32.

About bt classic i cant help you, i can only advice to open issue in esp-idf repo.

Maximvdw, 2018-10-25:
I managed to get the WIFI + BT Classic discovery working in the latest IDF

So I guess we should wait for this repo to be updated

closed 2019-08-15 by stale[bot]