espressif/arduino-esp32 #1998: Receive error occurs at I2C port

Schrscat, 2018-10-24:
We are using the ESP32 to receive the information from the MAX 30100. The returned value on the oscilloscope is right, which means that the value returned by the sensor is right. The value is 0x11. But received value from the esp32, showing on the PC, is not 0x11 but 0. Following is the shortest code to generate this error. This error could be a generous one. We have tried all the I2C ports, but this problem remains the same. 

```c
uint16_t readRegister(uint8_t address)
{
    Wire.beginTransmission(MAX30100_I2C_ADDRESS);
    Wire.write(address);
    Wire.endTransmission(false);
    Serial.print("byte count: ");
    Serial.println(Wire.requestFrom(MAX30100_I2C_ADDRESS, 1));

    return Wire.read();
}

void setup(){
    Serial.begin(115200);

    delay(1000);

    Wire.begin(21,22);
    Wire.setClock(I2C_BUS_SPEED);

    Serial.println(readRegister(0xff));
}

void loop(){}
```

The picture on the oscilloscope is 

[image]

lbernstone, 2018-10-24:
Please indicate which version of code you are using.  There were some issues with ReSTART in the release version.  This is fixed in the repository.

stickbreaker, 2018-10-24:
@schrscat same issue as #1962.

Chuck.

Schrscat, 2018-10-24:
> Please indicate which version of code you are using. There were some issues with ReSTART in the release version. This is fixed in the repository.

We are using v3.2-dev-39-gaaf12390.

Schrscat, 2018-10-24:
> @Schrscat same issue as #1962.
> 
> Chuck.

OK,we will try this. I would reply to you asap

closed 2019-08-15 by stale[bot]