espressif/arduino-esp32 #2000: Is it possible to do SPI transfers in a ISR safe way?

AlessandroAU, 2018-10-26:
I have one of the hardware timers calling an function to send some data over SPI every 20ms or so.
The ESP32 is also running a web-server serving files from SPIFFs. Currently this causes crashes because of issues with cache.

Is there a way to allow SPI transfers to be done in an ISR safe way?
[image]

stickbreaker, 2018-10-26:
@AlessandroAU How much latency can you accept? You could set up a task to handle the SPI operations.  This task would hang on a semaphore waiting to be triggered.  Your timer ISR would 'GIVE' the semaphore to release the waiting task. [Binary Semaphores](https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/system/freertos.html#c.xSemaphoreCreateBinary)

Chuck.

stickbreaker, 2018-10-26:
@AlessandroAU there are two cores in the esp32 one Wifi runs on core 0, when your app calls WiFi functions messages are passed between the cores, Arduino operations are linked to core 1. So, during a WiFi call the code on core 1 is hung waiting for a response from core 0.  If you create another task the FreeRTOS OS dispatcher can switch to it if this other task is ready(semaphore is 'given').  You can also code your app to `yield()` when it has nothing to do. (delay(n) where n>=2). 

You might actually use the [`xTaskNotify()`](https://www.freertos.org/RTOS_Task_Notification_As_Binary_Semaphore.html) instead of a Semaphore, it looks to be faster.

Chuck.

AlessandroAU, 2018-10-27:
@stickbreaker, Thank you very much for your help. It appears I need to do some reading the architecture of the ESP32.

Do you think it would be possible to assign both the wifi and websever functions to core 0 and leave core 1 free for timing critical code? 

Regards,
Alessandro

AlessandroAU, 2018-10-29:
Ok, I did some research, I think I understand the FreeRTOS methods now. Thanks for the help. Another question, does the arduino code itself (provided it does not define any tasks) count as a single task on its own? Therefore, If I then define another task the task scheduler can choose to run it but it is separate task entirely from the main arduino task?

chegewara, 2018-10-29:
Arduino setup and loop are part of one single freeRTOS task which is running on core 1. loop is wrapped in while(1). Now you can create as many tasks as you need just like you would do in freeRTOS app. You can event call `vTaskDelete(NULL)` from within setup or loop to destroy arduino default task.

AlessandroAU, 2018-10-29:
Thanks chegewara, does a pin or timer interrupt call always supersede whatever task is running?

AlessandroAU, 2018-10-29:
Ok, I cobbled together some basic code that seems to work, calling an ISR from a timer to give the Semaphore and waking up the high priority task., it seems to be stable up to several kHz with wifi and a webserver running. But all it does is toggle a pin, I merge it back into my main block next.

Any comments on the implementation?

```
static SemaphoreHandle_t timer_sem;
bool pinstate = false;

hw_timer_t * timer = NULL;
portMUX_TYPE timerMux = portMUX_INITIALIZER_UNLOCKED;

void IRAM_ATTR timer_isr_handler()
{
  static BaseType_t xHigherPriorityTaskWoken = pdFALSE;
  //Serial.println("TIMERISR");

  xSemaphoreGiveFromISR(timer_sem, &xHigherPriorityTaskWoken);
  if ( xHigherPriorityTaskWoken) {
    portYIELD_FROM_ISR(); // this wakes up sample_timer_task immediately
  }
}

void IRAM_ATTR sample_timer_task(void *param)
{
  timer_sem = xSemaphoreCreateBinary();

  timer = timerBegin(0, 40, true);
  timerAttachInterrupt(timer, &timer_isr_handler, true);
  timerAlarmWrite(timer, 100, true);
  timerAlarmEnable(timer);

  while (1) {
    xSemaphoreTake(timer_sem, portMAX_DELAY);

    digitalWrite(15, pinstate);
    pinstate = !pinstate;
    //Serial.println("TASK1");
  }
}

void setup() {
  Serial.begin(115200);

  xTaskCreate(
    sample_timer_task,          /* Task function. */
    "sample_timer_task",        /* String with name of task. */
    10000,            /* Stack size in words. */
    NULL,             /* Parameter passed as input of the task */
    10,                /* Priority of the task. */
    NULL);            /* Task handle. */

  pinMode(15, OUTPUT);
}

void loop() {
  delay(100);
}
```

everslick, 2018-11-04:
10000 words stack size seems to be on the save side, you might optimize it after you verified that your implementation works. other then that, it looks good to me.

AlessandroAU, 2018-11-09:
This might be a little OT, but how can I make this part of a library? 
[image]

closed 2018-11-16 by AlessandroAU

chegewara, 2018-11-16:
You cant pass class member function this way as a pointer. It needs to be C function or C++ static class member function.