espressif/arduino-esp32 #2001: Multithreading timing issue

josefinHeilig, 2018-10-26:
Hi everyone,

I want to use Multi threading on both cores of the ESP32
and 2 examples I  found work fine for me:
1: https://www.hackster.io/rayburne/esp32-in-love-with-both-cores-8dd948
2: https://github.com/copercini/esp32-iot-examples/blob/master/multiloop/multiloop.ino

But none of them gives a hint of how to execute the threads at a specific frequency. 

I want to use one thread for the updating of 3 synchronized bldc motors.
I drive the motors by changing the increment size of each motor to go through a PWM array, while the frequency is (should be) constant. Like in this example: http://www.berryjam.eu/2015/04/driving-bldc-gimbals-at-super-slow-speeds-with-arduino/
This means that I have 9 channels running 3 motors. 

I am not sure if it makes sense to have one thread per motor or one thread for all of them.
The other threads I want to use for sensor updates, battery check, etc. For Wifi I am using AsyncUDP.

In the second example for Multi threading a delay is used inside of the tasks, but in my understanding that is not making sure that the task is executed again after exact that time?

How can I time my Threads? 

I hope my problem becomes clear.

Cheers jh

josefinHeilig, 2018-10-26:
I just found vTaskDelayUntil() as opposed to  vTaskDelay() and delay() at 
https://www.freertos.org/vtaskdelayuntil.html

It might solve my issue, but other ideas are welcome.

atanisoft, 2018-10-26:
you may benefit from the Ticker library: https://github.com/espressif/arduino-esp32/tree/master/libraries/Ticker

lbernstone, 2018-10-26:
I don't understand your exact output requirements, but the [RMT peripheral](https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/peripherals/rmt.html) may be easier to use if you are trying to send a particular binary (pseudo-pwm) pattern at a cyclical frequency.  There is an arduino class in development, but the IDF code is not too difficult to work with.

closed 2019-08-15 by stale[bot]