espressif/arduino-esp32 #2003: format should temporarily disable the watchdog()

dirkx, 2018-10-27:
Board:							ESP32
Core Installation/update date:		2018-10-22
IDE name:						Arduino ID?
Flash Frequency:			40Mhz
Upload Speed:						115200?
Computer OS:                                            *BSD

During a full format of a large SPIFFs in setup() - one regularly gets the error

   Task watchdog got triggered. The following tasks did not reset the watchdog in time:
    - IDLE (CPU 0)
   Tasks currently running:
   CPU 0: ipc0
   CPU 1: IDLE

```cpp

//Change the code below by your sketch
#include <Arduino.h>

void setup() {
  SPIFF.format();
}

void loop() {
}
```

### Debug Messages:
```
   Task watchdog got triggered. The following tasks did not reset the watchdog in time:
    - IDLE (CPU 0)
   Tasks currently running:
   CPU 0: ipc0
   CPU 1: IDLE
```

lbernstone, 2018-10-27:
This is not good behavior, but you really should not use SPIFFS for large partitions.  It is not a particularly efficient filesystem.  Formatting FFAT on a 8MB SPI part does not give me a watchdog error.

everslick, 2018-11-04:
but SPIFFS has wear-leveling, which might be important. and the watchdog message gets triggere on much smaller SPIFFS partitions as well. so IMHO the OP has a point here.

lbernstone, 2018-11-04:
I am unable to reproduce.  Can someone that has the problem test this code and bump the wdt time up until it doesn't trigger?
```
#include <SPIFFS.h>
#include <esp_task_wdt.h>

void setup() {
  Serial.begin(115200);
  SPIFFS.begin(true);
}

void loop() {
  esp_task_wdt_init(2,false);
  uint32_t t = millis();
  SPIFFS.format();
  Serial.println(millis()-t);
}
```
@everslick : On the contrary, ffat uses wear levelling, SPIFFS does not.  This is also why it will be much more difficult to write a FS plugin for ffat than the one for SPIFFS.

everslick, 2018-11-04:
https://github.com/pellepl/spiffs

Spiffs is a file system intended for SPI NOR flash devices on embedded targets.

Spiffs is designed with following characteristics in mind:

* Small (embedded) targets, sparse RAM without heap
* Only big areas of data (blocks) can be erased
* An erase will reset all bits in block to ones
* Writing pulls one to zeroes
* Zeroes can only be pulled to ones by erase
* Wear leveling

anyway, i did not know that there was a FAT implementation with wear leveling, so thanks for the hint.

lbernstone, 2018-11-04:
Ah, the wear levelling in SPIFFS is contained in the library. The fat library uses a separate component.  Good to know.

everslick, 2018-11-04:
@lbernstone do you know by any chance if FFAT needs more resources then SPIFFS during runtime (heap) and on flash (code)? I'm not too happy about SPIFFS stability, but I'm tight on app partition size (and RAM).

lbernstone, 2018-11-04:
Fat is an indexed file system.  It may get slower over time as it needs to break files up into free space, but it is tracking the free space, and has significantly more testing on the failure conditions :smile:.  In my tests, FFat generally outperforms SPIFFS.  I would guess it takes more RAM, but I don't know exactly how much.  Pretty easy to test, as I copied the example pretty directly from SPIFFS.  Just throw some getFreeHeap statements into each.

everslick, 2018-11-04:
@lbernstone thanks for the comment, will do some benchmarks :-)

lbernstone, 2018-11-04:
In a minimal config (maxOpenFiles=1), it takes about 7K more heap for FFat, with about 4K additional per maxOpenFile.  Flash size is less than 5K additional.

closed 2019-08-15 by stale[bot]