espressif/arduino-esp32 #2004: HardwareSerial incorrect timing with non standard baudrate

aster94, 2018-10-28:
### Hardware:
Board:							ESP32 Lolin 32 mini
Core Installation/update date:			last fropm package manager (1.0.0?)
IDE name:						vscode
Flash Frequency:					tried both 40 adn 80MHz
PSRAM enabled:                                     don't know
Upload Speed:					460800
Computer OS:                                         xubuntu 18.04

### Description:
Using a baudrate different form the standard (10400) brings to incorrect timing of the serial communication
I made a few samples with a salea logic analizer (clone) with the above settings:
[image]

Correct behaviour from an arduino mega:
[image]
[mega 10400 o.txt](https://github.com/espressif/arduino-esp32/files/2522911/mega.10400.o.txt)

while esp plays a few jokes
[image]
[esp 10400 o.txt](https://github.com/espressif/arduino-esp32/files/2522912/esp.10400.o.txt)

same at 80mhz
[image]
[esp 80 10400 o.txt](https://github.com/espressif/arduino-esp32/files/2522913/esp.80.10400.o.txt)

same happens if i remove the odd parity bit
[image]
[esp 10400 n.txt](https://github.com/espressif/arduino-esp32/files/2522916/esp.10400.n.txt)

while at a standard baud rate (115200) all is ok
[image]
[esp 115200 n.txt](https://github.com/espressif/arduino-esp32/files/2522914/esp.115200.n.txt)

### Sketch:
for esp
```
HardwareSerial ser(2);

void setup()
{
    ser.begin(10400,SERIAL_8O1);
}

void loop()
{
    for (uint8_t n = 1; n < 10; n++)
    {
        ser.write(n);
        delay(10);
    }
    delay(100);
}
```

for arduino mega
```
void setup()
{
    Serial2.begin(10400,SERIAL_8O1);
}

void loop()
{
    for (uint8_t n = 1; n < 10; n++)
    {
        Serial2.write(n);
        delay(10);
    }
    delay(100);
}
```

aster94, 2019-08-01:
i suggest not to close this

aster94, 2019-09-30:
.

lbernstone, 2019-09-30:
Please modify and test the following code adapted directly from esp-idf [uart example](https://github.com/espressif/esp-idf/tree/master/examples/peripherals/uart/uart_echo) to see if the behavior is coming from the hardware or software.
https://docs.espressif.com/projects/esp-idf/en/v3.2/api-reference/peripherals/uart.html
```
/* UART Echo Example

   This example code is in the Public Domain (or CC0 licensed, at your option.)

   Unless required by applicable law or agreed to in writing, this
   software is distributed on an "AS IS" BASIS, WITHOUT WARRANTIES OR
   CONDITIONS OF ANY KIND, either express or implied.
*/
#include <stdio.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "driver/uart.h"

/**
 * This is an example which echos any data it receives on UART1 back to the sender,
 * with hardware flow control turned off. It does not use UART driver event queue.
 *
 * - Port: UART1
 * - Receive (Rx) buffer: on
 * - Transmit (Tx) buffer: off
 * - Flow control: off
 * - Event queue: off
 * - Pin assignment: see defines below
 */

#define ECHO_TEST_TXD  (GPIO_NUM_4)
#define ECHO_TEST_RXD  (GPIO_NUM_5)
#define ECHO_TEST_RTS  (UART_PIN_NO_CHANGE)
#define ECHO_TEST_CTS  (UART_PIN_NO_CHANGE)

#define BUF_SIZE (1024)
// Configure a temporary buffer for the incoming data
uint8_t *data = (uint8_t *) malloc(BUF_SIZE);  

void setup() {
    /* Configure parameters of an UART driver,
     * communication pins and install the driver */
    uart_config_t uart_config = {
        .baud_rate = 115200,
        .data_bits = UART_DATA_8_BITS,
        .parity    = UART_PARITY_DISABLE,
        .stop_bits = UART_STOP_BITS_1,
        .flow_ctrl = UART_HW_FLOWCTRL_DISABLE
    };
    uart_param_config(UART_NUM_1, &uart_config);
    uart_set_pin(UART_NUM_1, ECHO_TEST_TXD, ECHO_TEST_RXD, ECHO_TEST_RTS, ECHO_TEST_CTS);
    uart_driver_install(UART_NUM_1, BUF_SIZE * 2, 0, 0, NULL, 0);
}

void loop() {
        // Read data from the UART
        int len = uart_read_bytes(UART_NUM_1, data, BUF_SIZE, 20 / portTICK_RATE_MS);
        // Write data back to the UART
        uart_write_bytes(UART_NUM_1, (const char *) data, len);
}
```

aster94, 2019-10-05:
@lbernstone thanks for the help, i tried:

```
HardwareSerial ser(2);

void setup()
{
    ser.begin(10400,SERIAL_8O1);
}

void loop()
{
    for (uint8_t n = 1; n < 10; n++)
    {
        ser.write(n);
        delay(10);
    }
    delay(100);
}
```

and

```
#include <Arduino.h>

#include <stdio.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "driver/uart.h"

#define ECHO_TEST_TXD (GPIO_NUM_17)
#define ECHO_TEST_RXD (GPIO_NUM_16)
#define ECHO_TEST_RTS (UART_PIN_NO_CHANGE)
#define ECHO_TEST_CTS (UART_PIN_NO_CHANGE)

#define BUF_SIZE 1024

void setup()
{
  uart_config_t uart_config = {
      .baud_rate = 10400,
      .data_bits = UART_DATA_8_BITS,
      .parity = UART_PARITY_DISABLE,
      .stop_bits = UART_STOP_BITS_1,
      .flow_ctrl = UART_HW_FLOWCTRL_DISABLE};
  uart_param_config(UART_NUM_2, &uart_config);
  uart_set_pin(UART_NUM_2, ECHO_TEST_TXD, ECHO_TEST_RXD, ECHO_TEST_RTS, ECHO_TEST_CTS);
  uart_driver_install(UART_NUM_2, BUF_SIZE * 2, 0, 0, NULL, 0);
}

void loop()
{
  const char *data = "test";
  uart_write_bytes(UART_NUM_2, data, 5);
  delay(2);
}
```

This is the output: 
[test.zip](https://github.com/espressif/arduino-esp32/files/3693092/test.zip)

To me looks like an hardware problem, any thoughts? may I have modified in a wrong way your example sketch?

lbernstone, 2019-10-05:
I can't see anything in your Matrix data, Cypher.
If you are seeing the same result from the uart code as you get from HardwareSerial, then I suggest you open up an issue at the [ESP-IDF forum](https://github.com/espressif/esp-idf/issues)

aster94, 2019-12-04:
Don't close it

aster94, 2020-02-03:
Still waiting for this to be fixed on the esp-idf

koobest, 2020-03-17:
Hi @aster94 @stale
Has the issue been solved?   I can not  reproduce this issue based on ESP-IDF
thanks !!

aster94, 2020-03-17:
Hi @koobest first of all thanks to have a look to the issue

I just finished to run a few tests, this is the version of the core in my machine:
```
PLATFORM: Espressif 32 1.11.2 > WEMOS LOLIN32 (Platformio version)
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 3.10004.200129 (1.0.4)
 - tool-esptoolpy 1.20600.0 (2.6.0)
 - toolchain-xtensa32 2.50200.80 (5.2.0)
```
Which, if i am not wrong uses 4 IDF 3.2

I tested two sketches
Arduino version
```
#include <Arduino.h>

HardwareSerial ser(2); // TX: 17

void setup()
{
    ser.begin(9600,SERIAL_8O1);
}

void loop()
{
    for (uint8_t n = 1; n < 10; n++)
    {
        ser.write(n);
        delay(20);
    }
    delay(100);
}
```

IDF Version:
```
#include <Arduino.h>

#include <stdio.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "driver/uart.h"

#define ECHO_TEST_TXD (GPIO_NUM_17)
#define ECHO_TEST_RXD (GPIO_NUM_16)
#define ECHO_TEST_RTS (UART_PIN_NO_CHANGE)
#define ECHO_TEST_CTS (UART_PIN_NO_CHANGE)

#define BUF_SIZE 1024

void setup()
{
  uart_config_t uart_config = {
      .baud_rate = 10400,
      .data_bits = UART_DATA_8_BITS,
      .parity = UART_PARITY_DISABLE,
      .stop_bits = UART_STOP_BITS_1,
      .flow_ctrl = UART_HW_FLOWCTRL_DISABLE};
  uart_param_config(UART_NUM_2, &uart_config);
  uart_set_pin(UART_NUM_2, ECHO_TEST_TXD, ECHO_TEST_RXD, ECHO_TEST_RTS, ECHO_TEST_CTS);
  uart_driver_install(UART_NUM_2, BUF_SIZE * 2, 0, 0, NULL, 0);
}

void loop()
{
  const char *data = "test";
  uart_write_bytes(UART_NUM_2, data, 5);
  delay(10);
}
```

And here there is the logic analizer capture of the two + another capture using the arduino sketch and setting a baudrate of 9600
[esp32-baud.zip](https://github.com/espressif/arduino-esp32/files/4343945/esp32-baud.zip)

It looks like there is always a frame error at the end of every byte
[image]

And sometimes two bytes are send one after the other
[image]

What do you think?

aster94, 2020-04-05:
@koobest did you had the time to run any test?

lbernstone, 2020-04-05:
Please merge and test https://github.com/espressif/arduino-esp32/pull/3713 and https://github.com/espressif/arduino-esp32/pull/3664

aster94, 2020-04-05:
good news! the #3664 did the trick

In the zip file there is a logic analyzer capture from the stable release via package manager, a capture from master (so with 3664) and a capture from master with #3713 merged

[LOGIC.zip](https://github.com/espressif/arduino-esp32/files/4435082/LOGIC.zip)

aster94, 2020-04-06:
@lbernstone I am closing the issue because to me seems that everything is ok, if you note something wrong in the attacched logic analyzer captures let me know!

closed 2020-04-06 by aster94