espressif/arduino-esp32 #2005: I2C Wire got bug when used on Core 0

human890209, 2018-10-29:
### Hardware:
Board:							MH-ET ESP32MiniKit
Core Installation/update date:			1/9/2018
IDE name:							Arduino IDE
Flash Frequency:					40Mhz
PSRAM enabled:                                          no
Upload Speed:						921600
Computer OS:                                            Windows 10

### Description:
Hi,
 I'm trying to move some communication function to Core 0 by creating a new task on core 0
`	xTaskCreatePinnedToCore(loop2, "loop2Task", 2000, NULL, 1, NULL, 0);`
Most functions work well for now.
But Wire got some bugs.
The read or write result is wrong, got some wild number like 96, 255, 253 may be some symbol of I2C protocol.
I've found that the first transmit or request is always right and using Wire.begin() could recover the hung bus.
I call begin() after each transmits or request to reset the Wire. This works for some simple implementations. But this can't fix all kinds of problem and Wire is used by a lot of sensor library.

Hope expert like @stickbreaker could help me out.

stickbreaker, 2018-10-29:
@human890209 compile this code to see which `Wire()` version you are using:
```c++
#include "Wire.h"

void setup(){
char ch[]=STICKBREAKER;
}

void loop(){}
```
It will generate a compiler error that displays the Version, such as:

>
 >   C:\Users\user\Documents\Arduino\hardware\espressif\esp32\libraries\Wire\src/Wire.h:33:24: error: too many decimal points in number
>
 >   #define STICKBREAKER V1.0.1

And I will need to see the code you are using to communicate with the Device, and I need to know which device you are trying to access.

Also, realize the `Wire()` library is single threaded.  This means you cannot use the I2C bus from multiple tasks simultaneously.  You must coordinate all access such that there is no overlap.

Chuck.

chinswain, 2018-10-29:
Try this:

```C
#include <Wire.h>

static int taskCore = 0;
unsigned long previousMillis = 0;

void scan() {
  Serial.println("\n Scanning I2C Addresses");
  uint8_t cnt = 0;
  for (uint8_t i = 0; i < 0x7F; i++) {
    Wire.beginTransmission(i);
    uint8_t ec = Wire.endTransmission(true);
    if (ec == 0) { // Device ACK'd
      if (i < 16)Serial.print('0');
      Serial.print(i, HEX);
      cnt++;
    }
    else Serial.print("..");
    Serial.print(' ');
    if ((i & 0x0f) == 0x0f)Serial.println();
  }
  Serial.print("Scan Completed, ");
  Serial.print(cnt);
  Serial.println(" I2C Devices found.");
}

void coreTask( void * pvParameters ) {

  Wire.begin();

  while (true) {
    unsigned long currentMillis = millis();
    if (currentMillis - previousMillis >= 1000) {
      previousMillis = currentMillis;
      scan();
    }
  }
}

void setup() {

  Serial.begin(112500);
  delay(1000);

  Serial.print("Starting to create task on core ");
  Serial.println(taskCore);

  xTaskCreatePinnedToCore(
    coreTask,   /* Function to implement the task */
    "coreTask", /* Name of the task */
    10000,      /* Stack size in words */
    NULL,       /* Task input parameter */
    0,          /* Priority of the task */
    NULL,       /* Task handle. */
    taskCore);  /* Core where the task should run */

  Serial.println("Task created...");
}

void loop() {
  delay(100);
}
```

human890209, 2018-10-30:
Hi, @stickbreaker 
I find the version is `#define STICKBREAKER V1.0.1` in Wire.h
A simpler code is 
```
// YANG add for process on Core 0
void I2CGeneralMaster::processCore0()
{
	// Not Processing or Processing2, return
	if (static_status != Processing && static_status != Processing2) return;

	TwoWireMOD* tw;

	switch (static_data[0])
	{
		default:
		//case 0:
			tw = &WireMOD;
		break;
		case 1:
			tw = &WireMOD2;
		break;
	}

	byte address = static_data[1];

	// Check if BUS is busy
	if (!tw->contactSlave(address))
	{
		return;
	}

	if (static_length > 0 && static_status == Processing)
	{
		tw->beginTransmission(address);
		for (byte i = 0; i < static_length; i++)
		{
			tw->write(static_buffer[i]);
		}
		byte result = tw->endTransmission();
		//spf("endTransmission result: ");
		//spl(result);

		// YANG add begin to recover hung bus on Core 0
		tw->begin();

		// Check result state
		if (result == I2C_ERROR_OK)
		{
			// Success
			static_status = Processing2;
		}else{
			// Failed
			return;
		}
	}

	byte output_quentity = static_data[4];

	if (output_quentity > 0)
	{
		// Request
		if (tw->requestFrom(address, output_quentity) == 0)
		{
			// YANG add begin to recover hung bus on Core 0
			tw->begin();
			return;
		}

		// Store the received data into buffer
		byte index = 0;
		for (byte i = 0; i < output_quentity; i++)
		{
			static_buffer[index] = tw->read();
			index++;
		}

		// YANG add begin to recover hung bus on Core 0
		tw->begin();
	}

	// Set status to Finish
	static_status = Finish;
}
```
TwoWireMOD adds some flags and custom functions and inherits Wire.
This is to communicate with an Arduino which is set up as an I2C Client to do some test. I find the bugs with this.
You could notice I add some tw->begin() to fix the bug for now. But that is not stable, I'm still testing this fix.

And I'm working on the extEEPROM library, trying to get it working on Core 0 with this fix, but it doesn't work 100% or now. So I post to ask for help.

Thought I2C tasks are on both my core. But I don't fire them at the same time. I'm not using a lot of sensor of I2C for now. Only this self-made general master and extEEPROM. And the test I do only use one of them.

human890209, 2018-10-30:
Hi, @chinswain.
Thank you for your example.

stickbreaker, 2018-10-30:
@human890209 the i2c subsystem has extensive debugging controls, start by uncommenting `ENABLE_I2C_DEBUG_BUFFER` here: [enable debug buffer](https://github.com/espressif/arduino-esp32/blob/85032b226c7775bae510e954112823a2ae32181a/cores/esp32/esp32-hal-i2c.c#L38)
then set the ESP32 debug level to VERBOSE

Any Errors will appear on the Serial Monitor.

You can manually force the I2C subsytem to output diagnostic data with
` Wire.setDebugFlags( uint32_t setBits, uint32_t resetBits);`

the values of `setBits` and `resetBits` are:
>  Debug actions have 3 currently defined locus 
 0xXX------ : at entry of ProcQueue, before peripheral ISR starts moving data 
 0x--XX---- : at exit of ProcQueue, after peripheral ISR, has completed
 0x------XX : at entry of Flush
> 
 >bit 0 causes DumpI2c to execute , shows global i2c control record
 bit 1 causes DumpInts to execute, list all interrupts serviced during this request
 bit 2 causes DumpCmdqueue to execute, lists each i2c queue (action) element
 bit 3 causes DumpStatus to execute, shows current i2c Peripheral status word
 bit 4 causes DumpFifo to execute, shows all data that is sent out through peripheral

I would identify where you see `Wire()` failing and turn on debuging before the failing `Wire()` call.

You said the `Wire.requestFrom()` was returning bad data, so add
```c++
        tw->setDebugFlags(0x00070000,0); // set debug to output at completion of i2c action
	if (tw->requestFrom(address, output_quentity) == 0)
	{
            tw->setDebugFlags(0,0x00070000); // turn off debug
        // .....
            return;
        }
       tw->setDebugFlags(0,0x00070000); // turn off debug
```

chuck.

human890209, 2018-10-30:
Hi, @stickbreaker.
I add the debug in to my code and comment out my tw->begin() fix to reproduce the bugs.
Things are complecated. I should explain more to help you understand my bug.
This is my Arduino I2C Test Slave's Sketch:
```
#include "Arduino.h"
#include "Wire.h"

#define ADDRESS 	0x30
#define FREQUANCY	400000

volatile byte buffer[32] = {8};
volatile byte buffer_pos = 1;
volatile bool updated = false;
volatile bool sent = false;

void setup()
{
	Serial.begin(115200);
	Serial.println("Start");

	Wire.begin(ADDRESS);
	Wire.setClock(FREQUANCY);
	Wire.onReceive(receiveCommand);
	Wire.onRequest(messageRequested);
}

void receiveCommand(int howMany)
{
	for (byte i = 0; i < howMany; i++)
	{
		buffer[i] = Wire.read();
	}
	buffer_pos = howMany;
	updated = true;
}

void messageRequested()
{
	Wire.write((byte*)buffer,buffer_pos);
	sent = true;
}

void loop()
{
	if (updated)
	{
		updated = false;
		Serial.print(F("Receive: "));
		for (byte i = 0; i < buffer_pos; i++)
		{
			if (i != 0)
			{
				Serial.print(' ');
			}
			Serial.print(buffer[i]);
		}
		Serial.println();
	}

	if (sent)
	{
		sent = false;
		Serial.print(F("Sent: "));
		for (byte i = 0; i < buffer_pos; i++)
		{
			if (i != 0)
			{
				Serial.print(' ');
			}
			Serial.print(buffer[i]);
		}
		Serial.println();
	}
}
```
It print the received and sent data in Serial Monitor, that's how I test my General Master Result.

Step 1, I comment out all my tw->begin() fix to reproduce the bug.
I use GeneralMaster transmit a Long 10000, which is encoded into 16 39 0 0 by my code and transmit with GeneralMaster on Core 0, and than request the 4 bytes.
And the Serial Monitor of Arduino Test Slave print:
```
Receive: 16 39 255 255
Receive: 255 255 255 255
```
I fire this with a button, so I press more times.
The result:
```
Receive: 16 39 255 255
Receive: 255 255 255 255(Last one)
Sent: 255 255 255 255
Receive: 255 255 255 255(press again)
Sent: 255 255 255 255
Receive: 255 255 255 255(press again)
Sent: 255 255 255 255
Receive: 255 255 255 255(press again)
```
And I reset my Master ESP32 and redo again.
```
Receive: 16 39 255 255
Receive: 255 255 255 255
Sent: 255 255 255 255
Receive: 255 255 255 255
Sent: 255 255 255 255
Receive: 255 255 255 255
Sent: 255 255 255 255
Receive: 255 255 255 255
Receive: 16 39 255 255(After reset first press)
Receive: 255 255 255 255
Sent: 255 255 255 255
Receive: 255 255 255 255
Sent: 255 255 255 255
Receive: 255 255 255 255
```
And I turned on the debug for I2C as you told me todo. 
I can't reproduce the bug.... I2C works normally.
That's why I need to explain more
```
Receive: 16 39 255 255
Receive: 255 255 255 255
Sent: 255 255 255 255
Receive: 255 255 255 255
Sent: 255 255 255 255
Receive: 255 255 255 255
Sent: 255 255 255 255
Receive: 255 255 255 255
Receive: 16 39 255 255
Receive: 255 255 255 255
Sent: 255 255 255 255
Receive: 255 255 255 255
Sent: 255 255 255 255
Receive: 255 255 255 255
Receive: 16 39 255 255
Receive: 255 255 255 255
Sent: 255 255 255 255
Receive: 255 255 255 255
Receive: 16 39 255 255
Receive: 255 255 255 255
Sent: 255 255 255 255
Receive: 255 255 255 255
Sent: 255 255 255 255
Receive: 255 255 255 255
Sent: 255 255 255 255
Receive: 255 255 255 255(Debug Off)
Receive: 16 39 0 0 (Debug ON)
Receive: 16 39 0 0
Sent: 16 39 0 0
Receive: 16 39 0 0
Sent: 16 39 0 0
Receive: 16 39 0 0
Sent: 16 39 0 0
Receive: 16 39 0 0
Sent: 16 39 0 0
Receive: 16 39 0 0
Sent: 16 39 0 0
Receive: 16 39 0 0
Sent: 16 39 0 0
Receive: 16 39 0 0
Sent: 16 39 0 0
Receive: 16 39 0 0
Sent: 16 39 0 0
```
I add debug before
`tw->beginTransmission(address);`
and reset it before each return and the last one, cause the transmission got bug, too.

human890209, 2018-10-30:
Hi, @stickbreaker 
After I turn the Debug On, no bug could be found.
Now, I comment out the debug lines to reproduce the bug again.

Test1-No delay
```
---Press Button
Receive: 16 39 255 255
Receive: 255 255 255 255
No I2C report

---Press Button 
Sent: 255 255 255 255
Receive: 255 255 255 255
I2C: endTransmission result: ACK

---Press Button
Sent: 255 255 255 255
Receive: 255 255 255 255
I2C: endTransmission result: ACK
```

Test2-Delay(200) between Transmit and Request
```
---Press Button
Receive: 16 39 0 0
Receive: 255 255 255 255

---Press Button
Sent: 255 255 255 255
Receive: 96 255 255 255
Receive: 255 255 255 255
I2C: endTransmission result: ACK

---Press Button
Sent: 255 255 255 255
Receive: 96 255 255 255
Receive: 255 255 255 255
I2C: endTransmission result: ACK
```

My Code:
```
// YANG add for process on Core 0
void I2CGeneralMaster::processCore0()
{
	// Not Processing or Processing2, return
	if (static_status != Processing && static_status != Processing2) return;

	TwoWireMOD* tw;

	switch (static_data[0])
	{
		default:
		//case 0:
			tw = &WireMOD;
		break;
		case 1:
			tw = &WireMOD2;
		break;
	}

	byte address = static_data[1];

	// Check if BUS is busy
	if (!tw->contactSlave(address))
	{
		return;
	}

	//tw->setDebugFlags(0x00070000,0); // set debug to output at completion of i2c action

	if (static_length > 0 && static_status == Processing)
	{
		tw->beginTransmission(address);
		for (byte i = 0; i < static_length; i++)
		{
			tw->write(static_buffer[i]);
		}
		byte result = tw->endTransmission();

		// YANG add begin to recover hung bus on Core 0
		//tw->begin();

		// Check result state
		if (result == I2C_ERROR_OK)
		{
			// Success
			static_status = Processing2;
		}else{
			// Failed
			//tw->setDebugFlags(0,0x00070000); // turn off debug

			spf("endTransmission result: ");
			spl(tw->getErrorText(result));

			return;
		}
	}

	delay(200);

	byte output_quentity = static_data[4];

	if (output_quentity > 0)
	{
		// Request
		if (tw->requestFrom(address, output_quentity) == 0)
		{
			// YANG add begin to recover hung bus on Core 0
			//tw->begin();

			//tw->setDebugFlags(0,0x00070000); // turn off debug

			spf("requestFrom result: ");
			spl(tw->getErrorText(tw->lastError()));

			return;
		}

		// Store the received data into buffer
		byte index = 0;
		for (byte i = 0; i < output_quentity; i++)
		{
			static_buffer[index] = tw->read();
			index++;
		}

		// YANG add begin to recover hung bus on Core 0
		//tw->begin();
	}

	// Set status to Finish
	static_status = Finish;

	//tw->setDebugFlags(0,0x00070000); // turn off debug
}
```

human890209, 2018-10-30:
And I used to use I2CGeneralMaster on Core 1 normally call in loop(), with the same Arduino Test Client. And same I2C settings(clock, address, pins, ...) Everything works fine. I think Arduino should work fine.

human890209, 2018-10-30:
Hi, @stickbreaker 
I enable Debug again, and this time I reproduced the bug. 
I don't put a delay(200) between Transmit and Request.

Arduino Serial Log:
```
Receive: 16 39 0 0
Receive: 255 255 255 255
Sent: 255 255 255 255
Receive: 96 255 255 255
Receive: 255 255 255 255
Sent: 255 255 255 255
Receive: 96 255 255 255
Receive: 255 255 255 255
```

ESP32 Log:

```
14:24:18.750 websocket.js:192 <--[I][esp32-hal-i2c.c:431] i2cTriggerDumps(): after ProcQueue
14:24:18.754 websocket.js:192 <--[E][esp32-hal-i2c.c:314] i2cDumpI2c(): i2c=0x3ffc1260
14:24:18.756 websocket.js:192 <--[I][esp32-hal-i2c.c:315] i2cDumpI2c(): dev=0x60013000 date=0x16042000
14:24:18.758 websocket.js:192 <--[I][esp32-hal-i2c.c:317] i2cDumpI2c(): lock=0x3ffd7528
14:24:18.760 websocket.js:192 <--[I][esp32-hal-i2c.c:319] i2cDumpI2c(): num=0
14:24:18.762 websocket.js:192 <--[I][esp32-hal-i2c.c:320] i2cDumpI2c(): mode=1
14:24:18.764 websocket.js:192 <--[I][esp32-hal-i2c.c:321] i2cDumpI2c(): stage=3
14:24:18.765 websocket.js:192 <--[I][esp32-hal-i2c.c:322] i2cDumpI2c(): error=1
14:24:18.766 websocket.js:192 <--[I][esp32-hal-i2c.c:323] i2cDumpI2c(): event=0x3ffd7608 bits=10
14:24:18.769 websocket.js:192 <--[I][esp32-hal-i2c.c:324] i2cDumpI2c(): intr_handle=0x3ffd7638
14:24:18.769 websocket.js:192 <--[I][esp32-hal-i2c.c:325] i2cDumpI2c(): dq=0x3ffb12f0
14:24:18.771 websocket.js:192 <--[I][esp32-hal-i2c.c:326] i2cDumpI2c(): queueCount=1
14:24:18.772 websocket.js:192 <--[I][esp32-hal-i2c.c:327] i2cDumpI2c(): queuePos=0
14:24:18.773 websocket.js:192 <--[I][esp32-hal-i2c.c:328] i2cDumpI2c(): errorByteCnt=0
14:24:18.774 websocket.js:192 <--[I][esp32-hal-i2c.c:329] i2cDumpI2c(): errorQueue=0
14:24:18.776 websocket.js:192 <--[I][esp32-hal-i2c.c:330] i2cDumpI2c(): debugFlags=0x00070000
14:24:18.777 websocket.js:192 <--[I][esp32-hal-i2c.c:287] i2cDumpDqData(): [0] 7bit 30 W STOP buf@=0x3ffc4f7e, len=4, pos=4, ctrl=11111
14:24:18.778 websocket.js:192 <--[I][esp32-hal-i2c.c:305] i2cDumpDqData(): 0x0000: .'..                             10 27 00 00 
14:24:18.779 websocket.js:192 <--[I][esp32-hal-i2c.c:341] i2cDumpInts(): 0 row	count	INTR	TX	RX	Tick 
14:24:18.780 websocket.js:192 <--[I][esp32-hal-i2c.c:345] i2cDumpInts(): [01]	0x0001	0x0002	0x0005	0x0000	0x00002835
14:24:18.781 websocket.js:192 <--[I][esp32-hal-i2c.c:345] i2cDumpInts(): [02]	0x0001	0x0200	0x0000	0x0000	0x00002835
14:24:18.782 websocket.js:192 <--[I][esp32-hal-i2c.c:345] i2cDumpInts(): [03]	0x0001	0x0080	0x0000	0x0000	0x00002836
14:24:18.783 websocket.js:192 <--[E][esp32-hal-i2c.c:243] i2cDumpCmdQueue(): [ 0]	Y	RSTART	val[0]	exp[0]	en[0]	bytes[0]
14:24:18.784 websocket.js:192 <--[E][esp32-hal-i2c.c:243] i2cDumpCmdQueue(): [ 1]	Y	WRITE	val[0]	exp[0]	en[1]	bytes[1]
14:24:18.785 websocket.js:192 <--[E][esp32-hal-i2c.c:243] i2cDumpCmdQueue(): [ 2]	Y	WRITE	val[0]	exp[0]	en[1]	bytes[4]
14:24:18.787 websocket.js:192 <--[E][esp32-hal-i2c.c:243] i2cDumpCmdQueue(): [ 3]	Y	STOP	val[0]	exp[0]	en[0]	bytes[0]
14:24:18.788 websocket.js:192 <--[E][esp32-hal-i2c.c:243] i2cDumpCmdQueue(): [ 4]	N	RSTART	val[0]	exp[0]	en[0]	bytes[0]
14:24:18.789 websocket.js:192 <--[E][esp32-hal-i2c.c:243] i2cDumpCmdQueue(): [ 5]	N	RSTART	val[0]	exp[0]	en[0]	bytes[0]
14:24:18.790 websocket.js:192 <--[E][esp32-hal-i2c.c:243] i2cDumpCmdQueue(): [ 6]	N	RSTART	val[0]	exp[0]	en[0]	bytes[0]
14:24:18.792 websocket.js:192 <--[E][esp32-hal-i2c.c:243] i2cDumpCmdQueue(): [ 7]	N	RSTART	val[0]	exp[0]	en[0]	bytes[0]
14:24:18.793 websocket.js:192 <--[E][esp32-hal-i2c.c:243] i2cDumpCmdQueue(): [ 8]	N	RSTART	val[0]	exp[0]	en[0]	bytes[0]
14:24:18.794 websocket.js:192 <--[E][esp32-hal-i2c.c:243] i2cDumpCmdQueue(): [ 9]	N	RSTART	val[0]	exp[0]	en[0]	bytes[0]
14:24:18.796 websocket.js:192 <--[E][esp32-hal-i2c.c:243] i2cDumpCmdQueue(): [10]	N	RSTART	val[0]	exp[0]	en[0]	bytes[0]
14:24:18.797 websocket.js:192 <--[E][esp32-hal-i2c.c:243] i2cDumpCmdQueue(): [11]	N	RSTART	val[0]	exp[0]	en[0]	bytes[0]
14:24:18.797 websocket.js:192 <--[E][esp32-hal-i2c.c:243] i2cDumpCmdQueue(): [12]	N	RSTART	val[0]	exp[0]	en[0]	bytes[0]
14:24:18.799 websocket.js:192 <--[E][esp32-hal-i2c.c:243] i2cDumpCmdQueue(): [13]	N	RSTART	val[0]	exp[0]	en[0]	bytes[0]
14:24:18.799 websocket.js:192 <--[E][esp32-hal-i2c.c:243] i2cDumpCmdQueue(): [14]	N	RSTART	val[0]	exp[0]	en[0]	bytes[0]
14:24:18.800 websocket.js:192 <--[E][esp32-hal-i2c.c:243] i2cDumpCmdQueue(): [15]	N	RSTART	val[0]	exp[0]	en[0]	bytes[0]

14:24:18.801 websocket.js:192 <--[I][esp32-hal-i2c.c:431] i2cTriggerDumps(): after ProcQueue
14:24:18.802 websocket.js:192 <--[E][esp32-hal-i2c.c:314] i2cDumpI2c(): i2c=0x3ffc1260
14:24:18.802 websocket.js:192 <--[I][esp32-hal-i2c.c:315] i2cDumpI2c(): dev=0x60013000 date=0x16042000
14:24:18.803 websocket.js:192 <--[I][esp32-hal-i2c.c:317] i2cDumpI2c(): lock=0x3ffd7528
14:24:18.805 websocket.js:192 <--[I][esp32-hal-i2c.c:319] i2cDumpI2c(): num=0
14:24:18.805 websocket.js:192 <--[I][esp32-hal-i2c.c:320] i2cDumpI2c(): mode=1
14:24:18.806 websocket.js:192 <--[I][esp32-hal-i2c.c:321] i2cDumpI2c(): stage=3
14:24:18.807 websocket.js:192 <--[I][esp32-hal-i2c.c:322] i2cDumpI2c(): error=1
14:24:18.808 websocket.js:192 <--[I][esp32-hal-i2c.c:323] i2cDumpI2c(): event=0x3ffd7608 bits=10
14:24:18.808 websocket.js:192 <--[I][esp32-hal-i2c.c:324] i2cDumpI2c(): intr_handle=0x3ffd7638
14:24:18.809 websocket.js:192 <--[I][esp32-hal-i2c.c:325] i2cDumpI2c(): dq=0x3ffb12f0
14:24:18.809 websocket.js:192 <--[I][esp32-hal-i2c.c:326] i2cDumpI2c(): queueCount=1
[... 231 lines cut ...]
[... cut ...]
14:24:22.779 websocket.js:192 <--[E][esp32-hal-i2c.c:243] i2cDumpCmdQueue(): [11]	N	RSTART	val[0]	exp[0]	en[0]	bytes[0]
14:24:22.780 websocket.js:192 <--[E][esp32-hal-i2c.c:243] i2cDumpCmdQueue(): [12]	N	RSTART	val[0]	exp[0]	en[0]	bytes[0]
14:24:22.780 websocket.js:192 <--[E][esp32-hal-i2c.c:243] i2cDumpCmdQueue(): [13]	N	RSTART	val[0]	exp[0]	en[0]	bytes[0]
14:24:22.794 websocket.js:192 <--[E][esp32-hal-i2c.c:243] i2cDumpCmdQueue(): [14]	N	RSTART	val[0]	exp[0]	en[0]	bytes[0]
14:24:22.802 websocket.js:192 <--[E][esp32-hal-i2c.c:243] i2cDumpCmdQueue(): [15]	N	RSTART	val[0]	exp[0]	en[0]	bytes[0]
```

human890209, 2018-10-30:
There are 4 extra 255 be received by the Slave.
And the later Received data's first byte is 96.
Got ACK error when transmitting.

human890209, 2018-10-30:
I use a logic analyzer to check the signal.
I disabled everything include  `tw->setDebugFlags(0x00070000,0)` and `tw->begin()` and set the delay between transmit and request to 100ms. The interval between my I2CGeneralMaster::processCore0() set to 300ms, that the phase could be identify by time easily.

```
// YANG add for process on Core 0
void I2CGeneralMaster::processCore0()
{
	// Not Processing or Processing2, return
	if (static_status != Processing && static_status != Processing2) return;

	TwoWireMOD* tw;

	switch (static_data[0])
	{
		default:
		//case 0:
			tw = &WireMOD;
		break;
		case 1:
			tw = &WireMOD2;
		break;
	}

	byte address = static_data[1];

	// Check if BUS is busy
	if (!tw->contactSlave(address))
	{
		return;
	}

	//tw->setDebugFlags(0x00070000,0); // set debug to output at completion of i2c action

	if (static_length > 0 && static_status == Processing)
	{
		tw->beginTransmission(address);
		for (byte i = 0; i < static_length; i++)
		{
			tw->write(static_buffer[i]);
		}
		byte result = tw->endTransmission();

		delay(100);

		// YANG add begin to recover hung bus on Core 0
		//delayMicroseconds(500);
		//tw->begin();

		// Check result state
		if (result == I2C_ERROR_OK)
		{
			// Success
			static_status = Processing2;
		}else{
			// Failed
			//tw->setDebugFlags(0,0x00070000); // turn off debug

			//spf("endTransmission result: ");
			//spl(tw->getErrorText(result));
			return;
		}
	}

	byte output_quentity = static_data[4];

	if (output_quentity > 0)
	{
		// Request
		if (tw->requestFrom(address, output_quentity) != output_quentity)
		{
			// YANG add begin to recover hung bus on Core 0
			//tw->begin();

			//tw->setDebugFlags(0,0x00070000); // turn off debug

			//spf("requestFrom result: ");
			//spl(tw->getErrorText(tw->lastError()));

			return;
		}

		// Store the received data into buffer
		byte index = 0;
		for (byte i = 0; i < output_quentity; i++)
		{
			static_buffer[index] = tw->read();
			index++;
		}

		// YANG add begin to recover hung bus on Core 0
		//tw->begin();
	}

	// Set status to Finish
	static_status = Finish;

	//tw->setDebugFlags(0,0x00070000); // turn off debug
}
```
Overall: 
[image]
There are 10 groups divided by 300ms gap.
I name the first group Group1. Group1 got 2 messages, so Messages will be called 1-1 and 1-2.
The latter groups got 3 messages each. Messages will be called 2-1, 2-2, 2-3, 3-1, 3-2, 3-3,....

Group 1
Message 1-1
[image]
I guess this is a correct 16 39 0 0 Transmit Message

Message 1-2
[image]
I think the 255 255 255 255 message is this. It's wired that the first group doesn't send Request Message.

Group 2
Message 2-1
[image]
The second group begins with a Request Message, it's wired.

Message 2-2
[image]
I guess this is sent by Slave.

Message 2-3
[image]
Another 255 255 255 255 message.

Group 3
Message 3-1
[image]
Same as Group2, started with a Request

Message 3-2
[image]
I guess this is sent by Slave. Just like Message 2-2

Message 3-3
[image]
Another 255 255 255 255 message. Just like Message 2-3

Group 4-10 is same like Group 2 or 3.

Arduino Slave's Log
```
Receive: 16 39 0 0
Receive: 255 255 255 255

Sent: 255 255 255 255
Receive: 96 16 39 0
Receive: 255 255 255 255

Sent: 255 255 255 255
Receive: 96 16 39 0
Receive: 255 255 255 255

Sent: 255 255 255 255
Receive: 96 16 39 0
Receive: 255 255 255 255
```

stickbreaker, 2018-10-30:
Transaction at 56.540 shows ACK ERROR on address byte.
56.551 ERROR_BYTE = -1 address byte in queue 0 
56.561 DUMPCMDQUEUE N means peripheral did not execute command(write of 4 data bytes) 
56.556 DUMPINTS 0x0400 ACK ERROR

54.414 Shows ACK ERROR also

human890209, 2018-10-30:
Hi, @stickbreaker 
I update more evidence. How to deal with this bug?

stickbreaker, 2018-10-30:
Not bug, slave did not answer.

When you see ACK ERROR(which means Not Acknowledge) retry command.

Slave was Busy, not listening to bus.

Chuck.

stickbreaker, 2018-10-30:
Just saw scope pictures.

stickbreaker, 2018-10-30:
Group 1 decode is invalid
Change scope to analog capture

Do you have pullups on SDA, SCL? Need 2.4k and esp32 is 3.3v Arduino Uno is 5v
Need level shifters between. ESP32 CANNOT handle 5v pullups. And Uno wont work with 3.3v pullups.

Chuck

human890209, 2018-10-30:
Hi, @stickbreaker 
I'm using a Level shifter(5v - 3.3v for I2C) module between Arduino Nano and ESP32, and it got Pullup resister on it.
I don't know if the resister is 2.4k.
My logic scope doesn't get an analog mode.

stickbreaker, 2018-10-30:
See little triangles on SCL signal? They mark rising Edge detection by scope, cannot have two in a row, need negative edge between.

Signal is glitchy, need to see raw signal. Set scope to 2v/div 10ms/div

Chuck.

stickbreaker, 2018-10-30:
What scope?

human890209, 2018-10-30:
An old model of Saleae

stickbreaker, 2018-10-30:
Describe level shifter, should have pullups on both sides, usually 10k to 3.3v and 10k to 5v

stickbreaker, 2018-10-30:
Saleae usb? I think it does have analog mode. Need to see raw signal.

human890209, 2018-10-30:
Before I switch the I2C to Core 0, I used it on Core 1 before. And tested with this Arduino, too. 
Everything works fine before. And I2C works fine with ExtEEPROM and MPU on Core 1. 
New Saleae got analog. Mine is an old model.
I will take a picture of my shifter.

stickbreaker, 2018-10-30:
How many channels on scope?

stickbreaker, 2018-10-30:
I2c doesnt show any bugs, problem outside esp32

human890209, 2018-10-30:
> How many channels on scope?
16 Logic Channel

stickbreaker, 2018-10-30:
Try to slow down Esp32 100k or 50k

human890209, 2018-10-30:
Shifter
[image]

> I2c doesnt show any bugs, problem outside esp32

Is the 255 255 255 255 and 96 16 39 0 normal? I don't know if this is a part of the protocol of I2C.

human890209, 2018-10-30:
Ok, I will try a lower frequency.
But if the speed is too slow, I have to use it on Core 1 again...
I'm wondering if Core 0 is different from Core 1? Did I miss some settings for the Task of Core 0?

stickbreaker, 2018-10-30:
No, not correct.
Group 3.3 write 255, 255, 255, 255
3.2 write 0x60 0x10 0x27 0x00 

Actually 3.2 is weird.

Output is 0x60 (address 0x30 + write)
0x60
0x10
0x27
0x00

Looks like merge of NAK Write and next Write?

stickbreaker, 2018-10-30:
Looks like issue with peripheral fifo, old address from NAK failure still in fifo. New data appended. So peripheral sent 0x60, 0x60(96 dec), 0x10(16 dec), 0x27(39 dec), 0x0
The last zero never left fifo.

Change debug cmd to
0x001f0000

human890209, 2018-10-30:
So should I use 400k as frequency and test again?

stickbreaker, 2018-10-30:
Pullups on level shifter only 10k (103 markings)

human890209, 2018-10-30:
Thanks, @stickbreaker 
I will clean up my codes and make some more test on 100k, I the error is reproduced, I will use the new debug flag and paste my code and I2C log here.

chinswain, 2018-10-30:
How long are the wires?

human890209, 2018-10-31:
Hi, @chinswain 
20cm lines on both sides of the Shifter Module.

human890209, 2018-10-31:
Hi, @stickbreaker 
I've used my logic scope to test the I2C work under 50k and 100 frequency. I see a different pattern from 400k. 

The delay setting is the same Transmit-100ms-Request-300ms-Transmit-100ms-Request-300ms...

Code:
```
// YANG add for process on Core 0
void I2CGeneralMaster::processCore0()
{
	// Not Processing, return
	if (static_status != Processing) return;

	TwoWireMOD* tw;

	switch (static_data[0])
	{
		default:
		//case 0:
			tw = &WireMOD;
		break;
		case 1:
			tw = &WireMOD2;
		break;
	}

	byte address = static_data[1];
	byte output_quentity = static_data[4];
	byte result;

	// YANG updated this with updateCore
	// Check if BUS is busy
	if (!tw->contactSlave(address))
	{
		return;
	}

	#ifdef I2C_DEBUG_FLAG
		// set debug to output at completion of i2c action
		tw->setDebugFlags(I2C_DEBUG_FLAG,0);
	#endif // I2C_DEBUG_FLAG

	if (static_length > 0 && static_status == Processing)
	{
		tw->beginTransmission(address);
		for (byte i = 0; i < static_length; i++)
		{
			tw->write(static_buffer[i]);
		}

		result = tw->endTransmission();

		delay(100);

//		if (output_quentity == 0)
//		{
//			// No request later
//			result = tw->endTransmission();
//		}else{
//			// Request later
//			result = tw->endTransmission(false);
//		}

		// Check result state
		if ( result != I2C_ERROR_OK && result != I2C_ERROR_CONTINUE)
		{
			// Failed
			#ifdef I2C_DEBUG_FLAG
				// turn off debug
				tw->setDebugFlags(0,I2C_DEBUG_FLAG);
			#endif // I2C_DEBUG_FLAG

			spf("endTransmission result: ");
			spl(tw->getErrorText(result));
			return;
		}
	}

	if (output_quentity > 0)
	{
		// Request
		if (tw->requestFrom(address, output_quentity) != output_quentity)
		{
			#ifdef I2C_DEBUG_FLAG
				// turn off debug
				tw->setDebugFlags(0,I2C_DEBUG_FLAG);
			#endif // I2C_DEBUG_FLAG

			spf("requestFrom result: ");
			spl(tw->getErrorText(tw->lastError()));
			return;
		}

		// Store the received data into buffer
		byte index = 0;
		for (byte i = 0; i < output_quentity; i++)
		{
			static_buffer[index] = tw->read();
			index++;
		}
	}

	// Set status to Finish
	static_status = Finish;

	#ifdef I2C_DEBUG_FLAG
		// turn off debug
		tw->setDebugFlags(0,I2C_DEBUG_FLAG);
	#endif // I2C_DEBUG_FLAG
}
```

The scope pattern of 50k and 100k are almost the same, so I use 100k's screenshot.

Test1
I2C Frequency: 100,000
Overall:
[image]
11 Groups divided by 300ms interval, each Group got 2 Messages
Group 1, Group 2,...
Message 1-1, Message 1-2, Message 2-1, Message 2-2...

Group 1
Message 1-1
[image]

Message 2-1
[image]

Group 2
Message 2-1
[image]

Message 2-2
[image]

I notice that there are thin signals, these thing signals also exists when the Frequency is 50k
[image]

Test2
I2C Frequency: 400,000
Overall:
[image]
11 Groups divided by 300ms interval
You could notice the 2 3 3 3 pattern, different from the 2 2 2 2 pattern of 100k

Group 1
Message 1-1
[image]
Same as last 400k test's screenshot.

Message 1-2
[image]
Same as last 400k test's screenshot. Why send 255 255 255 255?

Group 2
Message 2-1
[image]
Same as last 400k test's screenshot. Request Message?

Message 2-2
[image]
Same as last 400k test's screenshot. 

Message 2-3
[image]
Same as last 400k test's screenshot. 

I don't know if this strange 2333 pattern is caused by Physical RC problems.
I will use the new debug flag to get the I2C Log later.

human890209, 2018-10-31:
Hi, @stickbreaker 
I use the new debug flag 0x001f0000 to get the I2C log.
It's wired the I2C keep transmitting and don't stop. I guess the return of endTransmitt always not 0, and process0() is been retried again and again.
The operation is changed by debug on.

Overall:
[image]
No Group just keep retrying transmitting, no Request.

I2C Log:
```
09:22:22.055 websocket.js:192 <--[I][esp32-hal-i2c.c:431] i2cTriggerDumps(): after ProcQueue
09:22:22.066 websocket.js:192 <--[E][esp32-hal-i2c.c:314] i2cDumpI2c(): i2c=0x3ffc1260
09:22:22.067 websocket.js:192 <--[I][esp32-hal-i2c.c:315] i2cDumpI2c(): dev=0x60013000 date=0x16042000
09:22:22.067 websocket.js:192 <--[I][esp32-hal-i2c.c:317] i2cDumpI2c(): lock=0x3ffd7528
09:22:22.068 websocket.js:192 <--[I][esp32-hal-i2c.c:319] i2cDumpI2c(): num=0
09:22:22.069 websocket.js:192 <--[I][esp32-hal-i2c.c:320] i2cDumpI2c(): mode=1
09:22:22.071 websocket.js:192 <--[I][esp32-hal-i2c.c:321] i2cDumpI2c(): stage=3
09:22:22.072 websocket.js:192 <--[I][esp32-hal-i2c.c:322] i2cDumpI2c(): error=1
09:22:22.072 websocket.js:192 <--[I][esp32-hal-i2c.c:323] i2cDumpI2c(): event=0x3ffd7608 bits=10
09:22:22.073 websocket.js:192 <--[I][esp32-hal-i2c.c:324] i2cDumpI2c(): intr_handle=0x3ffd7638
09:22:22.074 websocket.js:192 <--[I][esp32-hal-i2c.c:325] i2cDumpI2c(): dq=0x3ffb12f0
09:22:22.074 websocket.js:192 <--[I][esp32-hal-i2c.c:326] i2cDumpI2c(): queueCount=1
09:22:22.075 websocket.js:192 <--[I][esp32-hal-i2c.c:327] i2cDumpI2c(): queuePos=0
09:22:22.075 websocket.js:192 <--[I][esp32-hal-i2c.c:328] i2cDumpI2c(): errorByteCnt=0
09:22:22.076 websocket.js:192 <--[I][esp32-hal-i2c.c:329] i2cDumpI2c(): errorQueue=0
09:22:22.077 websocket.js:192 <--[I][esp32-hal-i2c.c:330] i2cDumpI2c(): debugFlags=0x001F0000
09:22:22.078 websocket.js:192 <--[I][esp32-hal-i2c.c:287] i2cDumpDqData(): [0] 7bit 30 W STOP buf@=0x3ffc4f7e, len=4, pos=4, ctrl=11111
09:22:22.078 websocket.js:192 <--[I][esp32-hal-i2c.c:305] i2cDumpDqData(): 0x0000: .'..                             10 27 00 00 
09:22:22.079 websocket.js:192 <--[I][esp32-hal-i2c.c:341] i2cDumpInts(): 0 row	count	INTR	TX	RX	Tick 
09:22:22.079 websocket.js:192 <--[I][esp32-hal-i2c.c:345] i2cDumpInts(): [01]	0x0001	0x0002	0x0005	0x0000	0x00005d49
09:22:22.081 websocket.js:192 <--[I][esp32-hal-i2c.c:345] i2cDumpInts(): [02]	0x0001	0x0200	0x0000	0x0000	0x00005d49
09:22:22.082 websocket.js:192 <--[I][esp32-hal-i2c.c:345] i2cDumpInts(): [03]	0x0001	0x0080	0x0000	0x0000	0x00005d49
09:22:22.085 websocket.js:192 <--[E][esp32-hal-i2c.c:243] i2cDumpCmdQueue(): [ 0]	Y	RSTART	val[0]	exp[0]	en[0]	bytes[0]
09:22:22.086 websocket.js:192 <--[E][esp32-hal-i2c.c:243] i2cDumpCmdQueue(): [ 1]	Y	WRITE	val[0]	exp[0]	en[1]	bytes[1]
09:22:22.087 websocket.js:192 <--[E][esp32-hal-i2c.c:243] i2cDumpCmdQueue(): [ 2]	Y	WRITE	val[0]	exp[0]	en[1]	bytes[4]
09:22:22.088 websocket.js:192 <--[E][esp32-hal-i2c.c:243] i2cDumpCmdQueue(): [ 3]	Y	STOP	val[0]	exp[0]	en[0]	bytes[0]
09:22:22.088 websocket.js:192 <--[E][esp32-hal-i2c.c:243] i2cDumpCmdQueue(): [ 4]	N	RSTART	val[0]	exp[0]	en[0]	bytes[0]
09:22:22.089 websocket.js:192 <--[E][esp32-hal-i2c.c:243] i2cDumpCmdQueue(): [ 5]	N	RSTART	val[0]	exp[0]	en[0]	bytes[0]
09:22:22.090 websocket.js:192 <--[E][esp32-hal-i2c.c:243] i2cDumpCmdQueue(): [ 6]	N	RSTART	val[0]	exp[0]	en[0]	bytes[0]
09:22:22.091 websocket.js:192 <--[E][esp32-hal-i2c.c:243] i2cDumpCmdQueue(): [ 7]	N	RSTART	val[0]	exp[0]	en[0]	bytes[0]
09:22:22.092 websocket.js:192 <--[E][esp32-hal-i2c.c:243] i2cDumpCmdQueue(): [ 8]	N	RSTART	val[0]	exp[0]	en[0]	bytes[0]
09:22:22.093 websocket.js:192 <--[E][esp32-hal-i2c.c:243] i2cDumpCmdQueue(): [ 9]	N	RSTART	val[0]	exp[0]	en[0]	bytes[0]
09:22:22.094 websocket.js:192 <--[E][esp32-hal-i2c.c:243] i2cDumpCmdQueue(): [10]	N	RSTART	val[0]	exp[0]	en[0]	bytes[0]
09:22:22.094 websocket.js:192 <--[E][esp32-hal-i2c.c:243] i2cDumpCmdQueue(): [11]	N	RSTART	val[0]	exp[0]	en[0]	bytes[0]
09:22:22.095 websocket.js:192 <--[E][esp32-hal-i2c.c:243] i2cDumpCmdQueue(): [12]	N	RSTART	val[0]	exp[0]	en[0]	bytes[0]
09:22:22.096 websocket.js:192 <--[E][esp32-hal-i2c.c:243] i2cDumpCmdQueue(): [13]	N	RSTART	val[0]	exp[0]	en[0]	bytes[0]
09:22:22.097 websocket.js:192 <--[E][esp32-hal-i2c.c:243] i2cDumpCmdQueue(): [14]	N	RSTART	val[0]	exp[0]	en[0]	bytes[0]
09:22:22.098 websocket.js:192 <--[E][esp32-hal-i2c.c:243] i2cDumpCmdQueue(): [15]	N	RSTART	val[0]	exp[0]	en[0]	bytes[0]
09:22:22.100 websocket.js:192 <--[I][esp32-hal-i2c.c:379] i2cDumpStatus(): ack(0) sl_rw(0) to(0) arb(0) busy(0) sl(1) trans(0) rx(0) tx(0) sclMain(6) scl(6)
09:22:22.101 websocket.js:192 <--[I][esp32-hal-i2c.c:418] i2cDumpFifo(): WRITE 0x300010 0027 0000 0000 

09:22:22.257 websocket.js:192 <--[D][esp32-hal-i2c.c:1300] i2cProcQueue():  Gross Timeout Dead start=0x5dcd, end=0x5dff, =50, max=50 error=1
09:22:22.257 websocket.js:192 <--[E][esp32-hal-i2c.c:314] i2cDumpI2c(): i2c=0x3ffc1260
09:22:22.258 websocket.js:192 <--[I][esp32-hal-i2c.c:315] i2cDumpI2c(): dev=0x60013000 date=0x16042000
09:22:22.258 websocket.js:192 <--[I][esp32-hal-i2c.c:317] i2cDumpI2c(): lock=0x3ffd7528
09:22:22.258 websocket.js:192 <--[I][esp32-hal-i2c.c:319] i2cDumpI2c(): num=0
09:22:22.259 websocket.js:192 <--[I][esp32-hal-i2c.c:320] i2cDumpI2c(): mode=1
09:22:22.260 websocket.js:192 <--[I][esp32-hal-i2c.c:321] i2cDumpI2c(): stage=3
09:22:22.261 websocket.js:192 <--[I][esp32-hal-i2c.c:322] i2cDumpI2c(): error=1
09:22:22.261 websocket.js:192 <--[I][esp32-hal-i2c.c:323] i2cDumpI2c(): event=0x3ffd7608 bits=0
[... cut ...]
09:22:22.964 websocket.js:192 <--[E][esp32-hal-i2c.c:243] i2cDumpCmdQueue(): [14]	N	RSTART	val[0]	exp[0]	en[0]	bytes[0]
09:22:22.965 websocket.js:192 <--[E][esp32-hal-i2c.c:243] i2cDumpCmdQueue(): [15]	N	RSTART	val[0]	exp[0]	en[0]	bytes[0]
09:22:22.966 websocket.js:192 <--[I][esp32-hal-i2c.c:379] i2cDumpStatus(): ack(0) sl_rw(0) to(0) arb(0) busy(0) sl(1) trans(0) rx(0) tx(0) sclMain(6) scl(6)
09:22:22.966 websocket.js:192 <--[I][esp32-hal-i2c.c:418] i2cDumpFifo(): WRITE 0x300010 0027 0000 0000 
09:22:23.072 websocket.js:192 <--endTransmission result: TIMEOUT
```

human890209, 2018-10-31:
I doubt that the Frequency Limit is relative to Cores.
Though I did get the correct results by lowering the I2C frequency. But the frequency limits seems very different by different Cores.
I use an external EEPROM module with I2C set frequency to 400k get me correct results of hundreds of bytes by Core 1. I'm using this for weeks, never find weird results.
But I read or write 4 bytes by Core 0, I got 10% wrong result at 100k. I have to use 50k to get 100% correct results. The difference between 400k and 50k is very big.
I'm wondering if there are some logic difference should be taken cared by codes?

human890209, 2018-10-31:
Thanks, @stickbreaker 
I just finish a simple test sketch to prove your theory and my experience.

ESP32 Master Sketch
```
// A Test Sketch to prove that Core 0's I2C is not as good as Core 1
// Use Serial Command:
// 0:Test I2C on Core 0
// 1:Test I2C on Core 1
// A(a): Set Frequency to 50,000
// B(b): Set Frequency to 100,000
// R(r): Set Frequency to 400,000

#include "Arduino.h"
// Add to avoid watch dog timeout on Core0
#include "soc/timer_group_struct.h"
#include "soc/timer_group_reg.h"

#include "Wire.h"

#define I2C_SLA_PIN			27
#define I2C_SDA_PIN			25
#define I2C_FREQUENCY		400000
#define I2C_SLAVE_ADDRESS	0x30

byte message[] = {1,2,3,4,5,6,7,8,9,0};
byte message_length = 10;

bool test0 = false;
bool test1 = false;

void core0( void * parameter )
{
	setup2();
	while (true)
	{
		// Add to avoid watch dog timeout on Core0
		TIMERG0.wdt_wprotect=TIMG_WDT_WKEY_VALUE;
		TIMERG0.wdt_feed=1;
		TIMERG0.wdt_wprotect=0;

		loop2();
	}
}

void setup()
{
	xTaskCreatePinnedToCore(core0, "Core0Task", 10000, NULL, 1, NULL, 0);

	Serial.begin(115200);

	Wire.begin(I2C_SDA_PIN, I2C_SLA_PIN, I2C_FREQUENCY);
	Serial.print("Frequency: ");
	Serial.println(I2C_FREQUENCY);
}

void setup2()
{

}

void loop()
{
	while (Serial.available())
	{
		char command = Serial.read();
		Serial.print("Command: ");
		Serial.println(command);
		switch (command)
		{
			case '0':
				// Test Core 0
				Serial.println("Test Core 0");
				test0 = true;
			break;
			case '1':
				// Test Core 1
				Serial.println("Test Core 1");
				test1 = true;
			break;
			case 'a':
			case 'A':
				// Set Frequency 50k
				Wire.setClock(50000);
				Serial.print("Frequency: ");
				Serial.println(50000);
			break;
			case 'b':
			case 'B':
				// Set Frequency 100k
				Wire.setClock(100000);
				Serial.print("Frequency: ");
				Serial.println(100000);
			break;
			case 'r':
			case 'R':
				// Set Frequency I2C_FREQUENCY
				Wire.setClock(I2C_FREQUENCY);
				Serial.print("Frequency: ");
				Serial.println(I2C_FREQUENCY);
			break;
			default:
				Serial.println("Command Invalid.");
			break;
		}
	}

	if (test1)
	{
		test1 = !testI2C(&Wire);
	}
}

void loop2()
{
	if (test0)
	{
		test0 = !testI2C(&Wire);
	}
}

bool testI2C(TwoWire* tw)
{
	if (tw->busy()) return false;

	// Transmit
	tw->beginTransmission((byte)I2C_SLAVE_ADDRESS);
	tw->write(message, message_length);
	byte result = tw->endTransmission();

	delay(100);

	if ( result != I2C_ERROR_OK)
	{
		Serial.print("endTransmission result: ");
		Serial.println(tw->getErrorText(result));
		return false;
	}

	// Request
	if (tw->requestFrom((byte)I2C_SLAVE_ADDRESS, message_length) != message_length)
	{
		Serial.print("requestFrom result: ");
		Serial.println(tw->getErrorText(tw->lastError()));
		return false;
	}

	while (tw->available())
	{
		Serial.print(tw->read());
		Serial.print(' ');
	}
	Serial.println();
	return true;
}
```

Arduino Nano Slave Sketch
```
// An I2C Slave for test
// Put Received bytes into a 32 bytes buffer. Send last received message on requested.

#include "Arduino.h"
#include "Wire.h"

#define ADDRESS 	0x30
#define FREQUANCY	400000

volatile byte buffer[32] = {8};
volatile byte buffer_pos = 1;
volatile bool updated = false;
volatile bool sent = false;

void setup()
{
	Serial.begin(115200);
	Serial.println("Start");

	Wire.begin(ADDRESS);
	Wire.setClock(FREQUANCY);
	Wire.onReceive(receiveCommand);
	Wire.onRequest(messageRequested);
}

void receiveCommand(int howMany)
{
	for (byte i = 0; i < howMany; i++)
	{
		buffer[i] = Wire.read();
	}
	buffer_pos = howMany;
	updated = true;
}

void messageRequested()
{
	Wire.write((byte*)buffer,buffer_pos);
	sent = true;
}

void loop()
{
	if (updated)
	{
		updated = false;
		Serial.print(F("Receive: "));
		for (byte i = 0; i < buffer_pos; i++)
		{
			if (i != 0)
			{
				Serial.print(' ');
			}
			Serial.print(buffer[i]);
		}
		Serial.println();
	}

	if (sent)
	{
		sent = false;
		Serial.print(F("Sent: "));
		for (byte i = 0; i < buffer_pos; i++)
		{
			if (i != 0)
			{
				Serial.print(' ');
			}
			Serial.print(buffer[i]);
		}
		Serial.println();
	}
}

```

human890209, 2018-10-31:
Hi, @stickbreaker 
I noticed this:
`i2cReleaseISR(i2c); // ISR exists, release it before disabling hardware`
Can I use this to release the ISR?

stickbreaker, 2018-10-31:
yes, `i2cReleaseISR()` can be used, but `i2c` is internal to the `Wire()` object. So, you will have to add another method to TwoWire().

Chuck.

human890209, 2018-10-31:
I tried to modify my TwoWireMOD which inherit from TwoWire
And I try to use:
```
void TwoWireMOD::releaseISR()
{
    if(i2c->intr_handle) {
        esp_intr_free(i2c->intr_handle);
        i2c->intr_handle=NULL;
    }
}
```
I add `#include "esp32-hal-i2c.c"`

But I got the compiler error:
```
D:\Arduino\Software\Sloeber\MyHardware\ESP32\cores\esp32/esp32-hal-i2c.c:169:8: warning: 'i2c_struct_t' has a field 'i2c_struct_t::dev' whose type uses the anonymous namespace
 struct i2c_struct_t {
        ^
D:\Arduino\Software\Sloeber\MyHardware\ESP32\cores\esp32/esp32-hal-i2c.c:215:1: error: cannot convert 'I2C_MODE_t' to 'I2C_STAGE_t' in initialization
 };
 ^
D:\Arduino\Software\Sloeber\MyHardware\ESP32\cores\esp32/esp32-hal-i2c.c:215:1: error: cannot convert 'i2c_err_t' to 'I2C_ERROR_t' in initialization
D:\Arduino\Software\Sloeber\MyHardware\ESP32\cores\esp32/esp32-hal-i2c.c:215:1: error: cannot convert 'I2C_MODE_t' to 'I2C_STAGE_t' in initialization
D:\Arduino\Software\Sloeber\MyHardware\ESP32\cores\esp32/esp32-hal-i2c.c:215:1: error: cannot convert 'i2c_err_t' to 'I2C_ERROR_t' in initialization
D:\Arduino\Software\Sloeber\MyHardware\ESP32\cores\esp32/esp32-hal-i2c.c:353:23: warning: 'void i2cDumpStatus(i2c_t*)' defined but not used [-Wunused-function]
 static void IRAM_ATTR i2cDumpStatus(i2c_t * i2c){
                       ^
In file included from D:\Arduino\Software\Sloeber\MyLibrary\WireMOD/WireMOD.h:6:0,
                 from ..\sloeber.ino.cpp:11:
D:\Arduino\Software\Sloeber\MyHardware\ESP32\cores\esp32/esp32-hal-i2c.c:383:13: warning: 'void i2cDumpFifo(i2c_t*)' defined but not used [-Wunused-function]
 static void i2cDumpFifo(i2c_t * i2c){
             ^
```
I'm not very familiar to struct.

I want to add a variable to hold which core the ISR is on and when Request or Transmit check current core and release ISR if the core is changed.

stickbreaker, 2018-10-31:
@human890209 no fast update, me-no-dev has to update it. 
you don't need to include esp32-hal-i2c.c .

if you try to release the interrupt from the wrong core you will receive an ESP_ERR_INVALID_ARG error
see: [esp-idf#2487](https://github.com/espressif/esp-idf/issues/2487). the resolution of this issue will migrate to Arduino-Esp32 when me-no-dev does the update.  For you purposes I would just design your code to destroy(~) `Wire()` before you switch cores.

I would have the ISR on Core1 with the app on Core0, Wifi is a pig on Core0.

Chuck.

stickbreaker, 2018-10-31:
@human890209 my day is over 11pm here. 

Out.

Chuck.

human890209, 2018-10-31:
Thanks,
I will use the destroy solution.

closed 2018-11-01 by human890209

human890209, 2018-11-01:
Hi, @stickbreaker 
Thanks for your help, I'm now using I2C on Core 0 as fast as on Core 1.
And

> I would have the ISR on Core1 with the app on Core0, Wifi is a pig on Core0.

I'm a new player of core 0. I'm also using ESPNow and BluetoothSerial which I guess is run on Core 0.

Does ESPNow and BT will interrupt the I2C transmit or request?
How does core 0 handle the task of ESPNow, BT, I2C, HardwareSerial, and GPIOInterrupt?
Do these functions got hardware with FIFO or buffer to receive data and queue to send?
Or they interfere with each other and make communications messy?