espressif/arduino-esp32 #2008: After reset button, it seems that my device sometimes fail to continue, when connecting wifi

robertgregor, 2018-10-30:
Here is normal working connect:

Connecting wifi...
..............................................................................[W][WiFiGeneric.cpp:360] _eventCallback(): Reason: 2 - AUTH_EXPIRE
...[I][TestAsync.ino:48] print_wakeup_reason(): 

Connected to AP in 4226ms. IP: 192.168.1.50
2018-10-30T18:15:19Z
Voltage: 3.31V
ets Jun  8 2016 00:22:57

Here I have press reset button:

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:952
load:0x40078000,len:6084
load:0x40080000,len:7936
entry 0x40080310
Connecting wifi...
E (236) boot: Assert failed in esp_err_t nvs::PageManager::requestNewPage(), /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/nvs_flash/src/nvs_pagemanager.cpp:175 (usedEntries == newPage->getUsedEntryCount())
Task watchdog got triggered. The following tasks did not reset the watchdog in time:
 - IDLE (CPU 0)
Tasks currently running:
CPU 0: wifi
CPU 1: IDLE
Task watchdog got triggered. The following tasks did not reset the watchdog in time:
 - IDLE (CPU 0)
Tasks currently running:
CPU 0: wifi
CPU 1: IDLE
Task watchdog got triggered. The following tasks did not reset the watchdog in time:
 - IDLE (CPU 0)
Tasks currently running:
CPU 0: wifi
CPU 1: IDLE
Task watchdog got triggered. The following tasks did not reset the watchdog in time:
 - IDLE (CPU 0)
Tasks currently running:
CPU 0: wifi
CPU 1: IDLE

lbernstone, 2018-10-30:
This looks like it could be a problem with your nvs partition.  Try [wiping your device](https://desire.giesecke.tk/index.php/2018/05/04/upload-of-code-over-usb-serial-fails-because-of-erroneous-code/) and reprogramming.

robertgregor, 2018-10-31:
Hi, thanks, I did:
esptool --port COM8 erase_flash
esptool.py v2.3.1
Connecting......
Detecting chip type... ESP32
Chip is ESP32D0WDQ6 (revision (unknown 0xa))
Features: WiFi, BT, Dual Core, VRef calibration in efuse
Uploading stub...
Running stub...
Stub running...
Erasing flash (this may take a while)...
Chip erase completed successfully in 12.6s
Hard resetting via RTS pin...

And it didn't happen any more. Seems your suggestion helped!!!! Thanks!!!

closed 2018-10-31 by robertgregor