espressif/arduino-esp32 #2012: ESP_LOG_BUFFER_HEX_LEVEL causes "Assert failed in heap_caps_free"

chegewara, 2018-10-31:
I am trying to print few bytes from uint8_t* variable and it perfectly works with esp-idf but its causing assert fail in arduino. Macro i am trying to use:
`ESP_LOG_BUFFER_HEX_LEVEL("TAG", val123, 4, ESP_LOG_INFO);`
Logs:
```
E (6005) boot: Assert failed in heap_caps_free, /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/heap/heap_caps.c:261 (heap != NULL && "free() target pointer is outside heap areas")
Task watchdog got triggered. The following tasks did not reset the watchdog in time:
 - IDLE (CPU 0)
Tasks currently running:
CPU 0: Btc_task  //<--- probably not important what task it is
CPU 1: IDLE
```

copercini, 2018-11-03:
Arduino core is compiled only with IDF error log level, so using ESP_LOG_INFO it will stuck in this loop forever: https://github.com/espressif/esp-idf/blob/master/components/log/include/esp_log.h#L132 triggering the watchdog timer

Man, move to https://github.com/espressif/arduino-esp32/blob/master/docs/esp-idf_component.md so you can really control IDF and arduino configs and avoid those bugs :wink:

chegewara, 2018-11-03:
@copercini 
The problem is not only `ESP_LOG_BUFFER_HEX_LEVEL`, the same issue ive got with Serial.println((char*)val) where value is uint8_t* variable. Personally i am using esp-idf, it was issue in case of upwork job, but thanks for that valuable info, now i can debug with ESP_LOG_ERROR level.

chegewara, 2018-11-03:
@copercini I thought you are nice guy, but telling someone to go away to esp-idf because is opening issues. Thats not cool.

copercini, 2018-11-03:
@chegewara oh, sorry, that's not what I want to meant

I am here as a volunteer, so I have no reward for closing issues or anything like that, 

Just telling you, as a fellow developer, that using it as IDF component give you the best of both worlds: you have the Arduino fast coding and can change IDF configs any time, also mix both codes

----
About your issue, I was searching a way to do it using arduino Serial, maybe like this:
```
uint8_t value[3] = { 0x8e, 0x73, 0xb0};
Serial.printf("0x%x 0x%x 0x%x\n", value[0], value[1], value[2]);
```
what do you think?

chegewara, 2018-11-03:
Thanks, i already found solution to print it, only because problems with esp_log i lost one day and got headache:
```
      for(int i=0;i<4;i++) {
        Serial.print((uint8_t)val[i], HEX);
      }
```

You totally answered on this issue and i agree, it can be closed. Just code i could write in less than hour with testing and example usage took me way too long.

closed 2018-11-03 by chegewara