espressif/arduino-esp32 #2014: ESP32 interrupts + timers

HFATL, 2018-11-02:
I'd like to know if the following code will work. Basically I need to have two external interrupts in two GPIOs from LOW to High and additionally two hardware timers (5 seconds and 15 seconds) which will use variables of the ISR. Is this the correct way of coding? Something like this:

volatile int Steps;
volatile int Spins;
float Rot, Delta;

const byte data1 = 39;
const byte data2 = 34;

hw_timer_t * timer1 = NULL;
hw_timer_t * timer2 = NULL;

portMUX_TYPE timerMux1 = portMUX_INITIALIZER_UNLOCKED;
portMUX_TYPE timerMux2 = portMUX_INITIALIZER_UNLOCKED;

void IRAM_ATTR onTimer1() {
  portENTER_CRITICAL_ISR(&timerMux1);
  Delta = (5/3) + 1/Steps ;
  Steps = 0;
  portEXIT_CRITICAL_ISR(&timerMux1);
 
}

void IRAM_ATTR onTimer2() {
  portENTER_CRITICAL_ISR(&timerMux2);
  Rot = Spins * Spins;
  Spins = 0;
  portEXIT_CRITICAL_ISR(&timerMux2);
 
}

 void IRAM_ATTR handleInterrupt1() {
  portENTER_CRITICAL_ISR(&timerMux1);
  Steps++;
  portEXIT_CRITICAL_ISR(&timerMux1);
}

 void IRAM_ATTR handleInterrupt2() {
  portENTER_CRITICAL_ISR(&timerMux2);
  Spins++;
  portEXIT_CRITICAL_ISR(&timerMux2);
}

void setup() {
 
  Serial.begin(115200);
 
 
  pinMode(data1, INPUT);
  attachInterrupt(digitalPinToInterrupt(data1), handleInterrupt1, RISING);
  pinMode(data2, INPUT);
  attachInterrupt(digitalPinToInterrupt(data2), handleInterrupt2, RISING);
  
  timer1 = timerBegin(0, 80, true);
  timerAttachInterrupt(timer1, &onTimer, true);
  timerAlarmWrite(timer1, 5000000, true);
  timerAlarmEnable(timer1);
 
  timer2 = timerBegin(0, 80, true);
  timerAttachInterrupt(timer2, &onTimer, true);
  timerAlarmWrite(timer2, 15000000, true);
  timerAlarmEnable(timer2);
}
 
void loop() {
 
 Serial.println(Delta);
 Serial.println(Rot);
 
}

Many thanks for your support!

HFATL, 2018-11-05:
Any help please?

Thank you

everslick, 2018-11-06:
This is the ESP32 Arduino Core issue tracker. It is here to report and discuss bugs in the parent repository. Please do not misuse it as a help forum. For general discussion please visit the appropriate gitter channels and/or the espressif user forum.

cyberman54, 2019-03-24:
@me-no-dev This should be closed, since it's not issue.

closed 2019-03-24 by everslick