espressif/arduino-esp32 #2016: WiFiClient.cpp::setTimeout not working properly?

cyberman54, 2018-11-04:
I'm trying to set a connection timeout before opening a wifi connection.
My code is:

```
const uint32_t RESPONSE_TIMEOUT = 30;
WiFiClientSecure client;
client.setTimeout(RESPONSE_TIMEOUT);
if (!client.connect(currentHost.c_str(), port)) {
    ESP_LOGI(TAG, "Cannot connect to %s", currentHost.c_str());
    display(3, " E", "connection lost");
    goto failure;
  }
```
This openes the wifi connection successful, but it logs this error:
[E][WiFiClient.cpp:236] setSocketOption(): 1006 : 9

0x1006 is defined as
`#define SO_RCVTIMEO    0x1006 /* receive timeout */`

but what means error 9 ?

cyberman54, 2018-11-04:
My fault: i set the client.setTimeout command at wrong place before opening the stream.
Closing this.

Jeroen88, 2018-11-05:
As I understand it, the timeout is not used for the connection, but for reading and writing only. So you should use client.setTimeout() _after_ client.connect().

I may have found the error code here, though: http://www.nongnu.org/lwip/2_0_x/group__infrastructure__errors.html

Beware, the timeout for WiFiClient (ie effectively what is called using WiFiClientSecure.setTimeout()) is **not** passed on to Stream, causing timeout errors while reading. See my comments at PR #1979.

cyberman54, 2018-11-05:
Thanks for linking your RP #1979 , it explains exactly the problems in which i ran.
So again i close this issue, hoping that PR #1979 will be merged soon.

closed 2018-11-05 by cyberman54

Jeroen88, 2018-11-05:
You're welcome :), you can use this PR by checking it out locally on your computer

cyberman54, 2018-11-05:
Did that already. But my goal is not to keep a local update.cpp at all...

Jeroen88, 2018-11-05:
Sure, that is why I want it to be merged soon too!

usmanshahid001, 2019-09-30:
int connect(IPAddress ip, uint16_t port, int32_t timeout);
i think connection timeout is 3rd parameter .