espressif/arduino-esp32 #2020: WebServer

MartinDavidWaller, 2018-11-05:
Hi,

I've recently moved some code from the NodeMCU environment to the ESP32 environment and I've found something very odd with the WebServer code. I've refreshed my sources from today and it still fails.

I have an index.html file stored in SPIFFS that contains the header:

<head>
    <link rel="stylesheet" type="text/css" href="styles.css" media="all" />
    <script type="text/javascript" src="jquery-3.2.1.min.js"></script>

The server code makes use of the standard pattern:

    server.onNotFound([]() {    

      // Try to send the file requested
      
      if (!handleFileRead(server.uri())) {

        // We have not located the file, return not found.
        
        server.send(404, "text/plain", "404: Not Found");  
      }
    });

and the handlerFileRead method starts with:

bool handleFileRead(String path) {

  // A request has come in for a file, we can process that here.

  Serial.println("");
  Serial.println("File requested:" + path);

This all seems to work fine, the index.html file is served as expected and the styles.css file is severed as expected. The jquery-3.2.1.min.js request is never server when requested via the index.html page? Seemingly the request never gets as far as the handlerFileRead method.

However, if I request the jquery-3.2.1.min.js file directly then it works.

It's almost as if it can't handle all the requests from the index.html page. Using Chrome to try and debug this I can see the request going out correctly from index.html but Chrome shows no response?

I wonder if anyone else has similar issues. This worked fine on the NodeMCU device.

Any help / guidance much appreciated.

Martin

MartinDavidWaller, 2018-11-06:
[ESP32Failure.zip](https://github.com/espressif/arduino-esp32/files/2552432/ESP32Failure.zip)

Hi, the attached Sketch will demonstrate the issue. You will need to upload the data folder to the SPIFFS and then connect to the SSID "Hello" and browse to 192.168.4.1. If you then look at the output and the developer console in Chrome you'll see the failure to download jquery...

Martin

MartinDavidWaller, 2018-11-06:
The chip being used is the ESP32 chip with 2 CPU cores, WiFi/BT/BLE, silicon revision 1, 4MB external flash.

tfry-git, 2019-03-12:
I suspect this may be the same issue as #1921 . Likely the browser requests the .js file, while index.html is still loading. That will currently not work. You could try moving the <script>-tag all the way to the end of index.html (it's allowed anywhere in HTML), perhaps that will defer the loading long enough.

PilnyTomas, 2022-03-31:
Hello @MartinDavidWaller .
There has been a recent update (v 2.0.3-RC1) which should fix lots of WiFi issues.
I tried your sketch with the version to it seems ok.
Could you please try it out and let us know?

VojtechBartoska, 2022-04-11:
Ping

VojtechBartoska, 2022-05-04:
Closing as solved, if you still face the issue, please reopen it. Thanks for understanding.

closed 2022-05-04 by VojtechBartoska