espressif/arduino-esp32 #2021: esp_log_set_vprintf doesnt process correctly the log messages from sketch

robertgregor, 2018-11-05:
Hi, I want to redirect the logging to the internal buffer to be available for the webpage instead of serial port.

There is a function in the esp-idf - esp_log_set_vprintf. It works, however, it doesn't work for the messages in arduino sketch, only the idf messages are displayed...

I have it like that:

#include "esp_log.h"
#include "esp_system.h"

int buffer_logging_vprintf( const char *str, va_list l ) {
  Serial.println("aaaaaaa");
}

void setup(){
  esp_log_level_set("*", ESP_LOG_VERBOSE);
  esp_log_set_vprintf(buffer_logging_vprintf);
  Serial.begin(115200);
  log_i("Infoooooooo");
}

I expected output like only the aaaaaaa. However, the real output looks like:

[I][TestAsync.ino:316] setup(): Infoooooooo
[I][TestAsync.ino:102] connectToWifi(): Connecting wifi...
aaaaaaa
aaaaaaa
aaaaaaa
aaaaaaa
aaaaaaa
aaaaaaa
aaaaaaa
aaaaaaa
aaaaaaa
aaaaaaa
aaaaaaa
aaaaaaa
aaaaaaa
aaaaaaa
aaaaaaa
aaaaaaa
aaaaaaa
[W][WiFiGeneric.cpp:360] _eventCallback(): Reason: 2 - AUTH_EXPIRE
aaaaaaa
aaaaaaa
aaaaaaa
aaaaaaa
aaaaaaa
aaaaaaa
aaaaaaa
aaaaaaa
aaaaaaa
aaaaaaa
aaaaaaa
[I][TestAsync.ino:54] print_wakeup_reason(): Wakeup was not caused by deep sleep
[I][TestAsync.ino:338] setup(): Connected to AP in 4281 ms. IP: 192.168.1.50
[I][TestAsync.ino:342] setup(): Module time: 2018-11-06 00:04:48
[I][TestAsync.ino:348] setup(): Voltage: 3.32V
[I][TestAsync.ino:350] setup(): Starting Web server...

So as is seen, the idf messages are redirected correctly, but not the messages from sketch.

Any idea?

Robert

robertgregor, 2018-11-06:
I see why:
In fact the preprocessor will replace the esp_idf log_i with log_printf( "[" "I" "][%s:%u] %s()...........
I don't think, that this is correct. Why it is not replaced by the esp_idf function - log_i and thus, it will have the same format and same functionality like the other classes???

GeorgeFlorian, 2019-03-18:
Hello ! I've been trying to use `esp_log_set_vprintf` for the past week and I can't make it work.
I would like to redirect the log messages to a SPIFFS file and then to read them from that file.

Did you fix this ? Did you manage to use `esp_log_set_vprintf` to redirect your log messages ?

Thank you !

closed 2019-08-15 by stale[bot]

letmeknowFR, 2023-09-03:
I have the same issue, the only solution I have found is to use the esp_log_write to redirect the message