espressif/arduino-esp32 #2024: SD Card module throws Guru Meditation Error: Core 1 panic'ed (Load Prohibited)

sheikhumar93, 2018-11-06:
### Hardware:
IDE name:						Arduino IDE
Flash Frequency:					80Mhz
Upload Speed:				          115200
Computer OS:                                            macOS Mojave 10.14.0

### Description:
I have a MPU6050 and a microSD Card module connected to an ESP32. The MPU6050 runs at 20Hz and should save these values to the microSD card connected. I am getting variable mileage with this sometimes the values get saved for a couple of hours before the Guru Meditation Error occurs, sometimes it happens in a minute or two before the error happens and the ESP32 restarts. I only have access to Wifi when I first turn on the device to set the time on the esp32, if it restarts at any point in the future due to this error, Wifi is not available. I have tried pasting in the stack in EspExceptionDecoder but the decoder seems to be stuck at "Decoding stack results". I think its a issue with the append code to the txt file because without the card the MPU6050 works perfectly even for long periods of time. Any help would be great :)

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp

//Change the code below by your sketch
#include <WiFi.h>
#include <DNSServer.h>
#include <WiFiClient.h>
#include <WiFiUdp.h>
#include <OSCMessage.h>
#include <WiFiManager.h>         //https://github.com/tzapu/WiFiManager
#include "FS.h"
#include "SD.h"
#include "SPI.h"
#include "I2Cdev.h"

#include "MPU6050_6Axis_MotionApps20.h"

// Arduino Wire library is required if I2Cdev I2CDEV_ARDUINO_WIRE implementation
// is used in I2Cdev.h
#if I2CDEV_IMPLEMENTATION == I2CDEV_ARDUINO_WIRE
    #include "Wire.h"
#endif

// class default I2C address is 0x68
// specific I2C addresses may be passed as a parameter here
// AD0 low = 0x68 (default for SparkFun breakout and InvenSense evaluation board)
// AD0 high = 0x69
MPU6050 mpu;
//MPU6050 mpu(0x69); // <-- use for AD0 high

// MPU control/status vars
bool dmpReady = false;  // set true if DMP init was successful
uint8_t mpuIntStatus;   // holds actual interrupt status byte from MPU
uint8_t devStatus;      // return status after each device operation (0 = success, !0 = error)
uint16_t packetSize;    // expected DMP packet size (default is 42 bytes)
uint16_t fifoCount;     // count of all bytes currently in FIFO
uint8_t fifoBuffer[64]; // FIFO storage buffer

int16_t ax, ay, az, gx, gy, gz;

#define INTERRUPT_PIN 2 // use pin 15 on ESP8266

const char DEVICE_NAME[] = "mpu6050";

WiFiUDP udp;                                // A UDP instance to let us send and receive packets over UDP
const IPAddress outIp(192, 168, 8, 151);     // remote IP to receive OSC
const unsigned int outPort = 3333;          // remote port to receive OSC
const char * udpAddress = "192.168.8.150";
const int udpPort = 3333;

const char* NTP_SERVER = "pool.ntp.org";
const char* TZ_INFO    = "PKT-5";
struct tm timeinfo;

void appendFile(fs::FS &fs, const char * path, const char * message){
    File file = fs.open(path, FILE_APPEND);
    if(!file){
        appendFile(SD, "/log.txt", "Failed to open file for appending\n");
        Serial.println("Failed to open file for appending");
        return;
    }
    if(!file.print(message)){
        appendFile(SD, "/log.txt", "Append failed\n");
        Serial.println("Append failed");
    }
    file.close();
}

volatile bool mpuInterrupt = false;     // indicates whether MPU interrupt pin has gone high
void dmpDataReady() {
    mpuInterrupt = true;
}

void mpu_setup()
{
  // join I2C bus (I2Cdev library doesn't do this automatically)
#if I2CDEV_IMPLEMENTATION == I2CDEV_ARDUINO_WIRE
  Wire.begin();
  Wire.setClock(400000); // 400kHz I2C clock. Comment this line if having compilation difficulties
#elif I2CDEV_IMPLEMENTATION == I2CDEV_BUILTIN_FASTWIRE
  Fastwire::setup(400, true);
#endif

  // initialize device
  Serial.println(F("Initializing I2C devices..."));
  mpu.initialize();
  pinMode(INTERRUPT_PIN, INPUT);

  // verify connection
  Serial.println(F("Testing device connections..."));
  Serial.println(mpu.testConnection() ? F("MPU6050 connection successful") : F("MPU6050 connection failed"));

  // load and configure the DMP
  Serial.println(F("Initializing DMP..."));
  devStatus = mpu.dmpInitialize();
  mpu.setXAccelOffset(-714);
  mpu.setYAccelOffset(-1882);
  mpu.setZAccelOffset(1303); // 1688 factory default for my test chip

  // make sure it worked (returns 0 if so)
  if (devStatus == 0) {
    // turn on the DMP, now that it's ready
    Serial.println(F("Enabling DMP..."));
    mpu.setDMPEnabled(true);

    // enable Arduino interrupt detection
    Serial.println(F("Enabling interrupt detection (Arduino external interrupt 0)..."));
    attachInterrupt(digitalPinToInterrupt(INTERRUPT_PIN), dmpDataReady, RISING);
    mpuIntStatus = mpu.getIntStatus();

    // set our DMP Ready flag so the main loop() function knows it's okay to use it
    Serial.println(F("DMP ready! Waiting for first interrupt..."));
    dmpReady = true;

    // get expected DMP packet size for later comparison
    packetSize = mpu.dmpGetFIFOPacketSize();
  } else {
    // ERROR!
    // 1 = initial memory load failed
    // 2 = DMP configuration updates failed
    // (if it's going to break, usually the code will be 1)
    Serial.print(F("DMP Initialization failed (code "));
    Serial.print(devStatus);
    Serial.println(F(")"));
  }
}

void setup(void)
{
  Serial.begin(115200);
  
  if(!SD.begin()){
      Serial.println("Card Mount Failed");
      return;
  }
  uint8_t cardType = SD.cardType();

  if(cardType == CARD_NONE){
      Serial.println("No SD card attached");
      return;
  }
  WiFiManager wifiManager;

  if(!wifiManager.autoConnect(DEVICE_NAME)) {
    appendFile(SD, "/log.txt", "ESP restart called\n");
    Serial.println("failed to connect and hit timeout");
    delay(3000);
    //reset and try again, or maybe put it to deep sleep
    ESP.restart();
    delay(5000);
  } 

  Serial.print(F("WiFi connected! IP address: "));
  Serial.println(WiFi.localIP());

  configTzTime(TZ_INFO, NTP_SERVER);

  if (getLocalTime(&timeinfo, 20000)) {  // wait up to 10sec to sync
    Serial.println(&timeinfo, "Time set: %B %d %Y %H:%M:%S (%A)");
    WiFi.mode(WIFI_OFF);
    btStop();
  } else {
    appendFile(SD, "/log.txt", "Time not set");
    Serial.println("Time not set");
  }
  
  mpu_setup();
}

void mpu_loop()
{
  // if programming failed, don't try to do anything
  if (!dmpReady) return;

  // wait for MPU interrupt or extra packet(s) available
  if (!mpuInterrupt && fifoCount < packetSize) return;

  // reset interrupt flag and get INT_STATUS byte
  mpuInterrupt = false;
  mpuIntStatus = mpu.getIntStatus();

  // get current FIFO count
  fifoCount = mpu.getFIFOCount();

  // check for overflow (this should never happen unless our code is too inefficient)
  if ((mpuIntStatus & 0x10) || fifoCount == 1024) {
    // reset so we can continue cleanly
    mpu.resetFIFO();
    appendFile(SD, "/log.txt", "FIFO overflow!\n");
    Serial.println(F("FIFO overflow!"));

    // otherwise, check for DMP data ready interrupt (this should happen frequently)
  } else if (mpuIntStatus & 0x02) {
    // wait for correct available data length, should be a VERY short wait
    while (fifoCount < packetSize) fifoCount = mpu.getFIFOCount();

    // read a packet from FIFO
    mpu.getFIFOBytes(fifoBuffer, packetSize);

    // track FIFO count here in case there is > 1 packet available
    // (this lets us immediately read more without waiting for an interrupt)
    fifoCount -= packetSize;

    mpu.getMotion6(&ax, &ay, &az, &gx, &gy, &gz);
    String dataString = "";
    time_t now;
    time(&now);
    dataString += now; dataString += ",";

    dataString += ax; dataString += ","; dataString += ay; dataString += ","; dataString += az; dataString += "\n";
    appendFile(SD, "/accel.txt", dataString.c_str());
    Serial.print(dataString);
  }
}
void loop(void)
{
  mpu_loop();
}
```

### Debug Messages:
```
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x400e38c4  PS      : 0x00060030  A0      : 0x800e38f8  A1      : 0x3ffcf820  
A2      : 0x0000001f  A3      : 0x3f402009  A4      : 0x00060e20  A5      : 0x3ffd3460  
A6      : 0x0000000a  A7      : 0x3ffdc540  A8      : 0x800e38c4  A9      : 0x3ffcf810  
A10     : 0x00000005  A11     : 0x3f40200e  A12     : 0x000000ff  A13     : 0x0000ff00  
A14     : 0x00ff0000  A15     : 0xff000000  SAR     : 0x00000010  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x0000001f  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff  

Backtrace: 0x400e38c4:0x3ffcf820 0x400e38f5:0x3ffcf840 0x400d9251:0x3ffcf860 0x400d945d:0x3ffcf8a0 0x400d5706:0x3ffcf8c0 0x400d401d:0x3ffcf910 0x400d40c5:0x3ffcfa00

Rebooting...
```

sheikhumar93, 2018-11-07:
The exception decoder was not working before but now it started working. Here you go :)
```
PC: 0x400e39f4: Print::write(char const*) at /Users/umar/Library/Arduino15/packages/esp32/hardware/esp32/1.0.0/cores/esp32/Print.h line 66
EXCVADDR: 0x0000001f

Decoding stack results
0x400e39f4: Print::write(char const*) at /Users/umar/Library/Arduino15/packages/esp32/hardware/esp32/1.0.0/cores/esp32/Print.h line 66
0x400e3ab1: Print::print(char const*) at /Users/umar/Library/Arduino15/packages/esp32/hardware/esp32/1.0.0/cores/esp32/Print.cpp line 84
0x400d93a5: WiFiManager::DEBUG_WM (WiFiManager::wm_debuglevel_t, char const*, system_event_id_t) at /Users/umar/Documents/Arduino/libraries/WiFiManager/WiFiManager.cpp line 2280
0x400d95b1: WiFiManager::WiFiEvent(system_event_id_t, system_event_info_t) at /Users/umar/Documents/Arduino/libraries/WiFiManager/WiFiManager.cpp line 2571
0x400d585a: std::_Function_handler    (WiFiManager*, std::_Placeholder1>, std::_Placeholder2>)> >::_M_invoke(std::_Any_data const&, system_event_id_t&&, system_event_info_t&&) at /Users/umar/Library/Arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/1.22.0-80-g6c4433a-5.2.0/xtensa-esp32-elf/include/c++/5.2.0/functional line 600
0x400d4171: WiFiGenericClass::_eventCallback(void*, system_event_t*) at /Users/umar/Library/Arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/1.22.0-80-g6c4433a-5.2.0/xtensa-esp32-elf/include/c++/5.2.0/functional line 2271
0x400d4219: _network_event_task(void*) at /Users/umar/Library/Arduino15/packages/esp32/hardware/esp32/1.0.0/libraries/WiFi/src/WiFiGeneric.cpp line 70
```

lbernstone, 2018-11-07:
You need to narrow down what line is throwing the exception.  Also, print ESP.getFreeHeap() in the Append function and see if you are leaking memory somewhere.

vanvince, 2018-11-13:
Hi @sheikhumar93! 
I have a very similar issue. I want to write some IMU-data onto my sd card. 
At the start I create the log file and everything work but, when I have the imu data from Bosch BNO055, I can't write anything! It turns out a Guru Meditation Error: Core 1 panic'ed (LoadProhibited) I really don't understand why!! 
As I can see you have the same my issue  in the "decoding stack results",  a "Print:write (const char*) ..." 
Is It an error inside of Print library? Did you solve yout problem?

P.S. If you want to see my issue here is the link #1902 . My case is different in one thing, I have different RealTime Tasks.

closed 2019-08-15 by stale[bot]