espressif/arduino-esp32 #2025: WiFi.softAPI() give Core  0 panic'ed (InstrFetchProhibited)

MauroMombelli, 2018-11-06:
### Hardware:
Board:							LOLIN32
Core Installation/update date:			2018/11/06
IDE name:							Arduino IDE 1.8.7
Flash Frequency:					80Mhz
PSRAM enabled:                                          no? (not sure what is this)
Upload Speed:						115200
Computer OS:                                            Arch linux 64 bit

### Description:
I want to write a little sketch that make the ESP32 (LOLIN32) run in AP mode, and get its slef-assigned IP so i can print it on the serial; I need it because later i want to implement a TCP server

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp

  #include "WiFi.h"
   
  const char *ssid = "MyESP32AP";
  const char *password = "testpassword";
   
  void setup() {
   
    Serial.begin(115200);
    WiFi.softAP(ssid, password);
   
    Serial.println();
    Serial.print("IP address: ");
    Serial.println(WiFi.softAPIP());
   
  }
   
  void loop() {}
```

### Debug Messages:

i selected the ESP dev module and then set the core debug level to debug, unfortunately the exception does not change and seems it cant be decoded:
```
Guru Meditation Error: Core  0 panic'ed (InstrFetchProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x00000000  PS      : 0x00060630  A0      : 0x801025dd  A1      : 0x3ffb7010  
A2      : 0x00000000  A3      : 0x00000037  A4      : 0xffffffca  A5      : 0x3ffb7198  
A6      : 0x0000000c  A7      : 0x00000000  A8      : 0x80112960  A9      : 0x3ffb7000  
A10     : 0x3ffba1f0  A11     : 0x00000037  A12     : 0xffffffca  A13     : 0x3ffb70a4  
A14     : 0x6574a61c  A15     : 0x6574a61c  SAR     : 0x00000010  EXCCAUSE: 0x00000014  
EXCVADDR: 0x00000000  LBEG    : 0x4000c349  LEND    : 0x4000c36b  LCOUNT  : 0xffffffff  

Backtrace: 0x00000000:0x3ffb7010 0x401025da:0x3ffb7030 0x4010389f:0x3ffb7110 0x401211d9:0x3ffb7130 0x40088a72:0x3ffb7180

```

trying to decode return:
```
PC: 0x00000000
EXCVADDR: 0x00000000

Decoding stack results

```

lbernstone, 2018-11-07:
Please try decoding again.  This is not a valid decode.
I am unable to reproduce.  Try putting a delay(1000) in after the softAP method to see if it is just not yet prepared.

MauroMombelli, 2018-11-10:
Just tried again, (ESP32 DEV MODULE) same identic error.
Interesting i put a delay(10000), but the chip reset after 1-3 seconds

I attached my build files.

Wat arduino version are you using? maybe the difference lies in the included toolchain..

[arduino_build_282362.zip](https://github.com/espressif/arduino-esp32/files/2568799/arduino_build_282362.zip)

treii28, 2019-03-28:
getting a similar error:

`E (44363) event: mismatch or invalid event, id=63
E (44365) event: default event handler failed!
dhcps: send_offer>>udp_sendto result 0
Guru Meditation Error: Core  0 panic'ed (InstrFetchProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x00000000  PS      : 0x00060330  A0      : 0x8010c6a9  A1      : 0x3ffb4e20  
A2      : 0x3ffcfd5c  A3      : 0x3ffcff90  A4      : 0x3ffbaa80  A5      : 0x3ffba9fc  
A6      : 0x09080808  A7      : 0x13080808  A8      : 0x8010c54c  A9      : 0x3ffb4de0  
A10     : 0x3ffcfd6c  A11     : 0x3ffcff90  A12     : 0x3ffb4e2c  A13     : 0x00000044  
A14     : 0x00000001  A15     : 0x00000006  SAR     : 0x00000010  EXCCAUSE: 0x00000014  
EXCVADDR: 0x00000000  LBEG    : 0x4000c349  LEND    : 0x4000c36b  LCOUNT  : 0x00000000  

Backtrace: 0x00000000:0x3ffb4e20 0x4010c6a6:0x3ffb4e60 0x40119185:0x3ffb4e80 0x4011cb1d:0x3ffb4ec0 0x40121d52:0x3ffb4ee0 0x4010f683:0x3ffb4f00 0x4008f2ad:0x3ffb4f30
`

AdrienFelon, 2019-05-08:
Getting a similar error as well (with release 1.0.2)

```[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 15 - AP_STACONNECTED
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
Guru Meditation Error: Core  0 panic'ed (InstrFetchProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x00000000  PS      : 0x00060e30  A0      : 0x80154151  A1      : 0x3ffdc690
A2      : 0x3ffe5cd0  A3      : 0x3ffe5f04  A4      : 0x3ffafa80  A5      : 0x3ffe3dd4
A6      : 0x0204a8c0  A7      : 0x0c04a8c0  A8      : 0x80153ff0  A9      : 0x3ffdc650
A10     : 0x3ffe5ce0  A11     : 0x3ffe5f04  A12     : 0x3ffdc69c  A13     : 0x00000044
A14     : 0x00000001  A15     : 0x00000006  SAR     : 0x00000010  EXCCAUSE: 0x00000014
EXCVADDR: 0x00000000  LBEG    : 0x4000c349  LEND    : 0x4000c36b  LCOUNT  : 0x00000000

Backtrace: 0x00000000:0x3ffdc690 0x4015414e:0x3ffdc6d0 0x40160331:0x3ffdc6f0 0x40163ec9:0x3ffdc730 0x401691d6:0x3ffdc750 0x401576db:0x3ffdc770 0x40089499:0x3ffdc7a0
```

Which decodes to:

```
0x4015414e: handle_dhcp at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/apps/dhcpserver/dhcpserver.c line 1031
0x40160331: udp_input at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/lwip/src/core/udp.c line 401
0x40163ec9: ip4_input at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/lwip/src/core/ipv4/ip4.c line 740
0x401691d6: ethernet_input at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/lwip/src/netif/ethernet.c line 184
0x401576db: tcpip_thread at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/lwip/src/api/tcpip.c line 483
0x40089499: vPortTaskWrapper at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/port.c line 355 (discriminator 1)
```

AdrienFelon, 2019-05-09:
FYI, found that taking out the call to `WiFiAPClass::softAPConfig(IPAddress local_ip, IPAddress gateway, IPAddress subnet)` made the crash go away

watarat, 2019-05-18:
I can confirm that not calling WiFi.softAPConfig(ip, ip, netmask); does(can??) stop the reset upon WiFi connecting. It can however work perfectly with the call active, so there is something strange going on.

In my case:
I have 4 RTOS tasks, (ADC,Bluetooth,SD_MMC & Wifi) - with all 4 running the problem exists and commenting out softAPConfig() fixes it.
Not starting the Bluetooth task - the problem doesn't exist, and I can reconfigure the IP address from the default using softAPConfig();
I'm using Arduino, V1.0.2, and Partition of 3MB no OTA to fit it all in. (&WiFi is highest priority task!)
Thankyou @AdrienFelon

xdraylin, 2019-05-23:
I get the same error:

```
dhcps: send_offer>>udp_sendto result 0
Guru Meditation Error: Core  0 panic'ed (InstrFetchProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x00000000  PS      : 0x00060930  A0      : 0x8019e4ad  A1      : 0x3ffdeae0  
A2      : 0x3ffe8b00  A3      : 0x3ffe9934  A4      : 0x3ffe8860  A5      : 0x3ffe8840  
A6      : 0x027ba8c0  A7      : 0x0c7ba8c0  A8      : 0x8019e34c  A9      : 0x3ffdeaa0  
A10     : 0x3ffe8b10  A11     : 0x3ffe9934  A12     : 0x3ffdeaec  A13     : 0x00000044  
A14     : 0x00000001  A15     : 0x00000006  SAR     : 0x00000010  EXCCAUSE: 0x00000014  
EXCVADDR: 0x00000000  LBEG    : 0x4000c349  LEND    : 0x4000c36b  LCOUNT  : 0x00000000  

Backtrace: 0x00000000:0x3ffdeae0 0x4019e4aa:0x3ffdeb20 0x40194891:0x3ffdeb40 0x40198265:0x3ffdeb80 0x4019d4e2:0x3ffdeba0 0x4018bda3:0x3ffdebc0 0x40093531:0x3ffdebf0
```
which decodes to:
```
Exception Cause: Not found

0x4000c349: ?? ??:0
0x4000c36b: ?? ??:0
0x4019e4aa: handle_dhcp at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/apps/dhcpserver/dhcpserver.c:1031
0x40194891: udp_input at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/lwip/src/core/udp.c:401
0x40198265: ip4_input at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/lwip/src/core/ipv4/ip4.c:740
0x4019d4e2: ethernet_input at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/lwip/src/netif/ethernet.c:184
0x4018bda3: tcpip_thread at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/lwip/src/api/tcpip.c:483
0x40093531: vPortTaskWrapper at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/port.c:355 (discriminator 1)
```
Somehow this error does not occur when setting the flag `CORE_DEBUG_LEVEL=ARDUHAL_LOG_LEVEL_VERBOSE` so I think this could be a timing related issue.

zuqualla, 2019-06-06:
very similar error...
```
send_offer>>udp_sendto result 0
Guru Meditation Error: Core  0 panic'ed (InstrFetchProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x00000000  PS      : 0x00060e30  A0      : 0x800ffeed  A1      : 0x3ffb3d30  
A2      : 0x3ffc9e68  A3      : 0x3ffca09c  A4      : 0x3ffb6374  A5      : 0x3ffbd548  
A6      : 0x0204a8c0  A7      : 0x6504a8c0  A8      : 0x800ffd90  A9      : 0x3ffb3cf0  
A10     : 0x3ffc9e78  A11     : 0x3ffca09c  A12     : 0x3ffb3d3c  A13     : 0x00000044  
A14     : 0x00000001  A15     : 0x00000006  SAR     : 0x00000010  EXCCAUSE: 0x00000014  
EXCVADDR: 0x00000000  LBEG    : 0x4000c349  LEND    : 0x4000c36b  LCOUNT  : 0x00000000  

Backtrace: 0x00000000:0x3ffb3d30 0x400ffeea:0x3ffb3d70 0x4010a5f1:0x3ffb3d90 0x4010df89:0x3ffb3dd0 0x40113702:0x3ffb3df0 0x40102a8f:0x3ffb3e10 0x4008877d:0x3ffb3e40
```
translates to 
```
PC: 0x00000000
EXCVADDR: 0x00000000

Decoding stack results
0x400ffeea: handle_dhcp at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/apps/dhcpserver/dhcpserver.c line 1031
0x4010a5f1: udp_input at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/lwip/src/core/udp.c line 401
0x4010df89: ip4_input at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/lwip/src/core/ipv4/ip4.c line 740
0x40113702: ethernet_input at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/lwip/src/netif/ethernet.c line 184
0x40102a8f: tcpip_thread at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/lwip/src/api/tcpip.c line 135
0x4008877d: vPortTaskWrapper at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/port.c line 143
```
when i change
`WiFi.softAP(WifiAPName.c_str(), "password");`
to
`WiFi.softAP(WifiAPName.c_str());`
the kernel panic disappears, weird! 

ps: i have a `delay(2000);` after the above statement, changing the time value didn't solve the error.

easyvolts, 2019-06-11:
Hi,
I have some similar problem:

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:5992
load:0x40078000,len:8800
ho 0 tail 12 room 4
load:0x40080400,len:6380
entry 0x4008072c
�[0;32mI (31) boot: ESP-IDF v3.2-103-g4aa1058e8 2nd stage bootloader�[0m
�[0;32mI (31) boot: compile time 12:23:23�[0m
�[0;32mI (31) boot: Enabling RNG early entropy source...�[0m
�[0;32mI (37) boot: SPI Speed : 40MHz�[0m
�[0;32mI (41) boot: SPI Mode : DIO�[0m
�[0;32mI (45) boot: SPI Flash Size : 4MB�[0m
�[0;32mI (49) boot: Partition Table:�[0m
�[0;32mI (52) boot: ## Label Usage Type ST Offset Length�[0m
�[0;32mI (60) boot: 0 nvs WiFi data 01 02 00009000 00005000�[0m
�[0;32mI (67) boot: 1 otadata OTA data 01 00 0000e000 00002000�[0m
�[0;32mI (75) boot: 2 app0 OTA app 00 10 00010000 00140000�[0m
�[0;32mI (82) boot: 3 app1 OTA app 00 11 00150000 00140000�[0m
�[0;32mI (90) boot: 4 eeprom Unknown data 01 99 00290000 00001000�[0m
�[0;32mI (97) boot: 5 spiffs Unknown data 01 82 00291000 0016f000�[0m
�[0;32mI (105) boot: End of partition table�[0m
�[0;32mI (109) boot: No factory image, trying OTA 0�[0m
�[0;32mI (114) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x394f4 (234740) map�[0m
�[0;32mI (204) esp_image: segment 1: paddr=0x0004951c vaddr=0x3ffb0000 size=0x030b0 ( 12464) load�[0m
�[0;32mI (209) esp_image: segment 2: paddr=0x0004c5d4 vaddr=0x40080000 size=0x00400 ( 1024) load�[0m
�[0;32mI (211) esp_image: segment 3: paddr=0x0004c9dc vaddr=0x40080400 size=0x03634 ( 13876) load�[0m
�[0;32mI (225) esp_image: segment 4: paddr=0x00050018 vaddr=0x400d0018 size=0x9c8b4 (641204) map�[0m
�[0;32mI (451) esp_image: segment 5: paddr=0x000ec8d4 vaddr=0x40083a34 size=0x0d4cc ( 54476) load�[0m
�[0;32mI (472) esp_image: segment 6: paddr=0x000f9da8 vaddr=0x400c0000 size=0x00064 ( 100) load�[0m
�[0;32mI (473) esp_image: segment 7: paddr=0x000f9e14 vaddr=0x50000000 size=0x00d84 ( 3460) load�[0m
�[0;32mI (492) boot: Loaded app from partition at offset 0x10000�[0m
�[0;32mI (492) boot: Disabling RNG early entropy source...�[0m
�[0;32mI (493) cpu_start: Pro cpu up.�[0m
�[0;32mI (497) cpu_start: Starting app cpu, entry point is 0x4008117c�[0m
�[0;32mI (0) cpu_start: App cpu up.�[0m
�[0;32mI (507) heap_init: Initializing. RAM available for dynamic allocation:�[0m
�[0;32mI (514) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM�[0m
�[0;32mI (520) heap_init: At 3FFBA220 len 00025DE0 (151 KiB): DRAM�[0m
�[0;32mI (526) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM�[0m
�[0;32mI (533) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM�[0m
�[0;32mI (539) heap_init: At 40090F00 len 0000F100 (60 KiB): IRAM�[0m
�[0;32mI (545) cpu_start: Pro cpu start user code�[0m
�[0;32mI (138) cpu_start: Starting scheduler on PRO CPU.�[0m
�[0;32mI (0) cpu_start: Starting scheduler on APP CPU.�[0m

Free heap in esp32 (bytes): 311924
12[E][Preferences.cpp:38] begin(): nvs_open failed: NOT_FOUND
Battery voltage(mV): 4024
I (550) wifi: wifi driver task: 3ffc70f8, prio:23, stack:3584, core=0
I (1240) wifi: wifi firmware version: 2af77cc
I (1240) wifi: config NVS flash: enabled
I (1240) wifi: config nano formating: disabled
�[0;32mI (1250) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE�[0m
�[0;32mI (1260) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE�[0m
I (1280) wifi: Init dynamic tx buffer num: 32
I (1280) wifi: Init data frame dynamic rx buffer num: 16
I (1280) wifi: Init management frame dynamic rx buffer num: 16
I (1290) wifi: Init management short buffer num: 32
I (1290) wifi: Init static rx buffer size: 1600
I (1290) wifi: Init static rx buffer num: 16
I (1300) wifi: Init dynamic rx buffer num: 16
�[0;32mI (1370) phy: phy_version: 4008, 544f89f, Jan 24 2019, 14:54:06, 0, 0�[0m
I (1370) wifi: mode : sta (24:0a:c4:82:88:84)
I (1370) wifi: mode : softAP (24:0a:c4:82:88:85)
I (1370) wifi: Total power save buffer number: 16
I (1380) wifi: Init max length of beacon: 752/752
I (1380) wifi: Init max length of beacon: 752/752
Free heap after WIFI AP init(bytes): 260100

I (13860) wifi: n:1 1, o:1 0, ap:1 1, sta:255 255, prof:1
I (13860) wifi: station: 3c:77:e6:b0:cf:d5 join, AID=1, bgn, 40U
Guru Meditation Error: Core 1 panic'ed (InstrFetchProhibited). Exception was unhandled.
Core 1 register dump:
PC : 0x00000000 PS : 0x00060b30 A0 : 0x80153c0a A1 : 0x3ffc5790
A2 : 0x3ffd093c A3 : 0x00000000 A4 : 0x00000224 A5 : 0x3ffd0b64
A6 : 0x00000000 A7 : 0x00000224 A8 : 0x801601b1 A9 : 0x3ffc5750
A10 : 0x3ffd094c A11 : 0x3ffd0b64 A12 : 0x3ffc5790 A13 : 0x00000044
A14 : 0x0000002e A15 : 0x00000001 SAR : 0x00000010 EXCCAUSE: 0x00000014
EXCVADDR: 0x00000000 LBEG : 0x4000c349 LEND : 0x4000c36b LCOUNT : 0x00000000

Backtrace: 0x00000000:0x3ffc5790 0x40153c07:0x3ffc5800 0x40159b2d:0x3ffc5850 0x4015f64e:0x3ffc5880 0x40151fb9:0x3ffc58a0

Rebooting...

**Parsed backtrace**
0x40153c07: udp_input at D:/GitRepo/esp-adf/esp-idf/components/lwip/lwip/src/cor e/udp.c:401
0x40159b2d: ip4_input at D:/GitRepo/esp-adf/esp-idf/components/lwip/lwip/src/cor e/ipv4/ip4.c:740
0x4015f64e: ethernet_input at D:/GitRepo/esp-adf/esp-idf/components/lwip/lwip/sr c/netif/ethernet.c:184
0x40151fb9: tcpip_thread at D:/GitRepo/esp-adf/esp-idf/components/lwip/lwip/src/ api/tcpip.c:483

zuqualla, 2019-06-16:
anyone tried this? [#issuecomment-453956975](https://github.com/espressif/arduino-esp32/issues/2283#issuecomment-453956975). 

Just putting `WiFi.softAPConfig` after `WiFi.softAP`... seems to help here (for now, ;) this bug is erratic!)

xdraylin, 2019-06-19:
For me erasing the flash and setting WiFi.persistent(false) seems to get rid of the problem.

ThinkalVB, 2019-07-21:
I also got the same problem. When removing the 
WiFi.softAPConfig(local_IP, gateway, subnet);  the problem goes above.
IPAddress local_IP(192,168,1,1);
IPAddress gateway(192,168,1,0);
IPAddress subnet(255,255,255,0);
Anyone know how to fix it or what the problem is ?

stickbreaker, 2019-07-21:
> 
> 
> I also got the same problem. When removing the
> 
> ```
> IPAddress local_IP(192,168,1,1);
> IPAddress gateway(192,168,1,0);
> IPAddress subnet(255,255,255,0);
> WiFi.softAPConfig(local_IP, gateway, subnet);  
> ```
> 
> the problem goes away. Anyone know how to fix it or what the problem is?

Invalid Gateway address, review IP protocol rules.

Chuck.

ThinkalVB, 2019-07-21:
Ah, @stickbreaker I didn't see that one coming; thank's for the assist. I removed the code, however. Can you suggest me some materials from where I can master the concept of gateways, subnets, localIP and how I can play with these values in ESP32 / ESP8266? ( like can I use the local IP as the gateway ... and etc...)

stickbreaker, 2019-07-21:
ipv4 is really simple:
* each device has an address
* each address is a member of a group (net)
* each net has an all-call (everybody answers) address
* each net is defined by a mask(subnet).

With these rules everybody in the group(net) can both be identified and is unique.  The Mask value is used to verify whether an address is part of the group(net) or exists in another group(net).

The Mask is just a bit-wise "AND" filter.  In your example mask=255.255.255.0 is also 0xFFFFFF00 this says that any member of this group(net) has to have the same first three octets (i.e. 192.168.1 ).
from [IPAddress.h](https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/IPAddress.h)
```c++
IPAddress  ip1,ip2,mask;
uint32_t test;
mask =IPAddress(255,255,255,0);
ip1 = IPAddress(192,168,1,23);
ip2 = IPAddress(192,168,2,23);
// are ip1 and ip2 members of the same group (subnet)?
if( (ip1 & mask) == (ip2 & mask)){
   Serial.printf(" ip1(%s) and ip2(%s) are both members of subnet %s.",ip1.toString().c_str(), (),ip2.toString().c_str(), mask.toString().c_str());
} 
else {
   Serial.printf(" ip1(%s) and ip2(%s) are NOT  members of subnet %s.",ip1.toString().c_str(), (),ip2.toString().c_str(), mask.toString().c_str());
};
```
The all-call(broadcast) address is the mask address AND 0xFFFFFFFF, so for your example 192.168.1.255.

Address 0 of any group(subnet) is reserved(unused).

The gateway address is the address of the computer (gatekeeper) through which to access any address outside this group. To access a computer outside this local network, the request is encapsulated into a packet addressed to the gateway.  The gateway receives this packet, opens it up and reads the destination address, It then compares this destination to a routing table inside the gateway, finds either the exact network that contains the destination or the next higher gateway, repacks the request with the newly found destination, and sends the packet.

In a simple home network, your "router" is the single gateway to the wider world.  The standard home network uses one of of "C" class private networks. i.e 192.168.x.x usually 192.168.1.x. Which is usually masked down 192.168.1.0/8 (192.168.1.1 to 192.168.1.254) This allows 254 devices to communicate with each other.  

I would recommend actually spending some time learning TCP/IP Joe Casad has written a good reference [TCP/IP in 24 Hours](https://www.amazon.com/TCP-Hours-Sams-Teach-Yourself/dp/0672337894/ref=dp_ob_title_bk)

Chuck.

lbernstone, 2019-07-22:
You absolutely can and should use the SoftAPIP as the gateway.  To learn about subnetting, just google "network essentials subnet".  There are millions of sites that will explain subnetting.  The default config is:
SoftAP IP: 192.168.4.1
Gateway: 192.168.4.1
Subnet Mask: 255.255.255.0
DNS: -----

ThinkalVB, 2019-07-22:
@stickbreaker Wow that was very much insightful. Thank you for the head start. Some 5 years from now I have read the entire book " Data Communications and Networking" Book by Behrouz A. Forouzan, however, this is the first time I was coming into practical sides of things. I will take it from here; thank you for your kind understanding and time.

reapzor, 2019-08-04:
I noticed that setting up a WIFI_AP or WIFI_AP_STA mode with a given SSID works fine the first time you flash the chip. The second time you flash it, there is an issue where a crash occurs any time a client tries to connect to it. 

I am not able to tell what is going on, but this patch below to WiFiAP.cpp works for me in softAP()
```
    //if(!softap_config_equal(conf, conf_current) && esp_wifi_set_config(WIFI_IF_AP, &conf) != ESP_OK) {
    //    return false;
    //}
	esp_wifi_set_config(WIFI_IF_AP, &conf);
```
This unconditionally sets the config each time softAP() is called

What I noticed is that softap_config_equal() returns false the first time I flash the chip with a new SSID to use. The second time I flash it, code unchanged, this starts to return true.
When this happens, a new config is never set, relying on what seems to be something in the flash already from the previous firmware upload. Setting the config unconditionally avoids whatever is wrong in the stale/older config, and gets things working.

The below sketch is the bare minimum to reproduce this problem. Note that you have to upload the sketch twice before it will start to fail. It is just the CaptivePortal.ino sketch. 
The first time it is uploaded, it works fine. The second time, it crashes whenever clients try to connect to it.

Anyone else with this behavior see this pattern or can test with the below sketch?
Repro:
1: Upload sketch to an ESP32 board
2: Using windows 10, connect to the AP it made: "DNSServer CaptivePortal example"
3: Observe success connecting to the AP
4: Re-upload the sketch unchanged
5: Repeat 2 (Connect to the AP again)
6: Observe esp32 reboots

```
#include <WiFi.h>
#include <DNSServer.h>

const byte DNS_PORT = 53;
IPAddress apIP(192, 168, 1, 1);
DNSServer dnsServer;
WiFiServer server(80);

String responseHTML = ""
  "<!DOCTYPE html><html><head><title>CaptivePortal</title></head><body>"
  "<h1>Hello World!</h1><p>This is a captive portal example. All requests will "
  "be redirected here.</p></body></html>";

void setup() {
  WiFi.mode(WIFI_AP);
  WiFi.softAPConfig(apIP, apIP, IPAddress(255, 255, 255, 0));
  WiFi.softAP("DNSServer CaptivePortal example");

  // if DNSServer is started with "*" for domain name, it will reply with
  // provided IP to all DNS request
  dnsServer.start(DNS_PORT, "*", apIP);

  server.begin();
}

void loop() {
  dnsServer.processNextRequest();
  WiFiClient client = server.available();   // listen for incoming clients

  if (client) {
    String currentLine = "";
    while (client.connected()) {
      if (client.available()) {
        char c = client.read();
        if (c == '\n') {
          if (currentLine.length() == 0) {
            client.println("HTTP/1.1 200 OK");
            client.println("Content-type:text/html");
            client.println();
            client.print(responseHTML);
            break;
          } else {
            currentLine = "";
          }
        } else if (c != '\r') {
          currentLine += c;
        }
      }
    }
    client.stop();
  }
}
```

reapzor, 2019-08-04:
The bug seems to get primed in WiFIGenericClass::mode(). Adding a small delay between espWiFiStart() and esp_wifi_set_mode() in WiFiGeneric.cpp gets the softAP() working reliably every time. The log message in _eventCallback was providing this delay, working around the issue. The below code snippet also works around this issue.

```
bool WiFiGenericClass::mode(wifi_mode_t m)
{
    wifi_mode_t cm = getMode();
    if(cm == m) {
        return true;
    }
    if(!cm && m){
        if(!espWiFiStart(_persistent)){
            return false;
        }
    } else if(cm && !m){
        return espWiFiStop();
    }
    sleep(1);  // Delay between espWiFiStart() and esp_wifi_set_mode()
    esp_err_t err;
    err = esp_wifi_set_mode(m);
    if(err){
        log_e("Could not set mode! %d", err);
        return false;
    }
    return true;
}
```

I don't think this is a fix, just posting observations.

dataman, 2019-08-04:
What you’re calling the all-call address is really the broadcast address.  Probably better to call it what it is.  I wrote a great white paper on supernetting and subnetting when I first got my mcse.

Sent from my iPhone Xs!
Charley Jones, PMP

xdraylin, 2019-08-07:
@reapzor Can you check the result of `esp_wifi_get_config(WIFI_IF_AP, &conf_current)` before `if(!softap_config_equal(conf, conf_current) && esp_wifi_set_config(WIFI_IF_AP, &conf) != ESP_OK)` in `WiFiAP.cpp` right before a crash occurs? Maybe `conf_current` is not set by the function due to an error and `softap_config_equal` will then try to access undefined memory causing the panic.

Yoshi20, 2019-08-14:
Wait for a station to connect before any DNS/request handling:
```
#include <esp_wifi.h>

/* setup accesspoint */
WiFi.mode(WIFI_MODE_APSTA)
WiFi.enableAP(true)
WiFi.softAP("mySSID");

/* wait for a station to connect */
wifi_sta_list_t station_list;
while(station_list.num == 0){
	esp_wifi_ap_get_sta_list(&station_list);
	delay(100);
}

/* some DNS/request handling */
...
```

kapyaar, 2019-09-11:
@reapzor Quite an interesting find. Spend hours troubleshooting an out of the blue panic error crash which kept happening after all was well. Came across your post, and realized that I had changed core debug level from Verbose to Error only.  

With verbose, board is holding up ok.

CircuitSetup, 2019-09-19:
> I noticed that setting up a WIFI_AP or WIFI_AP_STA mode with a given SSID works fine the first time you flash the chip. The second time you flash it, there is an issue where a crash occurs any time a client tries to connect to it.

Is this still happening in 1.0.3 where EEPROM was moved to NVS?

CircuitSetup, 2019-09-27:
@reapzor I noticed that there is something not right with handling and closing requests via captive portal, which I see that you are using. 

I removed the captive portal, and the crashes went away. In a related issue, requests weren't being handled correctly if a browser was left open, and/or a connection was dropped and reconnected, which is fixed with the latest builds of AsyncTCP and ESPAsyncWebServer.

pekkin, 2019-10-19:
It's a bit magic, but in my case, it is working as 

WiFi.softAP(ssid, password);
delay(2000); // VERY IMPORTANT
WiFi.softAPConfig(apIP, apIP, IPAddress(255, 255, 255, 0));

ttlappalainen, 2019-11-07:
I also found that using default AP IP and without softAPConfig there is no problem. With softAPConfig without delay crahed 100% by calling ipconfig /renew. 

Then I passed AP IP config by doing it at very first of wifi configuration with code:

```
extern void tcpipInit();

// *****************************************************************************
void SetWiFiAdapter(IPAddress local_ip, IPAddress gateway, IPAddress subnet) {
  tcpip_adapter_dhcps_stop(TCPIP_ADAPTER_IF_AP);

  tcpip_adapter_ip_info_t info;
  info.ip.addr = static_cast<uint32_t>(local_ip);
  info.gw.addr = static_cast<uint32_t>(gateway);
  info.netmask.addr = static_cast<uint32_t>(subnet);

  if(tcpip_adapter_set_ip_info(TCPIP_ADAPTER_IF_AP, &info) == ESP_OK) {
      dhcps_lease_t lease;
      lease.enable = true;
      lease.start_ip.addr = static_cast<uint32_t>(local_ip) + (1 << 24);
      lease.end_ip.addr = static_cast<uint32_t>(local_ip) + (11 << 24);

      tcpip_adapter_dhcps_option(
          (tcpip_adapter_option_mode_t)TCPIP_ADAPTER_OP_SET,
          (tcpip_adapter_option_id_t)REQUESTED_IP_ADDRESS,
          (void*)&lease, sizeof(dhcps_lease_t)
      );
      tcpip_adapter_dhcps_start(TCPIP_ADAPTER_IF_AP);
  }
}

setup () {
  tcpipInit();
  SetWiFiAdapter(local_ip,gateway,subnet);
...
WiFi.softAP(ssid, password);
...
}
```

th0ma5w, 2019-12-07:
I was getting a core 0 panic InstrFetchProhibited, and then I erased the flash and reflashed and it all just works now.

BlueAndi, 2019-12-07:
I noticed that the problem seems to be NV related and erased the flash too. Additional I avoided to store the configuration in the NV.
```
WiFi.persistent(false);
```
With that combination its working for me now since october.

lucian0112, 2020-01-16:
> I noticed that the problem seems to be NV related and erased the flash too. Additional I avoided to store the configuration in the NV.
> 
> ```
> WiFi.persistent(false);
> ```
> 
> With that combination its working for me now since october.

Hi,
can you be more speciffic on your solution? I have same issue but don t really understand what you did actually to fix it,
Thank you

kapyaar, 2020-01-16:
@lucian0112  Add 

WiFi.persistent(false);

before you do Wifi.begin(...) in your code

lucian0112, 2020-01-16:
like this?

```
void setup() {
  Serial.begin(9600);
  Serial2.begin(9600, SERIAL_8N1, RXD2, TXD2);
  
  WiFi.persistent(false);
  WiFi.softAP(ssid, password);
  WiFi.softAPConfig(local_ip, gateway, subnet);
  
  server.on("/", handle_OnConnect);
  server.on("/obdon", handle_obdon);
  server.on("/obdoff", handle_obdoff);
  server.onNotFound(handle_NotFound);
  
  server.begin();

  webSocket.begin();
  webSocket.onEvent(webSocketEvent);
}
```

BlueAndi, 2020-01-16:
Yes, in your case before setting up the access point with WiFi.softAP(). Don't forget to erase the whole flash before you test it again.

lucian0112, 2020-01-16:
Thank you,
I will get Flash Download Tool and erase flash 
Thanks again for help

brightisolutions, 2020-01-26:
@BlueAndi , thanks a lot for the help. Is working fine now.

AndyUK1966, 2020-03-25:
I have the same issue, but only when I plug in 433mhz RF modules for LightwaveRF.h, then I have to comment out WiFi.softAPConfig(local_IP, gateway, subnet); , if I unplug the modules it works fine though. Strange.

mshoe007, 2020-07-05:
Hey Mr. Stale Bot. Nothing got fixed in the Esp32 code. Can we leave it open? AlphaLima put a workaround in some other code. The rest of us trying to use esp32 as an access point have this crash.

KdotL, 2020-10-06:
including ` WiFi.persistent(false);` before `WiFi.softAP()` and resetting the flash ( `platformio run --target erase` if you use platformio) solved the issue for me. 

Thanks

3ricj, 2020-11-26:
So we found a problem, identified a workaround, but didn't fix anything in the core and it's impacting lots of other users. Is there not a better solution here?

vshymanskyy, 2020-12-17:
@igrr please reopen this

JohnboyK, 2020-12-25:
Is there any real working solution for this issue?
I ported code form ESP8266 to ESP32 and exactly the problem occurs when:
+) New flash of the program including full erase of progmem
+) Then first time of start as SoftAP works (it seems that at this point ESP writes parameters into internal memory)
    I am able to connect from my mobile phone to the Soft-AP and can work with the webserver on the ESP32
+) After reboot the ESP and every RE-connect with my mobile phone ESP crashes with (InstrFetchProhibited)
All my tests of workarounds or solutions listed above (delay of 2000 ms;  move command softAPConfig behind SoftAP.begin, start without softAPConfig) gave no solution that I can use the ESP32 for more than once before reflash and erase full memory.
Beginning of the Android client try the second time to connect to the SoftAP the ESP32 crashes...
It seems that the Android client ask the ESP32 WiFi-AP to use the same IP-address as last connection and this request crashes the ESP32....
Question: Is there any chance to erase the network parameters which are stored into NVRAM by WiFi-internal procedures from my program? Then I could try to erase this part of NVRAM from my program and then I could use the ESP32 every second reboot....
Thanks in advance
Johann

mshoe007, 2020-12-25:
Have you tried WiFi.persistent(false); 

I have as this pretty near the front of any ESP8266 or ESP32 project's setup() function:
    WiFi.disconnect();
    WiFi.mode(WIFI_OFF);
    WiFi.persistent(false);

Then I set WiFi client or access mode, and this seems not to leave stuff lying around in persistent storage.

I can now create an access point just fine, and it no longer crashes when a client connects.

JohnboyK, 2020-12-25:
**Thank you!** Based on your input the following sequence seems to work, but **important task before upload the sketch:** 
Erase flash completeley  (e.g. with NodeMCU PyFlasher with option "Erase flash yes, wipes all data" )
`    WiFi.disconnect(true);             // that no old information is stored
    WiFi.mode(WIFI_OFF);               // switch WiFi off
    delay(1000);                       // short wait to ensure WIFI_OFF
    WiFi.persistent(false);            // avoid that WiFi-parameters will be stored in persistent memory
    Serial.print(F("\nNow we start soft-AP....."));
    LWB2 = WiFi.softAPConfig(AP_local_IP, AP_gateway, AP_subnet);   // configure AP with fixed IP-address
    if ( !LWB2 ) {         // if config-setting not successful
        Serial.print(F("\n###ERR: softAPConfig failed with SSID: "));
        Serial.print (gcfg_parSSID);
    }
    delay(1000);   
    WiFi.mode(WIFI_AP);                    //Access Point Only
    delay(1000);
    WiFi.persistent(false);                // avoid that congifuration is written to persistent memory
    LWB1 = WiFi.softAP(gcfg_parSSID, gcfg_parWLanPWD);   // Start ESP as Access-Point
    if ( LWB1 ) {
        Serial.print(F("\nWLAN-AP start successful; IP-Address: "));
        Serial.println( WiFi.softAPIP().toString() );  
    }  //  end  if (LWB1 )
    else   {
        Serial.print(F("\n### Start AccessPoint failed!"));
    }  //  end  else`
With delay-timers I have to test what I can reduce without impact to functionality

nemotora, 2021-01-02:
> **Thank you!** Based on your input the following sequence seems to work, but **important task before upload the sketch:**
> Erase flash completeley (e.g. with NodeMCU PyFlasher with option "Erase flash yes, wipes all data" )
> here the code:
> 
> ```
>     WiFi.disconnect(true);             // that no old information is stored  
>     WiFi.mode(WIFI_OFF);               // switch WiFi off  
>     delay(1000);                       // short wait to ensure WIFI_OFF  
>     WiFi.persistent(false);            // avoid that WiFi-parameters will be stored in persistent memory  
>     Serial.print(F("\nNow we start soft-AP....."));  
>     LWB2 = WiFi.softAPConfig(AP_local_IP, AP_gateway, AP_subnet);   // configure AP with fixed IP-address  
>     if ( !LWB2 ) {         // if config-setting not successful  
>         Serial.print(F("\n###ERR: softAPConfig failed with SSID: "));  
>         Serial.print (gcfg_parSSID);  
>     }  //  end  if (!LWB2)  
>     delay(1000);     
>     WiFi.mode(WIFI_AP);                    //Access Point Only  
>     delay(1000);  
>     WiFi.persistent(false);                // avoid that congifuration is written to persistent memory  
>     LWB1 = WiFi.softAP(gcfg_parSSID, gcfg_parWLanPWD);   // Start ESP as Access-Point  
>     if ( LWB1 ) {  
>         Serial.print(F("\nWLAN-AP start successful; IP-Address: "));  
>         Serial.println( WiFi.softAPIP().toString() );    
>     }  //  end  if (LWB1 )  
>     else   {  
>         Serial.print(F("\n### Start AccessPoint failed!"));  
>     }  //  end  else  
> ```
> 
> With delay-timers I have to test what I can reduce without impact to functionality

I was getting the same error. I can confirm that the above mentioned workaround helped me to make it work again.

adojang, 2021-01-06:
> Is there any real working solution for this issue?
> I ported code form ESP8266 to ESP32 and exactly the problem occurs when:
> +) New flash of the program including full erase of progmem
> +) Then first [...]

This workaround worked for me too, although I didn't have to flash my device, simply reuploading the sketch using the arduino ide was enough. It seems like the big secret is the 2000ms delay between the two

Code for the lost soul:
`
#include <WiFi.h>
#include <WebServer.h>
/* Put your SSID & Password */
const char* ssid = "Test";  // Enter SSID here
const char* password = "test12345";  //Enter Password here

/* Put IP Address details */
IPAddress local_ip(192,168,1,25);
IPAddress gateway(192,168,1,25);
IPAddress subnet(255,255,255,0);

WebServer server(80);

//Then put this inside your void setup() function:
WiFi.softAP(ssid, password);
delay(2000);
WiFi.softAPConfig(local_ip, gateway, subnet);
`

JohnboyK, 2021-01-06:
Have you tried to do the following
1)connect to this softAP with an Arduino-client (Mobile phone or tablet)
2)work with the webserver
3)disconnect from softAP-WLAN
4) connect again to softAP-WLAN with the same Arduino-client
5) work with the webserver
In my case at step 4 I got the ESP crash with (InstrFetchProhibited) because it seems that the Arduino client send to DHCP an offer and tries to get the same IP-Address as he had before.
So for my problem the real solution were only to fully disconnect and switch off WiFi, wait for a second and, then start as SoftAP or Client with non-persistent parameter.... Code as described in my posting above...
    WiFi.disconnect(true);             // that no old information is stored  
    WiFi.mode(WIFI_OFF);               // switch WiFi off  
    delay(1000);                       // short wait to ensure WIFI_OFF  
    WiFi.persistent(false);            // avoid that WiFi-parameters will be stored in persistent memory
    ......`

closed 2021-07-08 by stale[bot]

AdamMiltonBarker, 2021-08-21:
This library has serious issues

ktonder, 2021-09-09:
Struggled too with this issue, none of the tips above regarding the delay helped me. What did do the trick was to set the correct "FlashFreq" to 40Mhz. And I also reduced the CPUFreq from 240MHZ to 80mhz.

Seems like its a lot more stable for mye now.

owk444, 2022-05-06:
I've got identical problem on the some board. After reading this thread, my solution was:
1. Remove all WiFi.* from code. Compile and upload. All works fine.
2. Add WiFi.* one by one with delay(2000) and compile and upload on each step. I got a working solution
3. Remove all delay() one by one from the top of the code, compile and upload on each step. All works fine.

Vscode+Platformio. LOLIN32. No erasing flash.

Dependency Graph
|-- <ESP8266FtpServer> 1.0.1
|   |-- <FS> 2.0.0
|   |-- <WiFi> 2.0.0
|   |-- <SPIFFS> 2.0.0
|   |   |-- <FS> 2.0.0
|-- <ArduinoJson> 6.19.4
|-- <SPIFFS> 2.0.0
|   |-- <FS> 2.0.0
|-- <WebServer> 2.0.0
|   |-- <WiFi> 2.0.0
|   |-- <FS> 2.0.0
|-- <DNSServer> 2.0.0
|   |-- <WiFi> 2.0.0
|-- <WiFi> 2.0.0

Working part of the code:

        WiFi.mode(WIFI_AP);
	Serial.println("Setting mode to Access Point");
	WiFi.softAP(ssid, password);
	Serial.println("An Access Point started");
	IPAddress IP = WiFi.softAPIP();
	Serial.println(IP);

	DNSServer dnsServer;
	FtpServer ftpSrv;
	Serial.println("DNS + FTP server started");
		
	server.on("/", handleRoot);					// main page
	server.on("/store", handleStore);			// save config
  	server.onNotFound(handleNotFound);			// 404
  	server.begin();
  	Serial.println(F("HTTP server started"));

	while (true) {
		server.handleClient();
		Serial.print(".");
		//delay(2);//allow the cpu to switch to other tasks ( from example )
		delay(2000); // may be I delete this in the future, now it works well
	}