espressif/arduino-esp32 #2026: Serial2.begin cause wifi.begin broke

Huibean, 2018-11-07:
### Hardware:
Board:							ESP32 Wrover IB
Core Installation/update date:			26/Sep/2018
IDE name:							Arduino IDE
Flash Frequency:					80Mhz
PSRAM enabled:                                          yes
Upload Speed:						115200
Computer OS:                                            Mac OSX

### Description:
After adding Serial2.begin, WiFi.begin case core panic

```
#include <WiFi.h>

const char* ssid     = "ssid";
const char* password = "password";

void setup()
{
    Serial.begin(115200);
    Serial2.begin(115200);
    delay(10);

    // We start by connecting to a WiFi network

    Serial.println();
    Serial.println();
    Serial.print("Connecting to ");
    Serial.println(ssid);

    WiFi.begin(ssid, password);

    while (WiFi.status() != WL_CONNECTED) {
        delay(500);
        Serial.print(".");
    }

    Serial.println("");
    Serial.println("WiFi connected");
    Serial.println("IP address: ");
    Serial.println(WiFi.localIP());
}

void loop() {
  
}
```

### Debug Messages:
```
PC: 0x4009118b: multi_heap_malloc_impl at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/heap/multi_heap.c line 131
EXCVADDR: 0xffffffff

Decoding stack results
0x4009118b: multi_heap_malloc_impl at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/heap/multi_heap.c line 131
0x4009179a: multi_heap_malloc at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/heap/multi_heap_poisoning.c line 191
0x400868d5: heap_caps_malloc at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/heap/heap_caps.c line 110
0x40086986: heap_caps_malloc_prefer at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/heap/heap_caps.c line 188
0x40086031: wifi_malloc at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/wifi_os_adapter.c line 61
0x400dbd7d: esp_wifi_init at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/wifi_init.c line 51
0x400d1da9: WiFiGenericClass::mode(wifi_mode_t) at /Users/huibean/Documents/Arduino/hardware/espressif/esp32/libraries/WiFi/src/WiFiGeneric.cpp line 121
0x400d1e6d: WiFiGenericClass::enableSTA(bool) at /Users/huibean/Documents/Arduino/hardware/espressif/esp32/libraries/WiFi/src/WiFiGeneric.cpp line 538
0x400d1f55: WiFiSTAClass::begin(char const*, char const*, int, unsigned char const*, bool) at /Users/huibean/Documents/Arduino/hardware/espressif/esp32/libraries/WiFi/src/WiFiSTA.cpp line 118
0x400d1807: setup() at /Users/huibean/Documents/Arduino/Wifi_Test/Wifi_Test.ino line 19
0x401203fb: loopTask(void*) at /Users/huibean/Documents/Arduino/hardware/espressif/esp32/cores/esp32/main.cpp line 15
0x4008db5d: vPortTaskWrapper at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/port.c line 141
```

Huibean, 2018-11-07:
disable PSRAM problem disappear, but PSRAM is needed

stickbreaker, 2018-11-07:
@Huibean Declare what pins you want Serial2 to use:
```c++
    void begin(unsigned long baud, uint32_t config=SERIAL_8N1, int8_t rxPin=-1, int8_t txPin=-1, bool invert=false);
// something like
Serial2.begin(115200,SERIAL8N1, 18,19);
```
by default Serial2 uses 16,17 which are used by PSDRAM
Chuck.

Huibean, 2018-11-07:
oh no, 16,17 is exactly what  I use

Huibean, 2018-11-07:
but when I change to Serial1(with default pin ) this would happen
```
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Guru Meditation Error: Core  1 panic'ed (Unhandled debug exception)
Debug exception reason: Stack canary watchpoint triggered (loopTask) 
```

stickbreaker, 2018-11-07:
Serial2 defaults 16,17 are used by PSDRAM
Serial1 default 9,10 are used by FLASH

you have to use other pins.

Chuck.

Huibean, 2018-11-07:
Thanks! @stickbreaker

closed 2018-11-07 by Huibean

blaticslm, 2023-03-31:
> @Huibean Declare what pins you want Serial2 to use:
>
> ```c++
>     void begin(unsigned long baud, uint32_t config=SERIAL_8N1, int8_t rxPin=-1, int8_t txPin=-1, bool invert=false);
> // something like [...]

5 years later, I wish to see this comment earlier because I FACED THE SAME PROBLEM WHEN I INITIALIZED SERIAL2 AND USE PSRAM AT THE SAME TIME!  Now I move the pins to anther digital pins and finally there....