espressif/arduino-esp32 #2027: Latest version, error with no exception to decode ...

euquiq, 2018-11-07:
I just updated to latest Github Arduino, compiled my current project, and after a minute or so of working, it throws:

```
Guru Meditation Error: Core  1 panic'ed (Cache disabled but cached memory region accessed)
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Guru Meditation Error: Core  1 panic'ed (IllegalIGuru Meditation Error: Core  1 panic'ed (Unhandled debug exception)
Debug exception reason: Stack canary watchpoint triggered () 
Guru Meditation Error: Core  1 panic'ed (
```

Just like that. Same code, earlier esp32 arduino version, did have some problems (it took longer to crash)  but always gave the stack trace and I could decode it.

Is this a bug in current version ? Or is it (most probably!) me just not understanding something ?

Regards,

Enrique

chegewara, 2018-11-07:
`Stack canary` is heap memory debugging which seems to be turned on here:
https://github.com/espressif/arduino-esp32/blob/master/tools/sdk/sdkconfig#L458

It can be caused by memory allocation or de-allocation problem.

euquiq, 2018-11-08:
I am still chasing bugs inside my project, but things are quite difficult (and weird): As already stated above, whenever I manage to get my ESP32 to hang / crash, I just get this kind of output (AND THEN IT FREEZES, it does NOT reset itself).

```
Guru Meditation Error: Core  1 panic'ed (Cache disabled but cached memory region accessed)
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Guru Meditation Error: Core  1 panic'ed (IllegaGuru Meditation Error: Core  1 panic'ed (Unhandled debug exception)
Debug exception reason: Stack canary watchpoint triggered (loop��@4) 
Guru Meditation Error: Core  1 panic'ed (Guru Meditation Error: Core  1 panic'ed (Double exception)
  1�H@��?8�?  1�H@��?8�?  1�H@��?8�?  1�H@��?8�?BREAK instr X�?ets Jun  8 2016 00:22:57
```

ArTyGill, 2018-12-12:
I have the same problem, the watchdog does not work and the controller freezes. Fee esp32-wrover-i.
Guru Meditation Error: Core  1 panic'ed (Cache disabled but cached memory region accessed)
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Guru Meditation Error: Core  1 panic'ed (IllegalIns

me-no-dev, 2018-12-12:
guys... none of you posted ANY code at all... what good those error messages will do to anyone?
And BTW... the very first error line reads `Guru Meditation Error: Core  1 panic'ed (Cache disabled but cached memory region accessed)` which means that you are doing something you should not do in ISR. What ISR? Post some code and maybe we can help.

closed 2019-08-15 by stale[bot]