espressif/arduino-esp32 #2031: rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)

shreyaspaqs, 2018-11-08:
Make your question, not a Statement, inclusive. Include all pertinent information:

What you are trying to do.
Describe your system( Hardware, computer, O/S, core version, environment)
Describe what is failing
Show the shortest possible code that will duplicate the error
Show the EXACT error message(it doesn't work is not enough)
Then if someone is interested and knowledgeable you might get a answer.  All of this work on your part shows us that you have worked to solve YOUR problem. The more complete your issue posting is, the more likely someone will volunteer their time to help you.

If you have a Guru Meditation Error or Backtrace, ***please decode it***:
https://github.com/me-no-dev/EspExceptionDecoder

----------------------------- Remove above -----------------------------

### Hardware:
Board:							?ESP32 Dev Module? ?node32? ?ttgo_lora?
Core Installation/update date:			?11/jul/2017?
IDE name:							?Arduino IDE? ?Platform.io? ?IDF component?
Flash Frequency:					?40Mhz?
PSRAM enabled:                                          ?no?
Upload Speed:						?115200?
Computer OS:                                            ?Windows 10? ?Mac OSX? ?Ubuntu?

### Description:
Describe your problem here

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp

//Change the code below by your sketch
#include <Arduino.h>

void setup() {
}

void loop() {
}
```

### Debug Messages:
```
Enable Core debug level: Debug on tools menu of Arduino IDE, then put the serial output here 
```

closed 2018-11-08 by shreyaspaqs

atanisoft, 2018-12-21:
@huynh213 please open a new issue to track this rather than an old closed issue.

alexandreMarquis, 2020-08-16:
> Hello,
> 
> I got the same error and have been searching for an answer but no luck, and not sure if this's the right place to post the question. Please delete or move it if this's not a valid location.
> 
> Goal: Loading firmware to my custom board using ESP32 Download Tool.
> 
> I've been developing my code on my custom board (use ESP32-WROOM32D module) in Arduino IDE, and it loads the firmware perfectly and works great using Arduino:
> 
> **For Arduino**
> Board: ESP32 Dev Module
> Upload Speed: 115200
> Flash frequency: 80Mhz
> Flash Mode: DIO
> Flash Size 4MB
> Partition Scheme: Minimal SPIFFS (since I need more space for Wifi, BLE, and OTA libs)
> 
> **For the ESP32 Bootloader:**
> Crystal Frequency: 40M
> SPI speed: 80MHz
> SPI mode: DIO
> Flash size: 32Mbit
> SpiAuoSet: unchecked
> DonotChgBin: unchecked
> 
> For bin files
> 
> 1. For the bootloader:
>    I did try to use bootloader_dio_80m.bin at the address 0x1000
>    -> I got the file from Arduino\hardware\espressif\esp32\tools\sdk\bin
>    -> For the address, I am not sure which one to put there so I use a random address that I found online, where other people use it.
> 2. For the partition:
>    I have min_spiffs.bin at the address 0x4000
>    ->I used Python to generate the bin file from the min_spiffs.csv from the folder Arduino\hardware\espressif\esp32\tools\partitions
>    -> Same as the bootloader's address, I really don't know what to use for the partition address.
> 3. For my firmware:
>    I have the .bin file generated by Arduino, address at 0x10000. It looks like this is a valid address.
> 
> **Some other info for reference:**
> --------------------------Arduino Message when successful loading firmware and board works-------
> esptool.py v2.3.1
> Connecting......
> Chip is ESP32D0WDQ5 (revision 1)
> Features: WiFi, BT, Dual Core, Embedded Flash
> Uploading stub...
> Running stub...
> Stub running...
> Configuring flash size...
> Auto-detected Flash size: 8MB (**I will switch my hardware module to 4MB later)
> Compressed 8192 bytes to 47...
> 
> Writing at 0x0000e000... (100 %)
> Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.0 seconds (effective 65525.1 kbit/s)...
> Hash of data verified.
> Flash params set to 0x023f
> Compressed 13696 bytes to 8946...
> 
> Writing at 0x00001000... (100 %)
> Wrote 13696 bytes (8946 compressed) at 0x00001000 in 0.8 seconds (effective 134.8 kbit/s)...
> Hash of data verified.
> Compressed 1509936 bytes to 871891...
> 
> Writing at 0x00010000... (1 %)
> .................
> 
> --------------------------Using ESP32 Download Tool, the board doesn't work-------
> 
> From the ESP32 Download Tool:
> 
> .Uploading stub...
> Running stub...
> Stub running...
> test offset : 4096 0x1000
> case ok
> test offset : 16384 0x4000
> case ok
> test offset : 65536 0x10000
> case ok
> CONNECT BAUD: 115200
> .Efuse CONSOLE_DEBUG_DISABLE is already burned.
> Uploading stub...
> Running stub...
> Stub running...
> 
> Then from Arduino Serial Monitor after resetting the power
> 
> rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
> configsip: 0, SPIWP:0xee
> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
> mode:DIO, clock div:1
> load:0x3fff0018,len:4
> load:0x3fff001c,len:1324
> load:0x40078000,len:7788
> ho 0 tail 12 room 4
> load:0x40080400,len:6448
> csum err:0xd0!=0xff
> ets_main.c 371
> ets Jun 8 2016 00:22:57

I know this is old but this worked for me : esptool.py --chip esp32 --port /dev/tty.wchusbserial1410 erase_flash