espressif/arduino-esp32 #2032: rst:0x10 (RTCWDT_RTC_RESET),boot:0x13

shreyaspaqs, 2018-11-08:
Make your question, not a Statement, inclusive. Include all pertinent information:

What you are trying to do.
Describe your system( Hardware, computer, O/S, core version, environment)
Describe what is failing
Show the shortest possible code that will duplicate the error
Show the EXACT error message(it doesn't work is not enough)
Then if someone is interested and knowledgeable you might get a answer.  All of this work on your part shows us that you have worked to solve YOUR problem. The more complete your issue posting is, the more likely someone will volunteer their time to help you.

If you have a Guru Meditation Error or Backtrace, ***please decode it***:
https://github.com/me-no-dev/EspExceptionDecoder

----------------------------- Remove above -----------------------------

### Hardware:
Board:							?ESP32 Dev Module? ?node32? ?ttgo_lora?
Core Installation/update date:			?11/jul/2017?
IDE name:							?Arduino IDE? ?Platform.io? ?IDF component?
Flash Frequency:					?40Mhz?
PSRAM enabled:                                          ?no?
Upload Speed:						?115200?
Computer OS:                                            ?Windows 10? ?Mac OSX? ?Ubuntu?

### Description:
Describe your problem here

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp

//Change the code below by your sketch
#include <Arduino.h>

void setup() {
}

void loop() {
}
```

### Debug Messages:
```
Enable Core debug level: Debug on tools menu of Arduino IDE, then put the serial output here 
```

lbernstone, 2018-11-08:
What is it that you are uploading into OTA?  That is what is causing the continuous reboots.

shreyaspaqs, 2018-11-08:
Am uploading Arduino led  Blink example. and I have stored in my server in this format **blink.ino.esp32.bin**  calling the same file in OTA server link. 
or should I rename like **blink.ino.bin** in my server?

Thanks in advance.

shreyaspaqs, 2018-11-08:
> What is it that you are uploading into OTA? That is what is causing the continuous reboots.

Am uploading Arduino led Blink example. and I have stored in my server in this format blink.ino.esp32.bin calling the same file in OTA server link.
or should I rename like blink.ino.bin in my server?

Thanks in advance

copercini, 2018-11-08:
what happens if you try with another code instead  Arduino led Blink example ?

shreyaspaqs, 2018-11-08:
I have not tried with another code. i will try today.

> what happens if you try with another code instead Arduino led Blink example?

Hi,

I have not tried any other code, today I will try.

And am loading bin file in this format blink.ino.esp32.bin is this way is correct or I need change like blink.ino.bin or blink.bin?

shreyaspaqs, 2018-11-08:
**blinky code which I have uploaded to the server is** 

#define LED_BUILTIN 2
const int ledPin =  LED_BUILTIN;
int ledState = LOW;          
unsigned long previousMillis = 0;      
const long interval = 1000;         

void setup() {
  pinMode(ledPin, OUTPUT);
}
void loop() {
  unsigned long currentMillis = millis();

  if (currentMillis - previousMillis >= interval) {
    previousMillis = currentMillis;
    if (ledState == LOW) {
      ledState = HIGH;
    } else {
      ledState = LOW;
    }
    digitalWrite(ledPin, ledState);
  }
}

copercini, 2018-11-08:
try change `#define LED_BUILTIN` to 21 just for test

shreyaspaqs, 2018-11-08:
> try change `#define LED_BUILTIN` to 21 just for test

sure I will test and update you.

shreyaspaqs, 2018-11-09:
Hi, 

Now Code is started working issue was that File name should be changed to blink1.ino.esp32.bin to blink1.ino.bin

Thanks everyone.

closed 2018-11-09 by shreyaspaqs