espressif/arduino-esp32 #2034: esptool.exe issue, can't upload sketches on ESP32-PRO

kalata23, 2018-11-08:
### Hardware: 
Board:							ESP32-PRO
Core Installation/update date:			08/11/2018
IDE name:						Arduino IDE
Flash Frequency:					80Mhz
Upload Speed:						115200
Computer OS:                                          Windows 7/10

### Description:

I have ESP32-PRO board by OLIMEX and try to upload a sketch on it. 

You can see [HERE](https://github.com/OLIMEX/ESP32-PRO/blob/master/HARDWARE/REV-B/ESP32-PRO_Rev_B.pdf) the board schematic. 
As you can see from the scheme, there is PIC32 mcu soldered on board. The main idea is to use the PIC32 chip for USB<==>Serial convertor.  
To do so, I'm using ChipKit core to upload sketches onto PIC32. Then I upload a sketch for [USB->Serial bridge](https://github.com/OLIMEX/ESP32-PRO/blob/master/SOFTWARE/PIC32_part_Arduino_IDE/PIC32_USB-Serial_Bridge/pic32-usb-serial/pic32-usb-serial.ino) which is taking the input data from the PC and transmits it via UART to the ESP32 chip and vise versa. 

I had no issues using esp-idf, esptools.py or ARDUINO with arduino-esp32 core on Linux. Many people and I have a problem using the same setup, but on Windows. We have tried both Windows 7 and 10 with same versions of Arduino IDE, arduino-esp32 core and ESP32-PRO boards. The issue we get is:
```
esptool.py v 2.4.1
 Connecting.....____......____......._____.......____......____
 A fatal error occure: Failed to connect to ESP32: TImed out waiting for packet header`
```

There is no such board ESP32-PRO in the Boards menu in Arduino IDE, so I chose ESP32-EVB, because the two boards have very similar schematics. The major difference is that ESP32-PRO uses PIC32 as Virtual CDC device, while ESP32-EVB or ESP32-GATEWAY uses CH340 chip for USB-SERIAL converter.

The interesting part is that choosing ESP32-EVB or ESP32-Gateway with same version of Arduino IDE and arduino-esp32 core I can upload sketches to ESP32-PRO board with speed up to 1M baud under Linux based OS (Ubuntu, Mint). I have used Oscilloscope to determinate if ESP32 communicates with PC through PIC32 on Windows platform. There seems to be transaction, but the RTS line seemed to not toggle. I installed python 3.7 on Windows and tried to open same COM port which is associated with ESP32-PRO. When i toggle DTR line with python I see on the scope that DTR line changes state as should be, but when I try toggling RTS line i doesnt change it's state, unless I toggle DTR line twice before trying to change RTS line state.
First thing i did was to  change the Olimex USB->Serial sketch, so before asserting RTS line, to double toggle the DTR line. I succeeded to Upload a sketch over ESP32-PRO, but couldn't reset the board.
Then I have downloaded the esptool from  github:
https://github.com/espressif/esptool
After that I started esptool.py script and the board could successfully  read ESP32's mac address and even flash the chip. 
Then i compiled the esptool.py script into Windows binary (esptool.exe) using PyInstaller. Then i replaced the esptool.exe file which is in ` \packages\esp32\tools\esptool\2.31\ `               directory and tried to upload the sketch again on Windows. I have even replaced esptool.exe on other computer with my file and it worked also on the second computer.
I guess there is some issue with the binarry file esptool.

### Debug Messages:
```
esptool.py v 2.4.1
 Connecting.....____......____......._____.......____......____
 A fatal error occure: Failed to connect to ESP32: TImed out waiting for packet header`
```

lbernstone, 2018-11-08:
Please open this issue at https://github.com/espressif/esptool, as it is not an arduino-esp32 question.  You will likely need a driver for the UART device.  See if you have a COM port visible in Windows Control Panel.

kalata23, 2018-11-08:
The issue seems to be related with esptool.exe file which comes with arduino-esp32 core, not with esptool which can be downloaded from the link you gave above. I have proper driver installed, because, when I recompiled the esptool.exe with Pyinstaller using esptool from the git repository and replaced the esptool.exe file that came with arduino-esp32 with the exe I already compiled, the same board ran on same PC with same Windows with same driver. Also I replaced the esptool.exe file which is placed in 
`\packages\esp32\tools\esptool\2.31`  on another PC with Windows 7 installed. Just saying that, maybe the issue is related with the esptool.exe, which I believe is the binary version of esptool.py script.
If you still think i should open new issue, i will do it.

DanKoloff, 2019-02-21:
This seems fixed in the esptool binary in the latest development package 1.0.1, ESP32-PRO seems to work fine with this package.

closed 2019-08-15 by stale[bot]