espressif/arduino-esp32 #2036: EEPROM.h - Using more than 4kb?

1uuc4asb, 2018-11-08:
### Hardware:
Board:							ESP32 DevKit v1
IDE name:					        Arduino IDE
Upload Speed:					115200
Computer OS:                                          Windows 10

I'm in a project, and i need to use the flash to store information.
I need about 30kb of storage.
I'm using the EEPROM.h, but i saw that the maximum that i can get is 4kb.

EEPROM.begin(4096);

As far as i know i have 4mb of flash, but it is divided in pages of 4096b. 
Is there any way to use more than 4kb?

1uuc4asb, 2018-11-08:
@requeijaum

bertmelis, 2018-11-08:
https://github.com/espressif/arduino-esp32/tree/master/libraries/SPIFFS

1uuc4asb, 2018-11-19:
Thanks @bertmelis

bertmelis, 2018-11-19:
when typing this reply, there are 577 issues open. The majority is not an issue, but a general question, a programming error or hardware failure.

Please help to clean up by closing this issue. (after closing, you can still reply and discuss further)

1uuc4asb, 2018-11-19:
Ok! Thanks for the help and sorry. I'll use gitter from now on.

closed 2018-11-19 by 1uuc4asb