espressif/arduino-esp32 #2037: I2C - Strange sensor values - Si7021

7h30n3, 2018-11-08:
### Hardware:
Board:							NodeMCU 32S
Core Installation/update date:			?11/jul/2017?
IDE name:					       Arduino IDE
Flash Frequency:					80Mhz
PSRAM enabled:                                          ?no?
Upload Speed:						921600
Computer OS:                                          Windows 10

### Description:
I try to run a Si7021 on an NodeMCU 32S. At first I couldn't get any communication at all `Did not find Si7021 sensor`. After using the latestest git version of the esp32 library using this instruction [](https://github.com/espressif/arduino-esp32/blob/master/docs/arduino-ide/windows.md) the serial monitor only shows this line over and over again `Humidity:    4.74	Temperature: -31.75` no matter how the acutal temperature or humidity changes.
Just to clearify, these aren't by far the humidity and temperature of my working environmet.

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp

//Change the code below by your sketch
#include "Adafruit_Si7021.h"

Adafruit_Si7021 sensor = Adafruit_Si7021();

void setup() {
  Serial.begin(115200);

  // wait for serial port to open
  while (!Serial) {
    delay(10);
  }

  Serial.println("Si7021 test!");
  
  if (!sensor.begin()) {
    Serial.println("Did not find Si7021 sensor!");
    while (true);
  }
}

void loop() {
  Serial.print("Humidity:    "); Serial.print(sensor.readHumidity(), 2);
  Serial.print("\tTemperature: "); Serial.println(sensor.readTemperature(), 2);
  delay(1000);
}
```

### Debug Messages:
```
Enable Core debug level: Debug on tools menu of Arduino IDE, then put the serial output here 
```

stickbreaker, 2018-11-08:
@7h30n3 AdaFruit's library has some errors in it. I just submitted an pr [Fix ESP32 customization](https://github.com/adafruit/Adafruit_Si7021/pull/6).

Try using my modifications.

I don't have a Si7021 so this is untested code, but my changes are pretty straight forward.

Chuck.

7h30n3, 2018-11-09:
@stickbreaker Thank you. It now works.

closed 2018-11-09 by 7h30n3