espressif/arduino-esp32 #2038: ESP32 Feather not working with VEML7700 Gravity Digital Ambient Light Sensor

ddyast01, 2018-11-09:
Trying to receive ambient light data from a Gravity Digital Ambient Light Sensor connected to an Adafruit ESP32 Feather board. We have tried many different things to try and get this working with no success. We have been able to receive light sensor data on an Arduino Uno board.

Updated code to latest: 

cores\esp32\esp32-hal-i2c.h
cores\esp32\esp32-hal-i2c.c
libraries\Wire\src\Wire.h
libraries\Wire\src\Wire.cpp

We always receive either no lux value or a constant non-zero value depending on whether DFRobot_VEML7700.h includes (Wire.endTransmission(false);) or not.

[image]

We have confirmed the light sensor is connected with an I2C scanner. 

We are just using the example code from the light sensor to read the lux values.

#include <DFRobot_VEML7700.h>

/*!
 * @file readVEML770.ino
 * @brief DFRobot's Digital Ambient Light Sensor
 * @n This example read the Lux from Digital Ambient Light Sensor, and then print them
 *
 * @copyright  [DFRobot](http://www.dfrobot.com), 2016
 * @copyright GNU Lesser General Public License
 *
 * @version  V1.0
 * @date  2016-12-06
 */

#include <Wire.h>
#include "esp32-hal-i2c.h"

DFRobot_VEML7700 als;

void setup()
{
  Serial.begin(115200);
  als.begin();

}

void loop()
{
  float lux;
  als.getALSLux(lux);
  Serial.print("Lux: ");//Read Lux and print
  Serial.print(lux);
  Serial.println(" lx");
  delay(500);
};

ddyast01, 2018-11-09:
Used that library, but only received 0 values for lux.

[image]

ddyast01, 2018-11-14:
> @ddyast01 try this library instead [veml7700](https://github.com/stickbreaker/arduino-VEML7700)
> 
> Chuck.

Any other ideas for how to get this working?

Thanks,
Donald

sumanjha090, 2019-01-12:
I am also having same issue kindly anyone can help. I tried even with Arduino due board. But same result 0.0 lux

closed 2019-08-15 by stale[bot]