espressif/arduino-esp32 #2039: CMake dependency error "Can not found esp_bt.h" (menuconfiged)

wuyuanyi135, 2018-11-09:
### Hardware:
Board:							ESP32 Dev Module
Core Installation/update date:			08/Nov/2018
IDE name:							CLion
Flash Frequency:					40MHz
PSRAM enabled:                                          no
Upload Speed:						115200
Computer OS:                                            Ubuntu

### Description:
The native cmake and CLion built-in cmake do not compile the project with arduino-esp32 project residing in the local `components` directory.
The error was that the header `esp_bt.h` was not found.

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp

//Change the code below by your sketch
#include <Arduino.h>

void setup() {
}

void loop() {
}
```

### Debug Messages:
```
Enable Core debug level: Debug on tools menu of Arduino IDE, then put the serial output here 
```

ahornerr, 2018-12-09:
Check my PR here https://github.com/espressif/arduino-esp32/pull/2172

ahornerr, 2018-12-10:
My PR https://github.com/espressif/arduino-esp32/pull/2172 was merged, could you update and see if the issue still exists?

wuyuanyi135, 2018-12-10:
Thanks. Will check it later.

closed 2018-12-12 by me-no-dev