espressif/arduino-esp32 #2040: Reboot when I connect BLE

robcim, 2018-11-09:
I use WiFi, PubSubClient and BLE in my code arduino. 
When run the my device, if I connect in BLE to device his after second go in reboot.
The error is: E (125882) phy_init: failed to allocate memory for RF calibration data
                abort() was called at PC 0x40154fee on core 0

               Backtrace: 0x40094ca4:0x3fffb5c0 0x40094e79:0x3fffb5e0 0x40154fee:0x3fffb600 
               0x4018d42a:0x3fffb630 0x4018d4c6:0x3fffb660 0x4018d7c6:0x3fffb690 0x4018d946:0x3fffb6c0 
               0x4008dd44:0x3fffb6e0 0x40090c39:0x3fffb720

               Rebooting...

My board is Olimex Esp32-EVB, speed 115200 and freq 80Mhz

Lo sketch use 1511638 byte (74%) of the space for sketch. The max is 2031616 byte.
Le variabili globali usano 74764 byte (25%) di memoria dinamica, lasciando altri 220148 byte liberi per le variabili locali. Il massimo è 294912 byte.

Can you help me please?

chegewara, 2018-11-09:
First of all start with this tool toencode backtrace:
https://github.com/me-no-dev/EspExceptionDecoder

And error look like issue with esp-idf or board.

robcim, 2018-11-09:
Thank for help. I try with ESPExceptionDecoder. The result is:

0x40094ca4: invoke_abort at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/panic.c line 155
0x40094e79: abort at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/panic.c line 170
0x40154fee: esp_phy_load_cal_and_init at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/phy_init.c line 552
0x40090c39: vPortTaskWrapper at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/port.c line 141

Have you ever seen these errors.
Thank you.

chegewara, 2018-11-09:
Its problem with memory allocation which suggest you are short of heap:
https://github.com/espressif/esp-idf/blob/master/components/esp32/phy_init.c#L569-L576

closed 2019-08-15 by stale[bot]