espressif/arduino-esp32 #2041: secureClient.connect(SERVER_DOMAIN, 443) takes long time during handshake.

a-c-sreedhar-reddy, 2018-11-09:
### Hardware:
Board:							ESP32 Dev Module
IDE name:					       Arduino IDE
Upload Speed:					115200
Computer OS:                                        Windows 10

### Description:
I would like to POST data to a server every 10 sec. But after some requests the program enters this loop
```
    log_v("Performing the SSL/TLS handshake...");
    while ((ret = mbedtls_ssl_handshake(&ssl_client->ssl_ctx)) != 0) {
        if (ret != MBEDTLS_ERR_SSL_WANT_READ && ret != MBEDTLS_ERR_SSL_WANT_WRITE) {
            return handle_error(ret);
        }
	vTaskDelay(10 / portTICK_PERIOD_MS);
    }
```
in [ssl_client.cpp](https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFiClientSecure/src/ssl_client.cpp) and it never exits.
So is it good to have a timeout for this WHILE loop?

a-c-sreedhar-reddy, 2018-11-09:
@Jeroen88  Thanks. But  I am using wifiSecueClient which does not support reuseconnection.

Jeroen88, 2018-11-09:
reuse is handled by HTTP client, independent of the underlying transport provided by WiFiClient(-Secure). You can also send the headers yourself, and handle the (re-)connection yourself, but HTTP Client is far more easy. You could also take a look at my PR #1979, which changes the HTTP Client API (in line with the same changes I did for the ESP8266) to be used more easily with different WiFi clients

copercini, 2018-11-10:
@a-c-sreedhar-reddy what core version are you using? a similar bug was fix in the end of 2017

About your project design: is it not better to use MQTT? this keeps the connection open for you send and receive data all the time

a-c-sreedhar-reddy, 2018-11-10:
@copercini My Esp sdk version is v3.1-dev-661-gf586f5e6

a-c-sreedhar-reddy, 2018-11-10:
After adding a timeout to the WHILE loop it works good. 

But there is another problem. If at any point secureClient.connect(host, port) returns 0 then all subsequent connects (retries) fail for for many hours. Then  connect starts working automatically sometimes or I need to restart the device.  Should we be clearing anything in the memory or buffer before we retry?

copercini, 2018-11-10:
it seems that the old connections are not being closed properly, then at some point the memory will run out, causing all these problems (including the timeout issue)

Could you print the free memory using `Serial.println("Free heap: %u", ESP.getFreeHeap());` ?

a-c-sreedhar-reddy, 2018-11-10:
I am creating wifisecureclient variable in a function and everytime the function exits  the destructor is called which closes the connection. 
We also monitered the Heapmemory which is same.

closed 2019-02-21 by a-c-sreedhar-reddy