espressif/arduino-esp32 #2042: HardwareSerial print not readable....

pacmenDE, 2018-11-09:
### Hardware:
Board:							esp32-devkitc V4
Core Installation/update date:			?11/jul/2017?
IDE name:							?Arduino IDE
Flash Frequency:					80Mh
PSRAM enabled:                                          ?no?
Upload Speed:						?921000?
Computer OS:                                            ?Windows 10?

### Description:
Hello,
i have some trouble with esp32 an HardwareSerial!
here my script for testing the Serial.
The Serial.print statemt shows me the correct String "Hello World"
But when i use the HardwareSerial1 then i get some other characters.
[image]
(Serial.print()),
[image]
Myserial.print()

Where is the problem?

I would be glad if someone could help me with this....
Thank
Patrik

//Change the code below by your sketch
#include <HardwareSerial.h>
HardwareSerial Myserial(1);
void setup(){
 
   Serial.begin(9600);
   Myserial.begin(9600);
    delay(1000);
    Myserial.println("Hello World");
    Serial.println("Hello World");
}
void loop() {
{

lbernstone, 2018-11-09:
The bootloader uses 115200 baud. You are seeing the boot sequence garbled.  Set your code to use 115200 on Serial.

pacmenDE, 2018-11-09:
[image]
... the same.....

pacmenDE, 2018-11-09:
[image]
.....not resolved....:-(

lbernstone, 2018-11-09:
Turn off Serial1 (MySerial).  Run this code:
```
void setup() {
    Serial.begin(115200);
    Serial.println("Hello World!");
    //Serial1.begin(9600);
    //Serial1.println("Hello World!");
}
void loop(){}
```

pacmenDE, 2018-11-09:
[image]

"Serial" is running fine....

closed 2019-08-15 by stale[bot]