espressif/arduino-esp32 #2045: HardwareSerial must implement pure virtual Print::write(uint8*)

Bascy, 2018-11-10:
Using Eclipse, an error is shown in HardwareSerial.h stating that it instantiates HardwareSerial objects while Print::write(uint8_t) is not implemented.

Print.h indeed contains a pure virtual definition of write:

`virtual size_t write(uint8_t) = 0;`

Neither Stream.h nor HardwareSerial.h implement this method.

stickbreaker, 2018-11-10:
@Bascy really?
What about this? [HardwareSerial::write()](https://github.com/espressif/arduino-esp32/blob/85032b226c7775bae510e954112823a2ae32181a/cores/esp32/HardwareSerial.cpp#L114)

Chuck.

Bascy, 2018-11-10:
It misses the `override` derective

bertmelis, 2018-11-10:
https://stackoverflow.com/questions/13880205/is-the-override-keyword-just-a-check-for-a-overridden-virtual-method

Bascy, 2018-11-10:
It seems that only the C++ indexer in Eclipse is complaining about it, it actually does compile

kbickham, 2019-08-09:
It will compile but it will crash under certain 8 bit chunk protocols.first error
==================================

07:34:59.197 -> Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
07:34:59.197 -> Core 1 register dump:
07:34:59.197 -> PC      : 0x400d1faf  PS      : 0x00060f30  A0      : 0x800d21bd  A1      : 0x3ffb1f40  
07:34:59.197 -> A2      : 0x3ffbfea4  A3      : 0x00000001  A4      : 0x00000000  A5      : 0x00000000  
07:34:59.197 -> A6      : 0xffffffff  A7      : 0x3ffbebe0  A8      : 0x800d1f9e  A9      : 0x3ffb1f20  
07:34:59.197 -> A10     : 0x00000000  A11     : 0xfa000000  A12     : 0xc1400000  A13     : 0x00000001  
07:34:59.231 -> A14     : 0x00060120  A15     : 0x00000000  SAR     : 0x0000000a  EXCCAUSE: 0x0000001c  
07:34:59.231 -> EXCVADDR: 0x00000000  LBEG    : 0x4000c349  LEND    : 0x4000c36b  LCOUNT  : 0xffffffff  
07:34:59.231 -> 
07:34:59.231 -> Backtrace: 0x400d1faf:0x3ffb1f40 0x400d21ba:0x3ffb1f60 0x400d1a81:0x3ffb1f90 0x400d4a49:0x3ffb1fb0 0x4008543d:0x3ffb1fd0
07:34:59.264 -> 
07:34:59.264 -> Rebooting...
07:34:59.264 -> ets Jun  8 2016 00:22:57
07:34:59.264 -> 
07:34:59.264 -> rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
07:34:59.264 -> configsip: 0, SPIWP:0xee
07:34:59.264 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
07:34:59.264 -> mode:DIO, clock div:1
07:34:59.264 -> load:0x3fff0018,len:4
07:34:59.264 -> load:0x3fff001c,len:1100
07:34:59.264 -> load:0x40078000,len:9232
07:34:59.264 -> load:0x40080400,len:6400
07:34:59.264 -> entry 0x400806a8

=======================================================
0x400d1faf: Genie::autoPinger() at C:\Users\Lauren\Downloads\arduino-nightly\libraries\genieArduino\src\genieArduino.cpp line 368
0x400d21ba: Genie::DoEvents() at C:\Users\Lauren\Downloads\arduino-nightly\libraries\genieArduino\src\genieArduino.cpp line 416
0x400d1a81: loop() at C:\Port2ESPhdc\prelimAVRport1/prelimAVRport1.ino line 249
0x400d4a49: loopTask(void*) at C:\Users\Lauren\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.2\cores\esp32\main.cpp line 19
0x4008543d: vPortTaskWrapper at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/port.c line 143

kbickham, 2019-08-09:
It will compile, but I'm trying to rewrite a library for a display that is type casting to uint8_t as a write. If it was functional on the hardware, I would have slept last night.

<details>
<summary>Details</summary>

first error
==================================

07:34:59.197 -> Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
07:34:59.197 -> Core 1 register dump:
07:34:59.197 -> PC      : 0x400d1faf  PS      : 0x00060f30  A0      : 0x800d21bd  A1      : 0x3ffb1f40  
07:34:59.197 -> A2      : 0x3ffbfea4  A3      : 0x00000001  A4      : 0x00000000  A5      : 0x00000000  
07:34:59.197 -> A6      : 0xffffffff  A7      : 0x3ffbebe0  A8      : 0x800d1f9e  A9      : 0x3ffb1f20  
07:34:59.197 -> A10     : 0x00000000  A11     : 0xfa000000  A12     : 0xc1400000  A13     : 0x00000001  
07:34:59.231 -> A14     : 0x00060120  A15     : 0x00000000  SAR     : 0x0000000a  EXCCAUSE: 0x0000001c  
07:34:59.231 -> EXCVADDR: 0x00000000  LBEG    : 0x4000c349  LEND    : 0x4000c36b  LCOUNT  : 0xffffffff  
07:34:59.231 -> 
07:34:59.231 -> Backtrace: 0x400d1faf:0x3ffb1f40 0x400d21ba:0x3ffb1f60 0x400d1a81:0x3ffb1f90 0x400d4a49:0x3ffb1fb0 0x4008543d:0x3ffb1fd0
07:34:59.264 -> 
07:34:59.264 -> Rebooting...
07:34:59.264 -> ets Jun  8 2016 00:22:57
07:34:59.264 -> 
07:34:59.264 -> rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
07:34:59.264 -> configsip: 0, SPIWP:0xee
07:34:59.264 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
07:34:59.264 -> mode:DIO, clock div:1
07:34:59.264 -> load:0x3fff0018,len:4
07:34:59.264 -> load:0x3fff001c,len:1100
07:34:59.264 -> load:0x40078000,len:9232
07:34:59.264 -> load:0x40080400,len:6400
07:34:59.264 -> entry 0x400806a8

=======================================================
0x400d1faf: Genie::autoPinger() at C:\Users\Lauren\Downloads\arduino-nightly\libraries\genieArduino\src\genieArduino.cpp line 368
0x400d21ba: Genie::DoEvents() at C:\Users\Lauren\Downloads\arduino-nightly\libraries\genieArduino\src\genieArduino.cpp line 416
0x400d1a81: loop() at C:\Port2ESPhdc\prelimAVRport1/prelimAVRport1.ino line 249
0x400d4a49: loopTask(void*) at C:\Users\Lauren\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.2\cores\esp32\main.cpp line 19
0x4008543d: vPortTaskWrapper at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/port.c line 143

</details>

stickbreaker, 2019-08-09:
@kbickham 
>07:34:59.197 -> Guru Meditation Error: Core 1 panic'ed (LoadProhibited). Exception was unhandled.

is telling you that a memory storage location was referenced that is inaccessible.

Usually it means a pointer has a null value.  

What is `GENIE_READ_OBJ`?  You are telling the compiler to treat it as a single byte.  

Please post enough code to show your declaration, and the current value stored in it.

Memory map from page 25 of the [ESP32 Technical reference manual V4.0](https://www.espressif.com/sites/default/files/documentation/esp32_technical_reference_manual_en.pdf)
[image]

Only addresses in these range are valid to access. basically 0x3ffb0000 to 0x400c1FFF and 0x50000000 to 0x50001fff.

Chuck.

kbickham, 2019-08-09:
Well I feel a little dumb. I'm not that experienced, but I'm learning... I genuinely thank you for the response. The genie library is basically a uart protocol for commands to the display. I've invested so much into the interface design that I'm tied to it c. It sends/receives data in 8 bit chunks, wasn't my design. I've seen some ported libraries handling everything in 2 byte chunks (I actually can't find where they shift anything before output, as I don't think they're spending half their bandwidth transmitting zeros?).
`/////////////////////// GenieArduino 08/22/2017 ///////////////////////
//
//      Library to utilise the 4D Systems Genie interface to displays
//      that have been created using the Visi-Genie creator platform.
//      This is intended to be used with the Arduino platform.
//
//      Improvements/Updates by
//        4D Systems Engineering, August 2017, www.4dsystems.com.au
//		  	Antonio Brewer & 4D Systems Engineering, July 2017, www.4dsystems.com.au
//        4D Systems Engineering, October 2015, www.4dsystems.com.au
//        4D Systems Engineering, September 2015, www.4dsystems.com.au
//        4D Systems Engineering, August 2015, www.4dsystems.com.au
//        4D Systems Engineering, May 2015, www.4dsystems.com.au
//        Matt Jenkins, March 2015, www.majenko.com
//        Clinton Keith, January 2015, www.clintonkeith.com
//        4D Systems Engineering, July 2014, www.4dsystems.com.au
//        Clinton Keith, March 2014, www.clintonkeith.com
//        Clinton Keith, January 2014, www.clintonkeith.com
//        4D Systems Engineering, January 2014, www.4dsystems.com.au
//        4D Systems Engineering, September 2013, www.4dsystems.com.au
//      Written by
//        Rob Gray (GRAYnomad), June 2013, www.robgray.com
//      Based on code by
//        Gordon Henderson, February 2013, <projects@drogon.net>
//
//      Copyright (c) 2012-2014 4D Systems Pty Ltd, Sydney, Australia
/*********************************************************************
 * This file is part of genieArduino:
 *    genieArduino is free software: you can redistribute it and/or modify
 *    it under the terms of the GNU Lesser General Public License as
 *    published by the Free Software Foundation, either version 3 of the
 *    License, or (at your option) any later version.
 *
 *    genieArduino is distributed in the hope that it will be useful,
 *    but WITHOUT ANY WARRANTY; without even the implied warranty of
 *    MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
 *    GNU Lesser General Public License for more details.
 *
 *    You should have received a copy of the GNU Lesser General Public
 *    License along with genieArduino.
 *    If not, see <http://www.gnu.org/licenses/>.
 *********************************************************************/
#if defined (SPARK)
   #include "application.h"
   #define lowByte(w) ((uint8_t)((w) & 0xFF))
   #define highByte(w) ((uint8_t)((w) >> 8))
 #else
   #if defined(ARDUINO) && ARDUINO >= 100
     #include "Arduino.h"
   #else
     #include "WProgram.h"
   #endif
 #endif

#include "genieArduino.h"
#include <math.h>
#include <string.h>

#define DEC 10
#define HEX 16
#define OCT 8
#define BIN 2

// ######################################
// ## GENIE CLASS ####################### 
// ######################################
Genie::Genie() {
  UserHandler = NULL;
  UserByteReader = NULL;
[... 1432 lines cut ...]
                           HDC.HEAT.OUT.statusold=HDC.HEAT.OUT.statusnow;
                           HDC.HEAT.OUT.statusnow = true;
                           };
                     if (HDC.DRYP.state == false)                   
                          {HDC.AIR.OUT.statusold=HDC.AIR.OUT.statusnow;
                           HDC.AIR.OUT.statusnow = true;
                           HDC.HEAT.OUT.statusold=HDC.HEAT.OUT.statusnow;
                           HDC.HEAT.OUT.statusnow = false;     
                           };  
               };   
//      if (HDC.DRYP.mode == true)                                      
//              { if (HDC.DRYP.state == true)                     
//                   {HDC.AIR.OUT.statusold=HDC.AIR.OUT.statusnow;
//                    HDC.AIR.OUT.statusnow = true;
//                    HDC.HEAT.OUT.statusold=HDC.HEAT.OUT.statusnow;
//                    HDC.HEAT.OUT.statusnow = true;  
//                    };
//                if (HDC.DRYP.state == false)                   
//                   {HDC.AIR.OUT.statusold=HDC.AIR.OUT.statusnow;
//                    HDC.AIR.OUT.statusnow = false;
//                    HDC.HEAT.OUT.statusold=HDC.HEAT.OUT.statusnow;
//                    HDC.HEAT.OUT.statusnow = false;  
//                     }; 
             //  };
   //};

kbickham, 2019-08-09:
Well I feel a little dumb. I'm not that experienced, but I'm learning... I genuinely thank you for the response. The genie library is basically a uart protocol for commands to the display. I've invested so much into the interface design that I'm tied to it c. It sends/receives data in 8 bit chunks, wasn't my design. I've seen some ported libraries handling everything in 2 byte chunks (I actually can't find where they shift anything before output, as I don't think they're spending half their bandwidth transmitting zeros?).

<details>
<summary>Details</summary>
/////////////////////// GenieArduino 08/22/2017 ///////////////////////
//
//      Library to utilise the 4D Systems Genie interface to displays
//      that have been created using the Visi-Genie creator platform.
//      This is intended to be used with the Arduino platform.
//
//      Improvements/Updates by
//        4D Systems Engineering, August 2017, www.4dsystems.com.au
//		  	Antonio Brewer & 4D Systems Engineering, July 2017, www.4dsystems.com.au
//        4D Systems Engineering, October 2015, www.4dsystems.com.au
//        4D Systems Engineering, September 2015, www.4dsystems.com.au
//        4D Systems Engineering, August 2015, www.4dsystems.com.au
//        4D Systems Engineering, May 2015, www.4dsystems.com.au
//        Matt Jenkins, March 2015, www.majenko.com
//        Clinton Keith, January 2015, www.clintonkeith.com
//        4D Systems Engineering, July 2014, www.4dsystems.com.au
//        Clinton Keith, March 2014, www.clintonkeith.com
//        Clinton Keith, January 2014, www.clintonkeith.com
//        4D Systems Engineering, January 2014, www.4dsystems.com.au
//        4D Systems Engineering, September 2013, www.4dsystems.com.au
//      Written by
//        Rob Gray (GRAYnomad), June 2013, www.robgray.com
//      Based on code by
//        Gordon Henderson, February 2013, <projects@drogon.net>
//
//      Copyright (c) 2012-2014 4D Systems Pty Ltd, Sydney, Australia
/*********************************************************************
 * This file is part of genieArduino:
 *    genieArduino is free software: you can redistribute it and/or modify
 *    it under the terms of the GNU Lesser General Public License as
 *    published by the Free Software Foundation, either version 3 of the
 *    License, or (at your option) any later version.
 *
 *    genieArduino is distributed in the hope that it will be useful,
 *    but WITHOUT ANY WARRANTY; without even the implied warranty of
 *    MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
 *    GNU Lesser General Public License for more details.
 *
 *    You should have received a copy of the GNU Lesser General Public
 *    License along with genieArduino.
 *    If not, see <http://www.gnu.org/licenses/>.
 *********************************************************************/
#if defined (SPARK)
   #include "application.h"
   #define lowByte(w) ((uint8_t)((w) & 0xFF))
   #define highByte(w) ((uint8_t)((w) >> 8))
 #else
   #if defined(ARDUINO) && ARDUINO >= 100
     #include "Arduino.h"
   #else
     #include "WProgram.h"
   #endif
 #endif

#include "genieArduino.h"
#include <math.h>
#include <string.h>

#define DEC 10
#define HEX 16
#define OCT 8
#define BIN 2

// ######################################
// ## GENIE CLASS ####################### 
// ######################################
[... 1445 lines cut ...]
                           HDC.AIR.OUT.statusnow = true;
                           HDC.HEAT.OUT.statusold=HDC.HEAT.OUT.statusnow;
                           HDC.HEAT.OUT.statusnow = false;     
                           };  
               };   
//      if (HDC.DRYP.mode == true)                                      
//              { if (HDC.DRYP.state == true)                     
//                   {HDC.AIR.OUT.statusold=HDC.AIR.OUT.statusnow;
//                    HDC.AIR.OUT.statusnow = true;
//                    HDC.HEAT.OUT.statusold=HDC.HEAT.OUT.statusnow;
//                    HDC.HEAT.OUT.statusnow = true;  
//                    };
//                if (HDC.DRYP.state == false)                   
//                   {HDC.AIR.OUT.statusold=HDC.AIR.OUT.statusnow;
//                    HDC.AIR.OUT.statusnow = false;
//                    HDC.HEAT.OUT.statusold=HDC.HEAT.OUT.statusnow;
//                    HDC.HEAT.OUT.statusnow = false;  
//                     }; 
             //  };
   //};
           

 

</details>

stickbreaker, 2019-08-09:
@kbickham Ok, I tracked down the meaning of [`GENIE_READ_OBJ` here](https://github.com/4dsystems/ViSi-Genie-Arduino-Library/blob/5d06b0ff6d718b5d3196468ab1a93e2033ac8e4c/src/genieArduino.h#L67)

If I am understanding your code, it simplifies down to:
```c++
@define GENIE_READ_OBJ 0

Serial.write((uint8_t) GENIE_READ_OBJ );
```
But, I don't think that is your problem.  I scanned through your code, you redefine GENIE::BEGIN()
```c++
// ######################################
// ## Setup #############################
// ######################################
bool Genie::Begin(Stream &serial) {
deviceSerial = &serial;
display_uptime = millis(); // start uptime timer (ms)
genieStart = 1; // start form request on startup
ReadObject(GENIE_OBJ_FORM, (uint8_t)0x00); // send form request
uint32_t timeout_start = millis(); // timeout timer
while ( millis() - timeout_start <= 250 ) { // blocking loop, releases after 150ms to timeout, or sooner if display's detected.
if ( DoEvents() == GENIE_REPORT_OBJ && !genieStart ) return 1; // form is updated.
}
displayDetected = 0;
return 0; // timeout occured, status offline.
}
```

But, I never see anywhere in your code you actually call it?  so the `deviceSerial()` stream pointer is null? 

`deviceSerial->write()` would generate this error if `deviceSerial` is NULL.  

This is not a problem with arduino-esp32, your code has issues.

Chuck.

kbickham, 2019-08-10:
It's basically a hex command code define in a library I have to use for comms to display.I'm not I'm actually not utilizing that specific function,instead I used an event based approach sending flags from display...thank god. I found a work around, I apologize for borderline thread jacking.  should have turned the compiler flags all the way up. That thing gives warnings for unrelated libraries in the same parent directory that are missing returns without void type. It's working now. Appreciate your time.

closed 2019-10-23 by stale[bot]