espressif/arduino-esp32 #2047: HTTPClient getString halts on specific payload on latest core

mouridis, 2018-11-10:
### Hardware:
Board: WeMOS Lolin32 v.1.0.0 (using revision 1 ESP-WROOM-32 module)
Core Installation/update date: 11/11/2018
IDE name: Arduino IDE
Computer OS: Windows 10 x64

### Description:
I was using Arduino IDE Board Manager to install Arduino-ESP32 Core when I realized that the version of the core offered by this method was outdated. So, I uninstalled the Core and I reinstalled it using the Git method.

After this my application was not working as expected (compiled and flashed fine, it just wasn't downloading a specific payload from my http server). The payload is just a small JSON object. I tried to debug and found that the getString() function of my HTTPClient object was halting. Since, my sketch was very complicated, I used the library included BasicHttpClient example (using my URL) and the problem persisted.

The URL of the JSON object payload that allows the problem to be reproduced is: http://nikos.mouratid.is/ts2/timeSync.php

Let me note again, that using the ESP32 Core provided with the Arduino IDE Board Manager method, my application and the BasicHttpClient example work fine.

You can easily reproduce the issue using the sketch below (but also the BasicHttpClient example, using the URL above as payload). Just edit ssid and password.

Any ideas?

### Sketch:
```cpp

#include <WiFi.h>
#include <HTTPClient.h>
const char* ssid = "xxx";
const char* password =  "xxx";
void setup() {
  Serial.begin(115200);
  delay(4000);
  WiFi.begin(ssid, password);
  while (WiFi.status() != WL_CONNECTED) {
    delay(1000);
    Serial.println("Connecting to WiFi..");
  }
  Serial.println("Connected to the WiFi network");
}
void loop() {
  if ((WiFi.status() == WL_CONNECTED)) {
    HTTPClient http;
    http.begin("http://nikos.mouratid.is/ts2/timeSync.php");
    int httpCode = http.GET();
    if (httpCode > 0) { //Check for the returning code
        Serial.println(httpCode);
        String payload = http.getString();
        Serial.println(payload);
      }
    else {
      Serial.println("Error on HTTP request");
    }
    http.end();
  }
  delay(15000);
}

```

### Serial Output:
```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1324
load:0x40078000,len:7788
ho 0 tail 12 room 4
load:0x40080400,len:6448
entry 0x400806e8
Connecting to WiFi..
Connecting to WiFi..
Connected to the WiFi network
200
1
```

mouridis, 2018-11-11:
Done. I tested the above example using several commits to find the first commit where I can reproduce the issue. Results:

```
**Commit hash**					**Date**	**Result**
30b3eebabc32ba653defb4a4b4e3bc58624e83e3	20180728	works
a0f0bd930cfd2d607bf3d3288f46e2d265dd2e11	20180827	works
1e4bf14a3eef4894f65783b4940d98777d5e2263	20180918	works
ea61563c690ea4c41cc8cfa808799f84bcfa0540	20180918	works
96822d783f3ab6a56a69b227ba4d1a1a36c66268	20180921	issue reproduced
4e96bffe0ef0e3929ba0ecf540c9c6f299b31e14	20180918	works
85032b226c7775bae510e954112823a2ae32181a	20180927	issue reproduced
```

It seems that issue appears after commit 96822d783f3ab6a56a69b227ba4d1a1a36c66268.

lzw7603, 2018-11-13:
Roll back to the old version of the IDF maybe not the best solution.
Because the lwip bug can not be fix.
see https://github.com/espressif/esp-idf/releases/tag/v3.1.1  LWIP Related tag.

My low cost fix is changing the code in libraries/WiFi/src/WiFiClient.cpp
line 426 change "if(res < 0) {" to "if(res <= 0)"

mouridis, 2018-11-14:
> My low cost fix is changing the code in libraries/WiFi/src/WiFiClient.cpp
> line 426 change "if(res < 0) {" to "if(res <= 0)"

This seems to workaround the issue, but can you please roughly explain what it does and why I only had problem with my specific payload? It would like to understand exactly how "dirty" this fix is...

Thanks!

lzw7603, 2018-11-15:
If u roll back to old IDF, you will find that recv using with MSG_DONTWAIT flag will block the code sometime.
The blocking situation is random happened on some specified url.
For example the code on /libraries/WiFi/src/WiFiClient.cpp line 425
The code will block on "res = recv(fd(), buf, toRead, MSG_DONTWAIT);"
That is what i say about the LWIP bug.
The new IDF fix it , but make the "res" result on" /libraries/WiFi/src/WiFiClient.cpp" line 425 for zero
so "if(res < 0)" will make the while loop repeatly and not come out.

For your code, u use the http.getString(); will cause the http run to http.end() in the end.
And the code in http.end() will find that some data remain in rx buffer and must use flush() to clear.
And the code will block on the bug i talk about above.

mouridis, 2018-11-15:
@lzw7603 Thank you very much for the detailed explanation! Even though my code above does not halt ot http.end() but on http.getString() but your explanation covers it anyway. It seems flush() is also called from getString().

I guess a more "official" fix will follow...

mouridis, 2018-11-21:
Fixed after commit 01d22c8807cf9186e242c7c64198d67e88939db9

closed 2018-11-21 by mouridis