espressif/arduino-esp32 #2048: TLV493D-A1B6 board with a Adafruit HUZZAH32 not talking

tpitman, 2018-11-11:
### Hardware:
Board: Adafruit HUZZAH32
IDE name: Arduino IDE

### Description:
I am trying to use the TLV493D-A1B6 board with a Adafruit HUZZAH32 Arduino dev board.

I started with the magnetic sensor board pictured at the top of the github readme file associated with the TLV493D source.

I broke away the small sensor board from the main board so I could connect the HUZZAH32 to it. I made a cable that lets me connect it back to your board and that works fine. I have pull up resistors and have tried several different values including 8.7K and 1.2K.

If I use the PC software to get values everything works fine. On the scope I see several bytes of clock and data over and over really fast.

I install the TLV493D github library and it compiles and builds.

I connect it to my HUZZAH32 board and I get nothing.

When I watch the start up on a scope on the i2c lines I see the clock going and data going so I know I have the correct pins.

The problem is that when it tries to read data I get nothing. All of the examples do the same things. They just send an 8 byte worth of clock over and over about every second and nothing else.

During initialization I do see 4 bytes of clock and data, but am not sure what that means.

I did try replacing the files mentioned in other posts for the i2c. It didn't make a difference. I think that is because I have the latest esp32 library already.

I also downloaded a simple i2c scan test and it doesn't find any devices at any address. I can see the traffic on my scope, but nothing responds.

As I said above there is PC software that talks to the dev board that has the chip on it and when I watch that traffic there are responses and it works fine so I know the chip works. I have moved it back and forth between their dev board and it works and the HUZZAH32 and it doesn't.

The verbose debug log shows that nothing is acking and I know that is the case because the SDA never goes low when it is time for the TLV493D to respond.

How do I figure out what is going wrong?

stickbreaker, 2018-11-11:
@tpitman where are you probing with your scope? At the sensor or on the huzza?

I just read the user manual for your, sensor. It says the address should be 0x5E or 0x5F depending on address pin selection.

Try this scanner. Show me the output
[I2c scan](https://github.com/stickbreaker/arduino-esp32/blob/master/libraries/Wire/examples/i2c_scan/i2c_scan.ino)

Chuck.

tpitman, 2018-11-12:
I am probing near the huzza. I am going to check out the analog signal since so far I have just been looking at the digital scope.

That i2c scan is very similar to the one I tried and it doesn't give any devices found.

I will get back to you on what the signal looks like on the scope.

tpitman, 2018-11-12:
I have more information. I am able to use the nRF52 Feather from Adafruit and it works fine so there is something going on with the esp32 huzzah32 feather.

I have been interacting with the team from Infineon who make the sensor. Rather than repost everything we have done here I will post a link to that repo and the issue:

https://github.com/Infineon/TLV493D-A1B6-3DMagnetic-Sensor/issues/9

If someone can give some thoughts as to what might be going on with the esp32 huzzah32 that makes it not work with i2c with this sensor I would really appreciate it.

tpitman, 2018-11-12:
I am using the SCL and SDA lines that the board calls out on the silk screen. They are the last 2 pins on the shorter connector. They are also labeled pin 22 and 23.

tpitman, 2018-11-13:
> Can you test that those pins are working properly with a LED or something?
> Maybe a connection/power problem.

Um the pins seem to be getting driven properly as i see the clock and data
being just what it is supposed to be. The problem is the sensor doesn't ack
anything, but the same sensor acks fine when connected to other dev boards

tpitman, 2018-11-16:
Anyone have any ideas on this?

tpitman, 2018-11-16:
> @tpitman where are you probing with your scope? At the sensor or on the huzza?
>
> I just read the user manual for your, sensor. It says the address should be 0x5E or 0x5F depending on address pin [...]

The output is all ..

The sensor board doesn't seem to want to ack anything from the huzzah32. The exactly same board acks and works fine with the nrf52 feather.

Any ideas why that would be? Something about the signal, voltage or timing or something?

stickbreaker, 2018-11-16:
@tpitman can you try another I2C device? Discover where it is a Hardware fault or software fault.  I know the current `Wire()` driver works.  

Take a couple steps back, create a simple circuit to test:
*  just ESP32, Pullup Resistors(~2.4k) and one I2C device. 
* use previously unused pins on the ESP32
select `sdaPin, sclPin` from unused pins on the esp32, don't use 0-3,6-11,12,34-39
`Wire.begin(sdaPin,sclPin,100000);`
* connect your pullup resistors to these pins
* connect your I2C device to these pins
* use that scanner code, modified to use your pin selections.
* verify i2c device power

Chuck.

tpitman, 2018-11-16:
When you say just esp32 I am not sure what you mean. The esp32 is mounted on the adafruit feather huzzah32 board. I can't image how I would isolate it.

If you are talking about just doing what you say by using the adafruit huzzah32 board, then I can do that.

From what I can tell the SCL and SDA pins on the huzzah32 board are on pins 36 and 37 according to the schematic they have here:

https://learn.adafruit.com/assets/41630

That is if I read that right. If so, those are 2 pins in the 34-39 range you said not to use.

What pins would you suggest and how would I connect to them on this board?

Here is the link to the huzzah32 pinouts page:

https://learn.adafruit.com/adafruit-huzzah32-esp32-feather/pinouts

I am happy to do the experiment you ask if I am able to do so on this board.

Thank you for your time.

tpitman, 2018-11-16:
Looks like I might be mapping to the wrong pins in my previous message. I was looking at the physical pin numbers on that schematic I included when I thought the huzzah32 was using 36 and 37. If you meant IO numbers, then it would be fine. It looks like the huzzah32 maps scl and sda to IO22/23.

So can you clarify what pins I should not be using based on your message to me?

stickbreaker, 2018-11-16:
The numbers are base on the ESP32 GPIO values.  What is the board name you have selected inside the Arduion IDE?

IO22/23 are the standard pins for I2C GPIO22,GPIO23

tpitman, 2018-11-16:
[image]

stickbreaker, 2018-11-16:
try `Wire.begin(23,22,100000);` use the labeled SDA,SCL pins on the PCB.

you need GND, v3.3, SDA, SCL plus the Pullup resistors to connect the sensor.

If 23,22 doesn't work you could use 18,19 (as labeled on pcb)

When I said ESP32 I meant which ever module you were using.  I just wanted you to minimize the circuit complexity.  Nothing other than the basic I2C necessary parts connected.

Set the "Core Debug Level" to "verbose" this will display any warning to the Serial Monitor.

Chuck.

tpitman, 2018-11-16:
I don't have the sensor with me right now, so I will try this later tonight.

I have tried 8.7k and 1.2k resistors. Should those work or do I specifically need 2.4k?

stickbreaker, 2018-11-16:
I would uses between 2.4k and 4.7k depending on speed, with 10k speeds should be limited below 100kHz.  3.3k provides 1mA current which works well with 400kHz and 18 inches of wire length. I2C is only specified up to 1m(39.37 inches). 
The scan sketch must be able to see the device.  The device has to answer the probe, else there is a hardware fault.  Either wrong pins, wrong wiring, dead sensor.  You have validated the Sensor by testing it with other hardware, so that only leaves incorrect circuit wiring, damaged ESP32 pins, or Wrong ESP32 pins.

Chuck.

tpitman, 2018-11-16:
Looks like all I have near that are 4.7k. That should work, right? What speed would you limit it to at about 12 inches?

stickbreaker, 2018-11-16:
I would start with 100kHz.

Logic analyzers are problematic if it is a signal problem.  The analyzer has a hysteresis that may not match the actual devices. 

Do you know the hysteresis points for your analyzer,  What voltage defines a Low and what voltage defines a high. The ESP32 defines High as 0.75*Vdd and Low as 0.25*Vdd so voltages between 0.25*Vdd and 0.75*Vdd are not acknowledged.  With a 3.3V Vdd, A low is a signal below 0.825V, or a signal that rises to 2.475V.  When the signal exceeds 2.475V it becomes High until it fall back below 0.825V.

I would look at the actual voltages with an oscilloscope, **IF** there is any question of the voltages.

If one of the devices on the I2C bus is not driving the signals correctly, that could be an issue.  The ESP32 is spec'd to drive a low with 28mA at 0.495V.  With a 4.7k pullup, only 0.70mA required, Unless there is a hardware fault in the ESP32 .

the TLV493D-A1B6 is rated with a 3mA max output at 0.4V.  So the sensor is the weaker drive. 

It's hysteresis points are different they use 20% and 70% instead of 25% and 75% of the ESP32.

My simplest explanation is damaged pins on the ESP32. So, try different pins. or possibility the I2C peripheral is damaged. You could try using the second peripheral just by using `Wire1()` instead `Wire()`.

Chuck.

tpitman, 2018-11-16:
When I did look at the signals on the scope they appeared to be the same. I will check again though if I can't get this to work.

It could be pins so I will try to use different pins. I should be able to just pass any pins in to the Wire.begin as long as they come out on the header, right?

stickbreaker, 2018-11-16:
as long as you don't hit RX,TX or use one of the input only pins 34..39

tpitman, 2018-11-17:
I am not 100% sure what got it to finally work. I just added a delay (5 seconds), but then took that out and it still worked when I started from the IDE, but then when I power cycle nothing comes out on the display.

So I put a 1 second delay back in (this is right after initializing serial) and it works now.

I tried 4.7k ohm resistors right off and that works, but then I went back to my other board that has 1.2k resistors and it still works.

So suddenly everything is working on the huzzah32 and I don't have a good explanation. Doesn't feel right, but at least I can move forward.

Thanks for your help.

mhollfelder, 2018-11-19:
This is really a strange problem that you face. 

It could be that there are some internal timing issues and the sensor is not starting in a proper way on the HUZZAH32 board. So maybe a `delay()` gives time that some internal stuff can go on before resetting the sensor. However, it finally works which is got to hear.

Best regards,

Manuel

tpitman, 2018-11-19:
As I said in my last comment I did add a delay as part of some changes and it is now working, but I tool the delay back out and it is still working. So I am not sure why it is working, but it is so I am going to move on now.

closed 2018-11-19 by tpitman