espressif/arduino-esp32 #2049: Stream read time out Error

saaske, 2018-11-11:
### Hardware:
Board:							ESP32 Dev Module
IDE name:					        Arduino IDE
Flash Frequency:					80Mhz
Upload Speed:					115200?
Computer OS:                                          Windows 10
Core Version                                            3.2

> Got 948448 bytes from server
> [D][Updater.cpp:129] begin(): OTA Partition: app0
> Begin OTA. This may take a few mins to complete.
> Written only : 148899/948448. Retry?
> heap memory --283768
> Error Occurred. Error #: 6

when I try to do OTA from amazon S3 it fails most of the time. I tried increasing the timeout to 3sec but no use.

Jeroen88, 2018-11-11:
My PR #1979 solves timeout issues, most probably yours too.

saaske, 2018-11-12:
@Jeroen88 I pulled your code and tried running it but it still fails.

```
[V][HTTPClient.cpp:1072] handleHeaderResponse(): RX: 'Accept-Ranges: bytes'
[V][HTTPClient.cpp:1072] handleHeaderResponse(): RX: 'Content-Type: application/octet-stream'
[V][HTTPClient.cpp:1072] handleHeaderResponse(): RX: 'Content-Length: 948448'
[V][HTTPClient.cpp:1072] handleHeaderResponse(): RX: 'Server: AmazonS3'
[V][HTTPClient.cpp:1072] handleHeaderResponse(): RX: 'Connection: close'
[V][HTTPClient.cpp:1072] handleHeaderResponse(): RX: ''
[D][HTTPClient.cpp:1102] handleHeaderResponse(): code: 200
[D][HTTPClient.cpp:1105] handleHeaderResponse(): size: 948448
[D][HTTPUpdate.cpp:176] handleUpdate(): Header read fin.
[D][HTTPUpdate.cpp:177] handleUpdate(): Server header:
[D][HTTPUpdate.cpp:178] handleUpdate():  - code: 200
[D][HTTPUpdate.cpp:179] handleUpdate():  - len: 948448
[D][HTTPUpdate.cpp:185] handleUpdate(): ESP32 info:
[D][HTTPUpdate.cpp:229] handleUpdate(): runUpdate flash...
[D][Updater.cpp:136] begin(): OTA Partition: app1
[E][HTTPUpdate.cpp:339] runUpdate(): Update.writeStream failed! (Stream Read Timeout)
[E][HTTPUpdate.cpp:275] handleUpdate(): Update failed
[D][HTTPClient.cpp:358] disconnect(): tcp stop
```

Jeroen88, 2018-11-12:
Maybe in you try to declare a HTTPUpdate myHttpUpdate(15000) yourself and use that? The default timeout is 8000. 
If you call setLedPin(int ledPin = -1, uint8_t ledOn = HIGH) before starting the update (with the right LED pin), the led will start flashing during the update. It should flash regularly: it is on while reading a buffer from the server and off during writing a buffer to flash. This might give you a clue what is going on.

cyberman54, 2018-11-28:
I had similar problem and can confirm that the new Updater.cpp code solved the issue of disrupted writes streams. Be sure to set Timeout on your data stream (e.g. Wifi client) before starting OTA.

closed 2019-08-15 by stale[bot]