espressif/arduino-esp32 #2051: Adafruit HUZZAH32 - reboot

kamil2234, 2018-11-11:
### Hardware:
Board:							Adafruit HUZZAH32 - ESP32 Feather Board
Core Installation/update date:			?
IDE name:					       Arduino IDE
Flash Frequency:					80 MHz
PSRAM enabled:                                         ?
Upload Speed:					    9216000
Computer OS:                                            Windows 10

### Description:
Hi,

For example if I use function  blabla() with "return string" without 
assigning result (return "ok") to the variable like ( String bla_value = blabla(); ) from this function  my ESP32 is rebooting.  This is my code below.  Please let my know where is the issue.

`#include <WiFi.h>
#include <WiFiClient.h>
#include <ESP32WebServer.h>
#include <ESPmDNS.h>
 
#define SDA1 23
#define SCL1 22
#include <Wire.h>
#include <I2C_EEPROM.h>
AT24C32<0x57> eep; // Das EEProm auf der china üblichen RTC3231 
 
const char *ssid = "smartyAPek";
const char *password = "sdfasdfsdf";
 
ESP32WebServer server(80);  
 
int app = 0; 
String wynik;
 
struct MySetingsTab {
  char struct_start[10];
  char wifi_ssid[64]; 
  char wifi_password[64];
  int svrport; 
  int dhcp;
  int ip_0;
  int ip_1;
  int ip_2;
  int ip_3;
  int nm_0;
  int nm_1;
  int nm_2;
  int nm_3;
  int gw_0;
  int gw_1;
  int gw_2;
  int gw_3;
  int dns_p0;
  int dns_p1;
  int dns_p2;
  int dns_p3;
  int dns_s0;
  int dns_s1;
  int dns_s2;
  int dns_s3;
  char  system_login[64];
  char  system_password[64];
  // setings  for sleep mode option - work if this option is active
  int  active_time;
  int  sleep_time;
  // setings for remote raporting sensor values
  int  remote_raporting;
  int  frequency_sec;
  char remote_url[64];
 
  char struct_end[10];
 
} mysetingstab, setingsfromdb;
 
//test funkcji string
String blabla(){
 
Serial.println("blabla..");
 
return "ok";
}
 
void handleRoot() {
 
  server.send(200, "text/plain", "hello from ESP32!");
 
}
 
void handleNotFound(){
 
  String message = "File Not Found\n\n";
  message += "URI: ";
  message += server.uri();
  message += "\nMethod: ";
  message += (server.method() == HTTP_GET)?"GET":"POST";
  message += "\nArguments: ";
  message += server.args();
  message += "\n";
  for (uint8_t i=0; i<server.args(); i++){
    message += " " + server.argName(i) + ": " + server.arg(i) + "\n";
  }
  server.send(404, "text/plain", message);
 
}
 
String WifiClientMode(char wfssid[], char wfpassword[]) {
 
     Serial.println("try connect to wifi network");
     Serial.print("wifi network ssid: ");
     Serial.print("'");
     Serial.print(wfssid);
     Serial.println("'");
     Serial.print("wifi network  pass: ");
     Serial.print("'");
     Serial.print(wfpassword);
     Serial.println("'");
     WiFi.begin( wfssid, wfpassword);
 
     // Wait for connection
     int wait_time = 0;
     while (WiFi.status() != WL_CONNECTED) {
 
        delay(500);
        Serial.print(".");
 
     }
 
     Serial.println("");
     Serial.print("Connected to ");
     Serial.println( wfssid);
     Serial.println( wfpassword);
     Serial.print("IP address: ");
     Serial.println(WiFi.localIP());
     delay(10000);
     Serial.println("end wifi procedure starting");
}
 
void setup(void){
 
  Serial.begin(9600);
  delay(1000);
 
  Serial.println("esp starts ");
  delay(1000);
  Serial.println("wire begin");
 
// wire begin mode sda pin / scl pin // speed
  Wire.begin(SDA1,SCL1,100000); // piny dla esp32
 
  delay(2000);
 
  Serial.println("check db ");
  //GetSetingsfromdb();
 
   if(eep.ready()){
    Serial.println("Epprom ready...");
   }
 
  delay(2000);
 
 //String blavalue = 
 
 blabla();
   //  Serial.println(dupa);
 
  strcpy(mysetingstab.wifi_ssid ,"msdfsdfsdf"); 
  strcpy(mysetingstab.wifi_password ,"asdfsdfsdfasdf");
 
 if (strcmp(mysetingstab.wifi_ssid, ssid) != 0) {
    Serial.println("Try run Network run in WIFI_CLIENT_STA mode: ");
    WifiClientMode(mysetingstab.wifi_ssid, mysetingstab.wifi_password);
    Serial.println("Network run in WIFI_CLIENT_STA mode: ");
    //WiFi.printDiag(Serial);
    delay(5000);
 }
 
      if (MDNS.begin("esp32")) {
      Serial.println("MDNS responder started");
      }
 
        server.on("/", handleRoot);
 
        server.on("/inline", [](){
          server.send(200, "text/plain", "this works as well");
        });
 
        server.onNotFound(handleNotFound);
 
      server.begin();
 
      Serial.println("HTTP server started");
 
}
 
void loop(void){
  server.handleClient();
}`

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp

//Change the code below by your sketch
#include <Arduino.h>

void setup() {
}

void loop() {
}
`

### Debug Messages:
`IR⸮⸮1⸮⸮⸮⸮)))⸮⸮⸮⸮!⸮⸮⸮:⸮⸮⸮esp starts 
wire begin
check db 
Epprom ready...
blabla..
Try run Network run in WIFI_CLIENT_STA mode: 
try connect to wifi network
wifi network ssid: 'msdfsdfsdf'
wifi network  pass: 'asdfsdfsdfasdf'
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 2 - STA_START
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 0 - WIFI_READY
...[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 4 - STA_CONNECTED
.....................[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 7 - STA_GOT_IP
[D][WiFiGeneric.cpp:386] _eventCallback(): STA IP: 192.168.2.159, MASK: 255.255.255.0, GW: 192.168.2.1
.
Connected to msdfsdfsdf
asdfsdfsdfasdf
IP address: 192.168.2.159
end wifi procedure starting
Network run in WIFI_CLIENT_STA mode: 
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x4012b932  PS      : 0x00060c33  A0      : 0x8012ba64  A1      : 0x3ffb1e10  
A2      : 0x00000081  A3      : 0x00000003  A4      : 0x3ffbb4b8  A5      : 0x00000003  
A6      : 0x3ffbb468  A7      : 0x00000000  A8      : 0x00000081  A9      : 0x3ffb1de0  
A10     : 0x00000001  A11     : 0x3ffb8058  A12     : 0x00000001  A13     : 0x00000001  
A14     : 0x00060c20  A15     : 0x00000000  SAR     : 0x00000018  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000081  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  

Backtrace: 0x4012b932:0x3ffb1e10 0x4012ba61:0x3ffb1e30 0x4010c4d1:0x3ffb1e50 0x4010c4e7:0x3ffb1e70 0x400fab39:0x3ffb1e90 0x400e0a73:0x3ffb1ec0 0x400dfd27:0x3ffb1ef0 0x400e0329:0x3ffb1f10 0x400d673f:0x3ffb1f50 0x400d1f46:0x3ffb1f70 0x4012cc87:0x3ffb1fb0 0x4008c23d:0x3ffb1fd0

Rebooting...
⸮!⸮F⸮.⸮⸮b5⸮⸮gC⸮r^⸮A⸮sbC⸮esp starts 
wire begin
`
###Exception Decoder

PC: 0x4012b932: find_key at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/pthread/pthread_local_storage.c line 83
EXCVADDR: 0x00000081

Decoding stack results
0x4012b932: find_key at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/pthread/pthread_local_storage.c line 83
0x4012ba61: pthread_setspecific at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/pthread/pthread_local_storage.c line 215
0x4010c4d1: sys_thread_sem_init at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/port/esp32/freertos/sys_arch.c line 548
0x4010c4e7: sys_thread_sem_get at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/port/esp32/freertos/sys_arch.c line 511
0x400fab39: tcpip_api_call at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/lwip/src/api/tcpip.c line 423
0x400e0a73: _mdns_pcb_init at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/mdns/mdns_networking.c line 271
0x400dfd27: _mdns_enable_pcb at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/mdns/mdns.c line 2947
0x400e0329: mdns_init at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/mdns/mdns.c line 4027
0x400d673f: MDNSResponder::begin(char const*) at C:\Users\kamil\Documents\Arduino\hardware\espressif\esp32\libraries\ESPmDNS\src\ESPmDNS.cpp line 56
0x400d1f46: setup() at C:\Users\kamil\Documents\Arduino\esp32_serwer-test_v1_string_error/esp32_serwer-test_v1_string_error.ino line 162
0x4012cc87: loopTask(void*) at C:\Users\kamil\Documents\Arduino\hardware\espressif\esp32\cores\esp32\main.cpp line 15
0x4008c23d: vPortTaskWrapper at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/port.c line 141

stickbreaker, 2018-11-11:
@kamil2334 don't use lambda for callback. Interrupt callbacks need to be marked IRAM_ATTR to always be present in memory.

Chuck.

kamil2234, 2018-11-11:
Could you please let me know where I use lambda calback? I do not know what lambda is.

lbernstone, 2018-11-11:
Remark out lines until you determine exactly what is causing the error.  Your code has many issues, and this forum is not for training.

kamil2234, 2018-11-11:
But the same code works good on the esp8266.  I don't see obvious issue.

lbernstone, 2018-11-11:
Use the included Webserver.h instead of ESP32WebServer.h.

kamil2234, 2018-11-11:
I changed to #include <Web Server.h> and the same problem:

`Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x4012b49e  PS      : 0x00060033  A0      : 0x8012b5d0  A1      : 0x3ffb1e10  
A2      : 0x15122500  A3      : 0x00000003  A4      : 0x3ffbbd98  A5      : 0x00000003  
A6      : 0x3ffbbd48  A7      : 0x00000000  A8      : 0x15122500  A9      : 0x3ffb1de0  
A10     : 0x00000001  A11     : 0x3ffb8058  A12     : 0x00000001  A13     : 0x00000001  
A14     : 0x00060020  A15     : 0x00000000  SAR     : 0x00000018  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x15122500  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  

Backtrace: 0x4012b49e:0x3ffb1e10 0x4012b5cd:0x3ffb1e30 0x4010c03d:0x3ffb1e50 0x4010c053:0x3ffb1e70 0x400fa6a5:0x3ffb1e90 0x400e05c7:0x3ffb1ec0 0x400df87b:0x3ffb1ef0 0x400dfe7d:0x3ffb1f10 0x400d654f:0x3ffb1f50 0x400d1dc6:0x3ffb1f70 0x4012c7f3:0x3ffb1fb0 0x4008c1ad:0x3ffb1fd0

Rebooting...
`

`PC: 0x4012b49e: find_key at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/pthread/pthread_local_storage.c line 83
EXCVADDR: 0x15122500

Decoding stack results
0x4012b49e: find_key at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/pthread/pthread_local_storage.c line 83
0x4012b5cd: pthread_setspecific at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/pthread/pthread_local_storage.c line 215
0x4010c03d: sys_thread_sem_init at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/port/esp32/freertos/sys_arch.c line 548
0x4010c053: sys_thread_sem_get at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/port/esp32/freertos/sys_arch.c line 511
0x400fa6a5: tcpip_api_call at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/lwip/src/api/tcpip.c line 423
0x400e05c7: _mdns_pcb_init at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/mdns/mdns_networking.c line 271
0x400df87b: _mdns_enable_pcb at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/mdns/mdns.c line 2947
0x400dfe7d: mdns_init at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/mdns/mdns.c line 4027
0x400d654f: MDNSResponder::begin(char const*) at C:\Users\kamil\Documents\Arduino\hardware\espressif\esp32\libraries\ESPmDNS\src\ESPmDNS.cpp line 56
0x400d1dc6: setup() at C:\Users\kamil\Documents\Arduino\esp32_serwer-test_v1_string_error/esp32_serwer-test_v1_string_error.ino line 163
0x4012c7f3: loopTask(void*) at C:\Users\kamil\Documents\Arduino\hardware\espressif\esp32\cores\esp32\main.cpp line 15
0x4008c1ad: vPortTaskWrapper at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/port.c line 141
`

lbernstone, 2018-11-11:
https://hackingmajenkoblog.wordpress.com/2016/02/04/the-evils-of-arduino-strings/

kamil2234, 2018-11-12:
I have read this article and still I want to use dynamic memory allocation via String class.

kamil2234, 2018-11-13:
In my opinion there is a problem with handling structures and strings on ESP32 for arduino.  Look at this code below:

```
#include <ArduinoJson.h>
StaticJsonBuffer<2000> jsonBuffer;

struct MySetingsTab {
  char wifi_ssid[256]; 
} mysetingstab;

char* CovertStringToChararray(String data){
// Length (with one extra character for the null terminator)
   int str_len = data.length() + 1; 
// Prepare the character array (the buffer) 
   char char_array[str_len];
// Copy it over 
   data.toCharArray(char_array, str_len);
  return char_array;
} /* END CovertStringToChararray */ 

void setup() {
Serial.begin(9600);
JsonObject& root = jsonBuffer.createObject();
root["wifi_ssid"] = "blablablabla";
strcpy(mysetingstab.wifi_ssid ,CovertStringToChararray(root["wifi_ssid"]));
}

void loop() {
  // put your main code here, to run repeatedly:

}

```
And result:

```
Core 1 register dump:
PC      : 0x400013f9  PS      : 0x00060f30  A0      : 0x800d0bb4  A1      : 0x3ffb1f70  
A2      : 0x3ffbff88  A3      : 0x00000000  A4      : 0x000000ff  A5      : 0x0000ff00  
A6      : 0x00ff0000  A7      : 0xff000000  A8      : 0x00000000  A9      : 0x3ffb1f30  
A10     : 0x3ffbff88  A11     : 0x3ffb1f50  A12     : 0x3ffb1f5c  A13     : 0x00000000  
A14     : 0x00000000  A15     : 0x3ffb0060  SAR     : 0x0000001f  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x400013f9  LEND    : 0x4000140d  LCOUNT  : 0xffffffff  
```

PC: 0x400013f9
EXCVADDR: 0x00000000

Decoding stack results
```
0x400d0bb1: setup() at C:\Users\kamil\Documents\Arduino\esp32_string_and_array_test/esp32_string_and_array_test.ino line 40
0x400e3d0b: loopTask(void*) at C:\Users\kamil\Documents\Arduino\hardware\espressif\esp32\cores\esp32\main.cpp line 15
0x400870cd: vPortTaskWrapper at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/port.c line 141
```

chegewara, 2018-11-13:
Which one is the line:
`esp32_string_and_array_test.ino line 40`

kamil2234, 2018-11-13:
`strcpy(mysetingstab.wifi_ssid ,CovertStringToChararray(root["wifi_ssid"]));`

closed 2019-08-15 by stale[bot]