espressif/arduino-esp32 #2053: Protecting code and OTA update without showing code

Kamoool, 2018-11-13:
### Hardware:
Board:							ESP32-PICO-D$
IDE name:						Arduino IDE
Flash Frequency:					240Mhz

### Description:
Hello,
I'd like to enter market with my product based on ESP32-PICO-D4 module, but i have 2 problems:
1. How to protect code from reading by potential customers while still being able to put new code into the chip (OTA update)? - so eFuse is not a way to go.

2. I'd like to provide my customers with update files, so they can send them to ESP through browser (OTA update). But how to do it in the way they won't get anything from code? sending password protected .zip file? Or password protected .ino (somehow??)? Is there a way to do that?

Regards

lbernstone, 2018-11-13:
Encrypt the flash: https://docs.espressif.com/projects/esp-idf/en/latest/security/flash-encryption.html
Encrypt the updates with AES and an embedded private key: https://github.com/espressif/esp-idf/blob/master/components/mbedtls/port/include/aes_alt.h
Stream the firmware file through AES decrypt into OTA.

Kamoool, 2018-11-13:
Anyone has examples on that for Arduino IDE? Im not into ESP-IDF at all..

LByrne, 2018-11-20:
Newbie, just got a ESP32, old school, XP. Any help to use connect the IDE?

squidlydeadly, 2018-11-26:
> Anyone has examples on that for Arduino IDE? Im not into ESP-IDF at all..

Have you ever tried the PlatformIO add-on for Atom or VS code  ? If not, here is an example : https://www.youtube.com/watch?v=fhEermOCrMo . It lets you use the Arduino framwork without forcing you to use Arduino IDE.

Kamoool, 2018-11-27:
I've got whole code for my module written in arduino IDE, with OTA updates etc. The last thing I need is protecting it from being pulled out of chip and protecting future OTA updates. Thats why I'm asking for Arduino IDE examples.
I though that would b just some lines of code.

mouridis, 2018-11-28:
@Kamoool May I ask what exactly are you trying to protect? (I'm only asking to know if this is also useful for my project). Is there a way to pull the binary out of the flash memory and decompile it to source code?

bertmelis, 2018-11-28:
Is the Arduino license compatible with encrypted binaries? I thought you should provide relinkeable code.

mouridis, 2018-11-28:
@bertmelis It's kind of a gray area issue... http://forum.arduino.cc/index.php?topic=23326.0

TLS1000, 2018-12-03:
Hi, would using the chipid be a good option? in setup you set
if(chipid != xxxxx) donothing

the compiled binary is difficult to reverse-engineer, and by using chipid no one can use it on another board?

closed 2019-08-15 by stale[bot]

Ben79543, 2019-10-09:
Kamool: very interested to know if you found a solution way forward.
Did you manage encrypted otas binaries in your code? Or decrypt before upgrading?
Thanks.

TLS1000, 2019-10-10:
Hi, encrypting the binary file is not possible to my knowledge, I also think it doesn't really make sense.

I added code so the binary only works on one ESP32 chip, example:

`uint64_t chipid;
  chipid=ESP.getEfuseMac();//The chip ID is essentially its MAC address(length: 6 bytes).
  Serial.printf("ESP32 Chip ID = %04X",(uint16_t)(chipid>>32));//print High 2 bytes
  Serial.printf("%08X\n",(uint32_t)chipid);//print Low 4bytes.
  //B8E48DA4AE30
  while (chipid != 0x08DD95E350CC) {
    Serial.println("chip id and program id don't match!");
    delay(1000);
  }`

Nexus1212, 2020-03-26:
Hello,
I am also interested in a solution for this topic. Is there allready a possibility for:

1. Activate "Secure Boot V2" and "Flash encryption" with the arduino IDE or PlattformIO (manual for esp-idf: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/security/secure-boot-v2.html

2. Update via OTA encryptet the Firmware

A Code example could be helpfully!