espressif/arduino-esp32 #2054: nghttp2/nghttp2.h: No such file or directory

anteph, 2018-11-13:
Hi,

I'm trying to do some experiments with HTTP2 but I'm running into some troubles. From what I've been reading, we can use nghttp2, which is already included in IDF, for HTTP2 requests.

Nonetheless, I cannot seem to include the lib in the Arduino code. I've already pulled the latest version of the Arduino core and NGHTTP2 is indeed present in tools/sdk/includes.

I'm trying to include the lib pretty much like is done on the asio_http2.h file, with no luck so far:
#include <nghttp2/nghttp2.h>

It's probably something silly, but I really cannot make it work.

Additionally, if I try to include the nghttp2/asio_http2_client.h (I haven't yet analyzed in detail but seems to be a wrapper that maybe will be handy?) I also get an error in this include: 

fatal error: boost/system/error_code.hpp: No such file or directory

Thank you very much,
Nuno Santos

### Hardware:
Board:							ESP32 Dev Module
Core Installation/update date:			13/11/2018
IDE name:							Arduino IDE v1.8.5
Computer OS:                                            Windows 8.1

### Sketch:
```cpp

#include <nghttp2/nghttp2.h>

void setup() {}

  
void loop() {}
```

lbernstone, 2018-11-14:
I don't have any tips on how to use it, but the esp32 port seems to pack all the functionality into http_parser.h
`#include <http_parser.h>`

chegewara, 2018-11-14:
There is no `nghttp2.h` in tools/sdk/include path.

anteph, 2018-11-14:
Hi! Thanks for the feedback :)

@lbernstone thanks for pointing it. I'll take a look. But for now I was trying to somehow import the nghttp2.h. IDF has a wrapper called sh2lib.h that seemed very simple to use, but it imports the nghttp2.h.

@chegewara But in the asio_http2.h file, it is importing the nghttp2.h with the following include:
#include <nghttp2/nghttp2.h>

Shouldn't we also be able to import it, in case we want to use the nghttp2.h  functionalities directly?

Best regards,
Nuno Santos

chegewara, 2018-11-14:
Ok, i see the point. But now there we have something like that in esp-idf:
https://github.com/espressif/esp-idf/tree/master/components/nghttp

As you can see there is external library nghttp2. I dont know arduino very good, but i can see this library needs to be included in arduino-esp32 or somehow added into library.

anteph, 2018-11-14:
I'm also not sure how these lower level includes work. I was thinking that, since there's a reference to the nghttp2.h lib, that it was already somehow included in the Arduino core and we could directly use it.

One thing I did find was a libnghttp.a in tools/sdk/lib, although I'm not sure if this is a different lib for HTTP 1 or if it is effectively the nghttp2 lib.

Hopefully this has a solution, would really like to run some tests with HTTP2 :)

lbernstone, 2018-11-17:
Answers:
- Do not ascribe purpose to that which could easily be error
- If you want to make a library for HTTP2, it would be a great help to the community (in a couple years when we all catch up on the protocol)
- The asio_* headers seem to assume the boost libraries will be available.  It is not part of the IDF, and so really these headers should not be included.

I'll put together a PR to add the main library and remove the asio stuff.
Also, it is not unusual to need an extern "C" around some IDF bits.  That's part of what the library would wrap around the code.

lbernstone, 2018-11-17:
Thanks for your involvement.  This will clearly be a needed feature in the near future, and any content and examples you can add to the community knowledge will be very helpful.

PR #2068 (2018-11-19): Some fixes to nghttp2 to provide basic functionality. 
Added the missing main header. Removed the asio headers which require Boost libraries. Moved http_parser into the expected location.
Fixes #2054 (maybe?)
closed 2018-11-19 by me-no-dev