espressif/arduino-esp32 #2055: Bad handling of multiple devices connecting to BluetoothSerial. How to set a max limit of allowed BT connections.

mouridis, 2018-11-14:
### Hardware:
Board: WeMOS Lolin32 v.1.0.0 (using revision 1 ESP-WROOM-32 module)
Core Installation/update date: Commit 85032b226c7775bae510e954112823a2ae32181a
IDE name: Arduino IDE 1.8.6
Flash Frequency: 80MHz
PSRAM enabled: No
Upload Speed: 921600
Computer OS: Windows 10 x64

### Description:
Using BluetoothSerial, it's allowed for more than one device to pair and connect at the same time to ESP32. As it is now, if indeed a second device connects to ESP32, essentially it hijacks the connection. ESP32 receives data from both connected devices but all outgoing data is directed to the second (last) device that connected to it.

Searching through the library code, it seems the reason for this behavior is that the library only stores the BT handle of the last connected device and discards the old one. The following lines in BluetoothSerial.cpp are responsible for this:
```cpp
    case ESP_SPP_SRV_OPEN_EVT://Server connection open
        _spp_client = param->open.handle;
        xEventGroupSetBits(_spp_event_group, SPP_CONNECTED);
        log_i("ESP_SPP_SRV_OPEN_EVT");
        break;
```

I understand that modifying the library to handle multiple simultaneous connections, and offering methods to identify the source device of incoming data and recipient device for outgoing data, would require a huge overhaul of the library.

In the meantime, and in order to avoid BluetoothSerial misbehavior, maybe it would be a good idea to limit the number of allowed simultaneous connections to only 1.

The quick dirty fix that I did in my project is modifying this part of the BluetoothSerial.cpp:
```cpp
    case ESP_SPP_SRV_OPEN_EVT://Server connection open
        _spp_client = param->open.handle;
        xEventGroupSetBits(_spp_event_group, SPP_CONNECTED);
        log_i("ESP_SPP_SRV_OPEN_EVT");
        break;

    case ESP_SPP_CLOSE_EVT://Client connection closed
        _spp_client = 0;
        xEventGroupClearBits(_spp_event_group, SPP_CONNECTED);
        log_i("ESP_SPP_CLOSE_EVT");
        break;
```

to this:
```cpp
    case ESP_SPP_SRV_OPEN_EVT://Server connection open
        if (!_spp_client){
            _spp_client = param->open.handle;
        } else {
            secondConnectionAttempt = true;
            esp_spp_disconnect(param->open.handle);
        }
        xEventGroupSetBits(_spp_event_group, SPP_CONNECTED);
        log_i("ESP_SPP_SRV_OPEN_EVT");
        break;

    case ESP_SPP_CLOSE_EVT://Client connection closed
        if(secondConnectionAttempt) {
            secondConnectionAttempt = false;
        } else {
            _spp_client = 0;
        }
        xEventGroupClearBits(_spp_event_group, SPP_CONNECTED);
        log_i("ESP_SPP_CLOSE_EVT");
        break;
```

and of course adding the needed declaration for secondConnectionAttempt somewhere on the declarations section:
```cpp
static boolean secondConnectionAttempt;
```

In this way, when a second device attempts to connect, ESP32 disconnects from it immediately after connection and does not store it's BT handle.

The question is if the IDF offers a cleaner way to implement similar functionality by limiting the number of allowed simultaneous connections.

chegewara, 2018-11-14:
https://github.com/espressif/arduino-esp32/blob/master/tools/sdk/sdkconfig#L162

mouridis, 2018-11-14:
This appears to affect BLE. Does it apply to BT classic also?

chegewara, 2018-11-14:
This is controller setting so it is affecting both, BR/EDR and BLE:
`CONFIG_BTDM_CONTROLLER_BR_EDR_MAX_ACL_CONN`

copercini, 2018-11-14:
@mouridis please open a pull request with your changes, this is very helpful =)

mouridis, 2018-11-21:
Addressed on commit 0d564d7b1d7ed2a85dcbd80e4b4e6f6e393e486d

travis012, 2022-09-22:
This is still a problem in the 4.4.2 framework. There is no way to tell it to only support 1 connection?!  I have tried setting CONFIG_BT_ACL_CONNECTIONS to 1 and it only works the first time you connect. IE: If you connect then disconnect, you have to reset the chip to connect again.  A warning appears in the log for this second attempt:
`BT_BTM: btm_sec_l2cap_access_req() PSM: 3 no application registerd`

VojtechBartoska, 2022-09-23:
Reopening and adding to 2.0.6 milestone, we will investigate this issue for next bugsfix release @travis012 :)

travis012, 2022-09-23:
I have the correct fix for this now.  I didn't realize that I was posting on the Arduino esp32 project. When I did, I re-posted on the espidf project:
https://github.com/espressif/esp-idf/issues/9830#issuecomment-1254442333

Basically you should just have to set `BTDM_CTRL_BR_EDR_MAX_ACL_CONN` to 1.  I'm also using `esp_bt_gap_set_scan_mode` in the open and close events as [boblane1](https://github.com/boblane1) suggests but that doesn't seem to have any effect.

VojtechBartoska, 2022-11-16:
This seems to be solved already, @P-R-O-C-H-Y will retest it.

P-R-O-C-H-Y, 2022-11-23:
Hi @travis012, I have retested the BluetoothSerial and if 1 device is connected to ESP, than to be able to connect another device, the device 1 must be disconnected first and this issue was about BTSerial.

I see your last comment about `BTDM_CTRL_BR_EDR_MAX_ACL_CONN` so I will test it further more and see how to implement this option inside Arduino-esp32 core.

closed 2022-11-23 by P-R-O-C-H-Y