espressif/arduino-esp32 #2056: how to check if internet is available in ESP32

anandpanchal, 2018-11-14:
### Hardware:
Board:							        ESP32 Dev Module
Core Installation/update date:			2018
IDE name:							Arduino IDE
Flash Frequency:					        80Mhz
PSRAM enabled:                                              no
Upload Speed:						115200
Computer OS:                                                 Windows 7

Hello Everyone,

My problem is : Determine if WiFi is not available is quite simple.How can I check if internet connection is present on WiFi side?
For sure I can ping an external server periodically, but I would like to avoid this solution.

Thanks

WereCatf, 2018-11-14:
There is no magical "you has internets" - signal, the only way of checking is to, you know, try to connect to something on the Internet.

anandpanchal, 2018-11-15:
Thanks for the response @WereCatf 
is there any provision in esp32 to ping to any ip address, e.g 8.8.8.8,  to check that internet is coming or not ?

a-c-sreedhar-reddy, 2019-04-19:
@anandpanchal Hey did you find any solution?

anuragmishra81, 2019-06-12:
https://github.com/marian-craciunescu/ESP32Ping

closed 2019-08-25 by stale[bot]