espressif/arduino-esp32 #2057: SDWebServer example, files are taking too long to load

roman-podorvanov, 2018-11-14:
I'm using SDWebServer from examples without any changes, just uploaded my html, css, js etc. files to the card. The problem is the page loading is very slow, 1.3MB css took 19s to load. And some other files are not loading at all (net::ERR_CONNECTION_RESET).

[image]

MartinDavidWaller, 2018-11-15:
Interesting thought. I placed the ESP32 directly next to the machine talking to it and I still get WebServer failures when HTML such as:

    <link rel="stylesheet" type="text/css" href="styles.css" media="all" />
    <script type="text/javascript" src="jquery-3.2.1.min.js"></script>

gets processed. 

The other thing I noticed was that from a cold start - the first time the HTML gets processed - it all seems to work. If I then refresh the page it fails.

deniss1711, 2018-11-23:
Using Arduino 1.8.7, espressif/arduino-esp32 - 85032b226c7775bae510e954112823a2ae32181a
Same Issue here. Limiting the browser to network.http.max-persistent-connections-per-server = 1 in about:config -> all files (even the 700kB jscripts) are loaded correct from the sd card with 200 = OK.
This is not the default setting on any browser, so not a real solution for me.

Back to 6 connections i also get files with 0 Byte size and debug log shows
[D][WiFiClient.cpp:452] connected(): Disconnected: RES: 0, ERR: 128, without touching the files on the SD card.

zhivko, 2018-11-24:
I am seeing similar effects. Is lwip that esp32 rtos uses thread safe at all?

deniss1711, 2018-11-24:
Checked some things and found the server is blocking while sending the data stream. When now loading the index.html the browser tries to load all files from the sd card asynchronously.
I found no easy way to load the scripts with only one tcp connection.

Switched the server to this one which solved the problem:
https://github.com/boblemaire/ESPAsyncSDWebServer

Needs modfication on SD.begin and in void handleCreate (only used when uploading)

Make sure alle dependencies are solved. Building with the current version of Arduino IDE brings errors. 
I used PlatformIO with lib_deps = me-no-dev/AsyncTCP, alanswx/ESPAsyncWiFiManager, bblanchon/ArduinoJson, me-no-dev/ESPAsyncWebServer

Not done with testing on stability, but performace is better and i get 200 for all files.

Jorgen-VikingGod, 2019-02-26:
I can confirm, that the Async version is working "faster" - it is still slower than the ESP8266 sync version - but the normal WebServer (sync version) can not handle all the files from SPIFFS.

My ESP32 is running now for 5 days without reboot or issues on the ESPAsyncWebServer+AsyncTCP+ESPAsyncWiFiManager.
It controls some relays and servo motors and reading sensor data + has I2C OLED display in use.

closed 2019-08-15 by stale[bot]