espressif/arduino-esp32 #2058: USES part of the SDK function ,Esp32, rebooting repeatedly

Arduino8266, 2018-11-15:
### Hardware:
Board: ?was the IDE development board chosen for the ESP32 Dev Module, actually using ESP32 DEVKIT V1?  
Core Installation/update date:			?3/oct/2018?
IDE name:							?Arduino IDE 1.8.5?
Flash Frequency:					?80Mhz?
PSRAM enabled:                                          ?no?
Upload Speed:						?921600?
Computer OS:                                            ?Windows 10? 

### Description:
Some of the esp32 SDK functions used in the code, due to project requirements, caused the device to start rebooting repeatedly。

The following are the functions and serial port output that caused the reboot ，
Here are the functions that caused the reboot, 
(
1. esp_wifi_set_ps(WIFI_PS_MIN_MODEM);

2. esp_wifi_set_protocol( WIFI_IF_AP,  WIFI_PROTOCOL_11G );
)each of which has caused the reboot ，

The serial port output corresponding to the function has been listed ，Help me, thank you .

### Sketch:
```cpp
//Change the code below by your sketch
    WiFi.setHostname(spiffs.host_name.c_str());
    esp_base_mac_addr_set(sta_mac); 
    //esp_wifi_set_ps(WIFI_PS_MIN_MODEM); 
    //esp_wifi_set_protocol( WIFI_IF_AP,  WIFI_PROTOCOL_11G );
    WiFi.setTxPower((wifi_power_t)70);

```

### Debug Messages:
```
1.{
 started  
[INFO] Mounting FS...
【INFO】 Serial is : 115200
[INFO] Serial_config si : 134217756
[INFO] Config file found
[INFO] err_name file found
[INFO] We start to setup the network:
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x4014e45a  PS      : 0x00060730  A0      : 0x800db631  A1      : 0x3ffb1ec0  
A2      : 0x3ffc7368  A3      : 0x00000001  A4      : 0x3f401dca  A5      : 0x3ffb3916  
A6      : 0x00000011  A7      : 0x00000000  A8      : 0x00000000  A9      : 0x3ffb1eb0  
A10     : 0x00000000  A11     : 0x3ffc107e  A12     : 0x00000006  A13     : 0x3ffc6672  
A14     : 0x0000a100  A15     : 0x00000000  SAR     : 0x00000008  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000138  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  

Backtrace: 0x4014e45a:0x3ffb1ec0 0x400db62e:0x3ffb1ef0 0x400d24f0:0x3ffb1f80 0x4016837a:0x3ffb1fa0

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:808
load:0x40078000,len:6084
load:0x40080000,len:6696
entry 0x400802e4
⸮ }

2.{
 started  
[INFO] Mounting FS...
【INFO】 Serial is : 115200
[INFO] Serial_config si : 134217756
[INFO] Config file found
[INFO] err_name file found
[INFO] We start to setup the network:
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x4014e462  PS      : 0x00060730  A0      : 0x800db634  A1      : 0x3ffb1ec0  
A2      : 0x3ffc7368  A3      : 0x00000002  A4      : 0x00000001  A5      : 0x3ffb3916  
A6      : 0x00000011  A7      : 0x00000000  A8      : 0x00000000  A9      : 0x3ffb1eb0  
A10     : 0x00000000  A11     : 0x3ffc107e  A12     : 0x00000006  A13     : 0x3ffc6672  
A14     : 0x0000a100  A15     : 0x00000000  SAR     : 0x00000008  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000138  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  

Backtrace: 0x4014e462:0x3ffb1ec0 0x400db631:0x3ffb1ef0 0x400d24f0:0x3ffb1f80 0x401683f2:0x3ffb1fa0

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:808
load:0x40078000,len:6084
load:0x40080000,len:6696
entry 0x400802e4

}

```

Arduino8266, 2018-11-15:
> @ Arduino8266 loadprohibited通常表示无效指针。
> 
> 验证`spiffs.host_name.c_str();`实际包含值。
> 
> 如果`.c_str()`返回NULL，则会发生不好的事情。
> 查克。

Thank you for your response，I've done this before, Verify ` spiffs. Host_name; ` contain values, but not the results 。
Can't figure out why。 

  if (spiffs.host_name != "") {
    WiFi.setHostname(spiffs.host_name.c_str());
    esp_base_mac_addr_set(sta_mac);     
    //esp_wifi_set_ps(WIFI_PS_MIN_MODEM);  
    //esp_wifi_set_protocol( WIFI_IF_AP,  WIFI_PROTOCOL_11G );
    WiFi.setTxPower((wifi_power_t)70);
  }   
  if (spiffs.sta_ssid == "" || spiffs.sta_bssid == "") spiffs.network_mode=0;
#ifdef _DEBUG_
  Serial.print(F("[INFO] The network mode is:"));
  Serial.println(spiffs.network_mode);
  Serial.print(F("[INFO] The wifi gain is:"));
  Serial.println(WiFi.getTxPower());
#endif

stickbreaker, 2018-11-15:
@Arduino8266 increase your resolution on the content of host_name.  a cStr is terminated by a NULL, a String has an encoded length.

```c++
Serial.print("as string=");
Serial.println(spiffs.host_name);
Serial.print("length as string=");
Serial.println(spiffs.host_name.length(),DEC);
Serial.printf("cstr address =%p\n",spiffs.host_name.c_str());
Serial.printf("cstr Length=%d\n",strlen(spiffs.host_name.c_str()));
Serial.printf("as cStr=%s\n",spiffs.host_name.c_str());
```
are you using DBCS encoding for you host name?

Chuck.

Arduino8266, 2018-11-15:
> @ Arduino8266增加你对host_name内容的分辨率。cStr以NULL结尾，String具有编码长度。
> 
> ```c++
> Serial.print（“ as string = ”）;
> Serial.println（spiffs.host_name）;
> Serial.print（“ length as string = ”）;
> Serial.println（spiffs.host_name.length（），DEC）;
> Serial.printf（“ cstr address =％p \ n ”，spiffs.host_name.c_str（））;
> Serial.printf（“ cstr Length =％d \ n ”，strlen（spiffs.host_name.c_str（）））;
> Serial.printf（“ as cStr =％s \ n ”，spiffs.host_name.c_str（））;
> ```
> 
> 你是否使用DBCS编码作为主机名？
> 
> 查克。

1.  Thank you very much for your guidance, I am a programming small white, hope you can teach me 
  more, thank you.
2.  I don't know how to fix the device reboot yet .  thank  you .

stickbreaker, 2018-11-15:
What was the output when you ran this test code?

Chuck.

Arduino8266, 2018-11-16:
> 运行此测试代码时的输出是什么？
> 
> 查克。
  1.  esp_wifi_set_ps（WIFI_PS_MIN_MODEM）;    
    output ：
 {
 started  
[INFO] Mounting FS...
【INFO】 Serial is : 115200
[INFO] Serial_config si : 134217756
[INFO] Config file found
[INFO] err_name file found
[INFO] We start to setup the network:
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x4014e45a  PS      : 0x00060730  A0      : 0x800db631  A1      : 0x3ffb1ec0  
A2      : 0x3ffc7368  A3      : 0x00000001  A4      : 0x3f401dca  A5      : 0x3ffb3916  
A6      : 0x00000011  A7      : 0x00000000  A8      : 0x00000000  A9      : 0x3ffb1eb0  
A10     : 0x00000000  A11     : 0x3ffc107e  A12     : 0x00000006  A13     : 0x3ffc6672  
A14     : 0x0000a100  A15     : 0x00000000  SAR     : 0x00000008  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000138  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  

Backtrace: 0x4014e45a:0x3ffb1ec0 0x400db62e:0x3ffb1ef0 0x400d24f0:0x3ffb1f80 0x4016837a:0x3ffb1fa0

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:808
load:0x40078000,len:6084
load:0x40080000,len:6696
entry 0x400802e4
⸮ 
}

2.  esp_wifi_set_protocol( WIFI_IF_AP,  WIFI_PROTOCOL_11G );
  output
{
 started  
[INFO] Mounting FS...
【INFO】 Serial is : 115200
[INFO] Serial_config si : 134217756
[INFO] Config file found
[INFO] err_name file found
[INFO] We start to setup the network:
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x4014e462  PS      : 0x00060730  A0      : 0x800db634  A1      : 0x3ffb1ec0  
A2      : 0x3ffc7368  A3      : 0x00000002  A4      : 0x00000001  A5      : 0x3ffb3916  
A6      : 0x00000011  A7      : 0x00000000  A8      : 0x00000000  A9      : 0x3ffb1eb0  
A10     : 0x00000000  A11     : 0x3ffc107e  A12     : 0x00000006  A13     : 0x3ffc6672  
A14     : 0x0000a100  A15     : 0x00000000  SAR     : 0x00000008  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000138  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  

Backtrace: 0x4014e462:0x3ffb1ec0 0x400db631:0x3ffb1ef0 0x400d24f0:0x3ffb1f80 0x401683f2:0x3ffb1fa0

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:808
load:0x40078000,len:6084
load:0x40080000,len:6696
entry 0x400802e4 
}
Output, that's it. Thank you .

stickbreaker, 2018-11-16:
Where is the host_name displayed?

```c++
Serial.print("as string=");
Serial.println(spiffs.host_name);
Serial.print("length as string=");
Serial.println(spiffs.host_name.length(),DEC);
Serial.printf("cstr address =%p\n",spiffs.host_name.c_str());
Serial.printf("cstr Length=%d\n",strlen(spiffs.host_name.c_str()));
Serial.printf("as cStr=%s\n",spiffs.host_name.c_str());
```

Arduino8266, 2018-11-16:
> host_name在哪里显示？
> 
> ```c++
> Serial.print（“ as string = ”）; 
> Serial.println（spiffs.host_name）; 
> Serial.print（“ length as string = ”）; 
> Serial.println（spiffs.host_name.length（），DEC）; 
> Serial.printf（“ cstr address =％p \ n ”，spiffs.host_name.c_str（））; 
> Serial.printf（“ cstr Length =％d \ n ”，strlen（spiffs.host_name.c_str（）））; 
> Serial.printf（“ as cStr =％s \ n ”，spiffs.host_name.c_str（））;
> ```
1. host_name  string : 
   Serial.println（spiffs.host_name）;   
   Serial.printf（“ as cStr =％s \ n ”，spiffs.host_name.c_str（））;

2. host_name  string length : 
    Serial.println（spiffs.host_name.length（），DEC）; 
    Serial.printf（“ cstr Length =％d \ n ”，strlen（spiffs.host_name.c_str（）））; 

3. hsot_name string  address :  
    Serial.printf（“ cstr address =％p \ n ”，spiffs.host_name.c_str（））;

stickbreaker, 2018-11-16:
ok, we have a miscommunication, or purposeful misdirection.

I believe the original error was generated because the value pass in `host_name` is invalid, I have asked you to provide to me the value you are passing as `host_name`.  You have not chosen to provide this answer, why? 

Your last message is just a regurgitation of my code snippet.  It provides no additional information with which to solve your original problem. 

It appears that I cannot help you. good luck.

Chuck.

Arduino8266, 2018-11-16:
I'm sorry, Mr. Chuck .  I was lucky to meet you，Turns out my thinking is wrong，You don't  angry。
  Here's the output: 

       as string=esp32_hostname
       length as string=14
       cstr address =0x3ffb3df4 
       cstr Length=14
       as cStr=esp32_hostname

Looking forward to your reply. Thank you !

stickbreaker, 2018-11-16:
Ok. That proves my conjecture is incorrect.
Something else is the problem locus.

Post your actual `setup()` function, with all supporting definitions. 

Chuck.

Arduino8266, 2018-11-16:
//esp_wifi_set_ps(WIFI_PS_MIN_MODEM);  // ****** 1 
    //esp_wifi_set_protocol( WIFI_IF_AP,  WIFI_PROTOCOL_11G );   //*********  2
 
   These two lines, either line  be used,   the device will be restarted . 
   Not in use, normal operation.

------------------------------------------------------------------------------------------------------
#include <Arduino.h>
#include <AsyncMqttClient.h>       
#include <WiFi.h>
#include "esp_wifi.h"

extern "C" {   
    #include "esp_system.h" 
    #include "esp32-hal.h"
}

extern "C" {
  #include "freertos/FreeRTOS.h"
  #include "freertos/timers.h"
}

void ICACHE_FLASH_ATTR wifi_service::setup() {
   
#ifdef _DEBUG_
  Serial.println(F("[INFO] We start to setup the network:"));
#endif
  if (spiffs.host_name != "") {
    WiFi.setHostname(spiffs.host_name.c_str());
    esp_base_mac_addr_set(sta_mac);    
    WiFi.setTxPower((wifi_power_t)70); 
    //esp_wifi_set_ps(WIFI_PS_MIN_MODEM);  // ****** 1 
    //esp_wifi_set_protocol( WIFI_IF_AP,  WIFI_PROTOCOL_11G );   //*********  2

  }   
  if (spiffs.sta_ssid == "" || spiffs.sta_bssid == "") spiffs.network_mode=0;
#ifdef _DEBUG_
  Serial.print(F("[INFO] The network mode is:"));
  Serial.println(spiffs.network_mode);
//  Serial.print(F("[INFO] The phy mode of the newwork is:"));
//  //Serial.println(WiFi.getPhyMode());
  Serial.print(F("[INFO] The wifi gain is:"));
  Serial.println(WiFi.getTxPower());

  
#endif     
  switch(spiffs.network_mode){
    case 0:
#ifdef _DEBUG_
      Serial.println(F("[INFO] This is the AP model."));
#endif
      WiFi.mode(WIFI_AP);
      setup_ap(spiffs.ap_ip,spiffs.ap_mask,spiffs.host_name,spiffs.ap_pwd,spiffs.ap_dhcp);
      break;
    case 1:
#ifdef _DEBUG_
      Serial.println(F("[INFO] This is the STA model."));
#endif
      WiFi.mode(WIFI_AP);
      setup_ap(spiffs.ap_ip,spiffs.ap_mask,spiffs.host_name,spiffs.ap_pwd,spiffs.ap_dhcp);
      WiFi.mode(WIFI_STA);
      setup_sta(spiffs.sta_ssid,spiffs.sta_ssid_pwd,spiffs.sta_bssid,spiffs.sta_dhcp,spiffs.sta_ip,spiffs.sta_gw,spiffs.sta_mask,spiffs.sta_dns);
      break;
    case 2:
    case 3:
#ifdef _DEBUG_
      Serial.println(F("[INFO] This is the AP_STA model."));
#endif
      WiFi.mode(WIFI_AP_STA);
      setup_ap(spiffs.ap_ip,spiffs.ap_mask,spiffs.host_name,spiffs.ap_pwd,spiffs.ap_dhcp);
      setup_sta(spiffs.sta_ssid,spiffs.sta_ssid_pwd,spiffs.sta_bssid,spiffs.sta_dhcp,spiffs.sta_ip,spiffs.sta_gw,spiffs.sta_mask,spiffs.sta_dns);
      delay(500);
      break;
//deflaut:
      default:
#ifdef _DEBUG_
      Serial.println(F("[ERR] none"));
#endif
      break;    
  }
}//end setup()

stickbreaker, 2018-11-16:
It looks like you are using IDF with Arduino as a component.

I don't use Arduino as a component. 

I'm out of my depth, sorry.

Sorry.

closed 2019-08-15 by stale[bot]